riscv-collab/riscv-gnu-toolchain #20: "-msoft-float" flag generates broken binaries when using glibc

ccelio, 2015-01-26:
I attempted to use `riscv64-unknown-linux-gnu-gcc -msoft-float` to compile SPECINT (which has a shocking amount of floating point). 

The code compiles, but the resulting binaries compute garbage when it comes to floating point. I've been told that glibc does not "support" -msoft-float, so that's suddenly not a surprise.

**Question**

Is there a way for this combination to noisily complain to prevent users from thinking "my code compiled, thus it should work"?  Or should I just wait on a multi-lib implementation of glibc instead?

darius-bluespec, 2015-02-20:
I have created programs that use glibc and -msoft-float, though I have not tried SPECINT.  I am curious how you built your toolchain, as I suspect that is where the problem is.

In order to build a toolchain that supports -msoft-float with glibc, I did the following:
* Modify riscv-gnu-toolchain/Makefile to pass --without-fp to ./configure for glibc.
* Make the targets in riscv-gnu-toolchain/Makefile individually (build-binutils-linux, build-gcc-linux-stage1, etc.).
* When making build-glibc-linux, set: CFLAGS="-msoft-float -g -O2" ASFLAGS=-msoft-float

These steps ensures that glibc itself is compiled with software floating point.

It appears to me that multilib means something different for binutils as it does for glibc.  As far as I have been able to determine, glibc's idea of multilib is somewhat limited, typically only supporting 32 and 64 bit variations (and x32).

ccelio, 2015-02-20:
> "I am curious how you built your toolchain, as I suspect that is where the problem is." 

You are correct, I wasn't aware that I needed to modify riscv-gnu-toolchain to build glibc for soft-float. Silly me. Thanks for providing the required steps. Is it possible to provide this as a higher-level option for future users to leverage?

darius-bluespec, 2015-02-20:
On 02/20/2015 05:32 PM, Christopher Celio wrote:
> "I am curious how you built your toolchain, as I suspect that is
>     where the problem is."
>
> You are correct, I wasn't aware that I needed to modify
> riscv-gnu-toolchain to build glibc for [...]

Yes, that is certainly possible.  I have a couples patches to the
riscv-gnu-toolchain Makefile that I use to build various configurations
(soft-float, no-atomic, RV32).  I would be glad to clean it up a bit and
make a pull request.

ccelio, 2015-02-20:
Thanks, that'd be great.

closed 2015-02-20 by ccelio

sherry151, 2015-04-25:
Did this pull request make it into the riscv-tools repo?