riscv-collab/riscv-gnu-toolchain #24: Unable to build following instructions on readme page due to possible website problem.

sam-falvo, 2015-02-06:
After installing a bunch of undisclosed dependencies, I was finally able to execute:

    ./configure --prefox=/Volumes/workspace/opt/riscv
    make

After a moderate amount of text output, it fails with the following:

```
patching file libsanitizer/sanitizer_common/sanitizer_platform_limits_posix.h
patching file libsanitizer/sanitizer_common/sanitizer_linux.cc
mv src/gcc.tmp src/gcc
mkdir -p src
rm -rf src/original-newlib src/newlib-*
cd src && curl ftp://sourceware.org/pub/newlib/newlib-1.18.0.tar.gz | tar zx
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0curl: (7) Failed to connect to sourceware.org port 21: Connection refused
mv src/newlib-1.18.0 src/original-newlib
mv: rename src/newlib-1.18.0 to src/original-newlib: No such file or directory
make: *** [src/original-newlib] Error 1
```

In the interests of maximizing the liklihood of reproducible builds, is it please possible to vendor your dependencies?  It'd make getting started substantially easier.  Thanks!

aswaterman, 2015-02-06:
Looks like a network problem of some kind.  Perhaps ftp://sourceware.org
was temporarily down (but it seems to be up now).

Our internal approach to eliminate discrepancies due to different versions
of dependencies is to use the same Linux distro everywhere and install the
packages via the package manager.  A mitigating factor is that they are
standard packages required to build any gcc toolchain and are in no way
specific to the RISC-V one.

sam-falvo, 2015-02-06:
Thanks for the response.  I'll need to look further into this, if it's just a network blockage at work, or if it's something else.  If I can build the toolchain on my box at home, I'll close this issue.  Thanks again!

sbeamer, 2015-02-06:
If you do find a package that was missing and was necessary for you to install, please let us know.

Additionally, in your code snippet you appear to be using --prefox when your probably mean --prefix.

sam-falvo, 2015-02-07:
That's what I get for typing it in by hand instead of copying and pasting.

I tried at home, and it is able to access the site OK, so it looks like a network configuration issue while at work.

I am running into another issue that I don't know how to resolve.

    g++   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -DGENERATOR_FILE -static-libstdc++ -static-libgcc  -o build/gengtype \
	    build/gengtype.o build/errors.o build/gengtype-lex.o build/gengtype-parse.o build/gengtype-state.o build/version.o ../build-x86_64-unknown-linux-gnu/libiberty/libiberty.a
    build/gengtype.o: In function `adjust_field_rtx_def':
    /tmp/riscv-gnu-toolchain/src/newlib-gcc/gcc/gengtype.c:1137: undefined reference to `lexer_line'
    build/gengtype.o: In function `adjust_field_type(type*, options*)':
    /tmp/riscv-gnu-toolchain/src/newlib-gcc/gcc/gengtype.c:1413: undefined reference to `lexer_line'
    /tmp/riscv-gnu-toolchain/src/newlib-gcc/gcc/gengtype.c:1441: undefined reference to `lexer_line'
    /tmp/riscv-gnu-toolchain/src/newlib-gcc/gcc/gengtype.c:1435: undefined reference to `lexer_line'
    /tmp/riscv-gnu-toolchain/src/newlib-gcc/gcc/gengtype.c:1418: undefined reference to `lexer_line'
    ... listing of errors goes on for about 100 lines more ...

I noticed that I have bison installed, but not flex.

I'm thinking it'd be much more convenient to make sure adjust the top-most configure script to check for flex and bison.  I'd contribute a change myself if I knew what I was doing.  (Been a C/C++ hacker for decades; STILL don't have a clue how to properly use autotools.)

sam-falvo, 2015-02-07:
Running into another issue having fixed the missing flex dependency.

    checking for riscv64-unknown-elf-gcc... /tmp/riscv-gnu-toolchain/build-gcc-newlib/./gcc/xgcc -B/tmp/riscv-gnu-toolchain/build-gcc-newlib/./gcc/ -nostdinc -B/tmp/riscv-gnu-toolchain/build-gcc-newlib/riscv64-unknown-elf/newlib/ -isystem /tmp/riscv-gnu-toolchain/build-gcc-newlib/riscv64-unknown-elf/newlib/targ-include -isystem /tmp/riscv-gnu-toolchain/src/newlib-gcc/newlib/libc/include -B/tmp/riscv-gnu-toolchain/build-gcc-newlib/riscv64-unknown-elf/libgloss/riscv64 -L/tmp/riscv-gnu-toolchain/build-gcc-newlib/riscv64-unknown-elf/libgloss/libnosys -L/tmp/riscv-gnu-toolchain/src/newlib-gcc/libgloss/riscv64 -B/opt/riscv/riscv64-unknown-elf/bin/ -B/opt/riscv/riscv64-unknown-elf/lib/ -isystem /opt/riscv/riscv64-unknown-elf/include -isystem /opt/riscv/riscv64-unknown-elf/sys-include   
    checking for suffix of object files... configure: error: in `/tmp/riscv-gnu-toolchain/build-gcc-newlib/riscv64-unknown-elf/libgcc':
    configure: error: cannot compute suffix of object files: cannot compile
    See `config.log' for more details.
    make[2]: *** [configure-target-libgcc] Error 1
    make[2]: Leaving directory `/tmp/riscv-gnu-toolchain/build-gcc-newlib'
    make[1]: *** [all] Error 2
    make[1]: Leaving directory `/tmp/riscv-gnu-toolchain/build-gcc-newlib'
    make: *** [build-gcc-newlib] Error 2

Yet, when I checked the /tmp/riscv-gnu-toolchain/build-gcc-newlib/config.log file, I see this:

    configure:4427: checking for suffix of object files
    configure:4449: gcc -c   conftest.c >&5
    configure:4453: $? = 0
    configure:4474: result: o

This looks like a problem I cannot resolve on my own.  Thoughts?  Thanks for your help.

sbeamer, 2015-02-09:
On the README for the repo under [Prerequisites](https://github.com/ucb-bar/riscv-gnu-toolchain#prerequisites) we list bison and flex as recommended packages. I would double check that you have the rest of that list if you are linux. If you are on OS X, I would give the homebrew [packaged](https://github.com/riscv/homebrew-riscv) install a try.

sam-falvo, 2015-02-09:
Thanks, but your update doesn't address the current issue I'm seeing.

sam-falvo, 2015-02-09:
I'm going to close this issue.  I'm confident I'm not communicating correctly, I'm being a burden, or both, neither of which is my intention.

I was hoping to be able to get a clean build of the GNU toolchain working so that I can compare my home-made assembler implementation against something that is known working.  I'll try the homebrew package, but if that doesn't work (I've had problems with it w/ another RISC-V package), then I'll just pray my code works and hope for bug fixes from the community.

closed 2015-02-09 by sam-falvo

sbeamer, 2015-02-09:
We are sorry we were unable to fix your problem. When building gcc the relevant error messages sometimes get buried in the avalanche of output. If you find any bugs in our release or have suggestions for the documentation, please open a pull request or comment.

sam-falvo, 2015-02-09:
No worries!  Thanks again for the time you've been able to provide.  I'll
tackle this again when I have more resources. For now, hoping I get
everything right.  I can always change things later if needed.  :-)