riscv-collab/riscv-gnu-toolchain #31: Build error: ‘TARGET_ALIAS’ undeclared (first use in this function)

zhemao, 2015-02-26:
I'm trying to build the toolchain, but I keep getting the following error message. I suspect it may have something to do with the version of gcc installed on my system, which is fairly new. I'm using GCC 4.9.2.

    /home/zhehao/programs/research/reference-chip/riscv-tools/riscv-gnu-toolchain/build/src/binutils/gas/as.c:224:14: error: ‘TARGET_ALIAS’ undeclared (first use in this function)
         VERSION, TARGET_ALIAS, BFD_VERSION_STRING);
                  ^
    /home/zhehao/programs/research/reference-chip/riscv-tools/riscv-gnu-toolchain/build/src/binutils/gas/as.c:224:14: note: each undeclared identifier is reported only once for each function it appears in
    /home/zhehao/programs/research/reference-chip/riscv-tools/riscv-gnu-toolchain/build/src/binutils/gas/as.c: In function ‘parse_args’:
    /home/zhehao/programs/research/reference-chip/riscv-tools/riscv-gnu-toolchain/build/src/binutils/gas/as.c:635:5: error: ‘TARGET_ALIAS’ undeclared (first use in this function)
         TARGET_ALIAS);
         ^
    /home/zhehao/programs/research/reference-chip/riscv-tools/riscv-gnu-toolchain/build/src/binutils/gas/as.c:649:44: error: ‘TARGET_CANONICAL’ undeclared (first use in this function)
        fprintf (stderr, _("canonical = %s\n"), TARGET_CANONICAL);
                                                ^
    /home/zhehao/programs/research/reference-chip/riscv-tools/riscv-gnu-toolchain/build/src/binutils/gas/as.c:650:43: error: ‘TARGET_CPU’ undeclared (first use in this function)
        fprintf (stderr, _("cpu-type = %s\n"), TARGET_CPU);

aswaterman, 2015-02-26:
This is a generic binutils build issue issue (try googling for "error:
‘TARGET_ALIAS’ undeclared"), but the cause seems unspecific. I would
examine the build/build-binutils-newlib/config.log and config.status to see
if TARGET_ALIAS is defined and work backwards from there.

closed 2015-05-10 by aswaterman

e19293001, 2016-10-28:
Encountered this problem. 

Found out that C_INCLUDE_PATH had a trailing colon and solved my problem with:

if [ -z "$C_INCLUDE_PATH" ]; then
    export C_INCLUDE_PATH=/include/path
else
    export C_INCLUDE_PATH=/include/path:$C_INCLUDE_PATH
fi