riscv-collab/riscv-gnu-toolchain #42: stack pointer not initialized at runtime

oddish2211, 2015-05-04:
Maybe i'm totally wrong here, i'm not that familiar with GCC.

I'm trying to compile a simple c program for a softcore with this toolchain. By viewing the assembly/machine code i've noticed that the stack pointer starts at zero, which is undesired behaviour and makes my program crash.

By examining the default riscv.ld linker script i can see that there is a constant named _gp that is used to initialize the global pointer (gp register) in the default startup code crt0.S. But the stack pointer (sp register) is never initialized.

Did i stumble on a bug? or do i have to do something else to intialize the sp without writing my own startup code?

My toolchain is configured to use newlib and i compile my c code to RV32I machinecode.

aswaterman, 2015-05-04:
The execution environment (e.g., OS kernel) is expected to have initialized sp before jumping to the program's entry point. If you're in an embedded domain with different constraints, you need to use a different crt0 that sets up that kind of thing.

oddish2211, 2015-05-04:
Then i'm still curious about what the GP represents.

But thanks anyways, this already helps a lot.

closed 2015-05-04 by oddish2211

aswaterman, 2015-05-04:
Fair enough. gp is the ABI global pointer to the small data section. The OS doesn't initialize it because it's a property of the binary, not the execution environment (unlike the stack)