riscv-collab/riscv-gnu-toolchain #49: RVC: c.bnez not encoded correctly

svenstucki, 2015-06-14:
When compiling code for RV32C, c.bnez instructions are translated incorrectly from RV32. It looks like a simple encoding issue, where a few bits got lost or were shifted. Both the immediate and the involved register is wrong.

Example objdump output (RV32I):

    1c000052: 02051463            bnez  a0,1c00007a <synch_start>
    1c000056 <memory_init>:
    [...]
    1c00006a <zero_loop>:
    [...]
    1c00007a <synch_start>:

Corresponding RV32C code:

    1c000046: 601a                  bnez  s0,1c00004c <memory_init+0x4>
    1c000048 <memory_init>:
    [...]
    1c00005c <zero_loop>:
    [...]
    1c00006a <synch_start>:

svenstucki, 2015-06-15:
Here's another very simple example with code:

    example:
        bne x8, x0, bla
        c.bnez x8, bla

    bla:
        nop

Compile with `riscv64-unknown-elf-gcc -c -mrvc -march=RV32IC` to get:

    Disassembly of section .text:

    00000000 <example>:
       0: 00041363            bne x8,x0,6 <bla>
       4: 601a                  c.bnez  x8,a <bla+0x4>

    00000006 <bla>:
       6: c002                  c.nop

In this case the `c.bnez` is always off by 4 byte (i.e. it stays the same if e.g. nops are added after the branches). It's also interesting that the first branch isn't converted to its RVC equivalent.

aswaterman, 2015-06-15:
Thanks for the bug report.  Happily, it turns out that it was fairly
benign.  Branch offsets are only finalized at link time, and the ones that
show up in the object code are only there to improve the readability of the
objdump.  The assembler got it wrong, but the linker ultimately got it
right.

I pushed a patch that fixes this.

closed 2015-06-17 by aswaterman