riscv-collab/riscv-gnu-toolchain #52: Build of gcc fails in Stage 2 configure, "Unable to detect exception model."

lluixhi, 2015-06-25:
Commands run:
```
./configure --prefix=/opt/riscv --enable-multilib
make linux
```

Output:
```
Checking multilib configuration for libgcc...
mkdir -p -- riscv-unknown-linux-gnu/libgcc
Configuring in riscv-unknown-linux-gnu/libgcc
configure: creating cache ./config.cache
checking build system type... x86_64-unknown-linux-gnu
checking host system type... riscv-unknown-linux-gnu
checking for --enable-version-specific-runtime-libs... no
checking for a BSD-compatible install... /usr/bin/install -c
checking for gawk... /usr/bin/gawk
checking for riscv-unknown-linux-gnu-ar... /opt/riscv/riscv-unknown-linux-gnu/bin/ar
checking for riscv-unknown-linux-gnu-lipo... riscv-unknown-linux-gnu-lipo
checking for riscv-unknown-linux-gnu-nm... /home/lluixhi/git/riscv/riscv-gnu-toolchain-build/build-gcc-linux-stage1/./gcc/nm
checking for riscv-unknown-linux-gnu-ranlib... /opt/riscv/riscv-unknown-linux-gnu/bin/ranlib
checking for riscv-unknown-linux-gnu-strip... /opt/riscv/riscv-unknown-linux-gnu/bin/strip
checking whether ln -s works... yes
checking for riscv-unknown-linux-gnu-gcc...  /home/lluixhi/git/riscv/riscv-gnu-toolchain-build/build-gcc-linux-stage1/./gcc/xgcc -B/home/lluixhi/git/riscv/riscv-gnu-toolchain-build/build-gcc-linux-stage1/./gcc/ -B/opt/riscv/riscv-unknown-linux-gn
u/bin/ -B/opt/riscv/riscv-unknown-linux-gnu/lib/ -isystem /opt/riscv/riscv-unknown-linux-gnu/include -isystem /opt/riscv/riscv-unknown-linux-gnu/sys-include   
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether  /home/lluixhi/git/riscv/riscv-gnu-toolchain-build/build-gcc-linux-stage1/./gcc/xgcc -B/home/lluixhi/git/riscv/riscv-gnu-toolchain-build/build-gcc-linux-stage1/./gcc/ -B/opt/riscv/riscv-unknown-linux-gnu/bin/ -B/opt/riscv/riscv-u
nknown-linux-gnu/lib/ -isystem /opt/riscv/riscv-unknown-linux-gnu/include -isystem /opt/riscv/riscv-unknown-linux-gnu/sys-include    accepts -g... yes
checking for  /home/lluixhi/git/riscv/riscv-gnu-toolchain-build/build-gcc-linux-stage1/./gcc/xgcc -B/home/lluixhi/git/riscv/riscv-gnu-toolchain-build/build-gcc-linux-stage1/./gcc/ -B/opt/riscv/riscv-unknown-linux-gnu/bin/ -B/opt/riscv/riscv-unkno
wn-linux-gnu/lib/ -isystem /opt/riscv/riscv-unknown-linux-gnu/include -isystem /opt/riscv/riscv-unknown-linux-gnu/sys-include    option to accept ISO C89... unsupported
checking how to run the C preprocessor...  /home/lluixhi/git/riscv/riscv-gnu-toolchain-build/build-gcc-linux-stage1/./gcc/xgcc -B/home/lluixhi/git/riscv/riscv-gnu-toolchain-build/build-gcc-linux-stage1/./gcc/ -B/opt/riscv/riscv-unknown-linux-gnu/
bin/ -B/opt/riscv/riscv-unknown-linux-gnu/lib/ -isystem /opt/riscv/riscv-unknown-linux-gnu/include -isystem /opt/riscv/riscv-unknown-linux-gnu/sys-include    -E
checking size of double... 8
checking size of long double... 8
checking for inttypes.h... no
checking for stdint.h... no
checking for stdlib.h... no
checking for ftw.h... no
checking for unistd.h... no
checking for sys/stat.h... no
checking for sys/types.h... no
checking for string.h... no
checking for strings.h... no
checking for memory.h... no
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... no
checking whether decimal floating point is supported... no
configure: WARNING: decimal float is not supported for this target, ignored
checking whether fixed-point is supported... no
checking whether to use setjmp/longjmp exceptions... no
checking if the linker (/home/lluixhi/git/riscv/riscv-gnu-toolchain-build/build-gcc-linux-stage1/./gcc/collect-ld) is GNU ld... yes
checking for thread model used by GCC... single
checking whether assembler supports CFI directives... yes
checking for target glibc version... 0.0
checking for __attribute__((visibility("hidden")))... yes
checking whether the target assembler supports thread-local storage... yes
checking whether the thread-local storage support is from emutls... no
checking for init priority support... yes
configure: updating cache ./config.cache
configure: creating ./config.status
config.status: creating Makefile
config.status: creating auto-target.h
config.status: linking /home/lluixhi/git/riscv/riscv-gnu-toolchain-build/src/gcc/libgcc/enable-execute-stack-empty.c to enable-execute-stack.c
config.status: linking /home/lluixhi/git/riscv/riscv-gnu-toolchain-build/src/gcc/libgcc/unwind-generic.h to unwind.h
config.status: linking /home/lluixhi/git/riscv/riscv-gnu-toolchain-build/src/gcc/libgcc/config/no-unwind.h to md-unwind-support.h
config.status: linking /home/lluixhi/git/riscv/riscv-gnu-toolchain-build/src/gcc/libgcc/config/no-sfp-machine.h to sfp-machine.h
config.status: linking /home/lluixhi/git/riscv/riscv-gnu-toolchain-build/src/gcc/libgcc/gthr-single.h to gthr-default.h
config.status: executing default commands
[... 34 lines cut ...]
checking for stdlib.h... no
checking for ftw.h... no
checking for unistd.h... no
checking for sys/stat.h... no
checking for sys/types.h... no
checking for string.h... no
checking for strings.h... no
checking for memory.h... no
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... no
checking whether decimal floating point is supported... no
configure: WARNING: decimal float is not supported for this target, ignored
checking whether fixed-point is supported... no
checking whether to use setjmp/longjmp exceptions... unknown
configure: error: unable to detect exception model
Makefile:10901: recipe for target 'configure-target-libgcc' failed
make[2]: *** [configure-target-libgcc] Error 1
make[2]: Leaving directory '/home/lluixhi/git/riscv/riscv-gnu-toolchain-build/build-gcc-linux-stage1'
Makefile:143: recipe for target 'stamps/build-gcc-linux-stage1' failed
make[1]: *** [stamps/build-gcc-linux-stage1] Error 2
make[1]: Leaving directory '/home/lluixhi/git/riscv/riscv-gnu-toolchain-build'
Makefile:55: recipe for target 'linux' failed
make: *** [linux] Error 2
```

ccelio, 2015-06-25:
You need to look at the config.log to see what is actually going wrong.  Best guess is a path is wrong somewhere and a compiler or assembler is failing to build something.

lluixhi, 2015-06-25:
The path it's looking for includes on is "/opt/riscv/riscv-unknown-linux-gnu/sys-include:/opt/riscv/riscv-unknown-linux-gnu/include", but there is nothing at those locations.

It looks like glibc isn't being built before gcc's stage 2 is attempted.

a0u, 2015-06-28:
The issue lies with gcc stage 1, not stage 2.

Here is failing test from `build-gcc-linux-stage1/riscv-unknown-linux-gnu/32/libgcc/config.log`:

```
configure:4556: checking whether to use setjmp/longjmp exceptions
configure:: /scratch/a_ou/src/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/xgcc -B/scratch/a_ou/src/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/ -B/scratch/a_ou/tools2/riscv-unknown-linux-gnu/bin/ -B/scratch/a_ou/tools2/riscv-unknown-linux-gnu/lib/ -isystem /scratch/a_ou/tools2/riscv-unknown-linux-gnu/include -isystem /scratch/a_ou/tools2/riscv-unknown-linux-gnu/sys-include  -m32 -c --save-temps -fexceptions  conftest.c >&5
conftest.s: Assembler messages:
conftest.s:17: Error: register save offset not a multiple of 8
conftest.s:19: Error: register save offset not a multiple of 8
configure:: $? = 1
configure: failed program was:
| /* confdefs.h */
| #define PACKAGE_NAME "GNU C Runtime Library"
| #define PACKAGE_TARNAME "libgcc"
| #define PACKAGE_VERSION "1.0"
| #define PACKAGE_STRING "GNU C Runtime Library 1.0"
| #define PACKAGE_BUGREPORT ""
| #define PACKAGE_URL "http://www.gnu.org/software/libgcc/"
| #define SIZEOF_DOUBLE 8
| #define SIZEOF_LONG_DOUBLE 8
| #define HAVE_GETIPINFO 1
| /* end confdefs.h.  */
| 
| void bar ();
| void clean (int *);
| void foo ()
| {
|   int i __attribute__ ((cleanup (clean)));
|   bar();
| }
| 
configure:4587: result: unknown
configure:4603: error: unable to detect exception model
```

Lines 17 and 19 of `conftest.s` are `.cfi_offset 1, -4` and `.cfi_offset 1, -12`, respectively.

`xgcc` is invoking `gas` as follows:
```
/scratch/a_ou/src/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/as -v --traditional-format -m32 -o conftest.o conftest.s
```
which, by itself, doesn't seem problematic.

Below is `conftest.s` reproduced in full:
```Assembly
	.file	"conftest.c"
	.text
	.align	2
	.globl	foo
	.type	foo, @function
foo:
.LFB0:
	.cfi_startproc
	.cfi_personality 0x9b,DW.ref.__gcc_personality_v0
	.cfi_lsda 0x1b,.LLSDA0
	add	sp,sp,-32
	.cfi_def_cfa_offset 32
	sw	ra,28(sp)
	sw	s0,24(sp)
	sw	s1,20(sp)
	sw	s2,16(sp)
	.cfi_offset 1, -4
	.cfi_offset 8, -8
	.cfi_offset 9, -12
	.cfi_offset 18, -16
	add	s0,sp,32
	.cfi_def_cfa 8, 0
.LEHB0:
	call	bar
.LEHE0:
	li	s1,1
.L5:
	add	a5,s0,-20
	mv	a0,a5
.LEHB1:
	call	clean
	li	a5,1
	bne	s1,a5,.L8
	j	.L9
.L6:
	mv	s2,a0
	mv	s1,zero
	j	.L5
.L8:
	mv	a5,s2
	mv	a0,a5
	call	_Unwind_Resume
.LEHE1:
.L9:
	lw	ra,28(sp)
	lw	s0,24(sp)
	lw	s1,20(sp)
	lw	s2,16(sp)
	add	sp,sp,32
	jr	ra
	.cfi_endproc
.LFE0:
	.globl	__gcc_personality_v0
	.section	.gcc_except_table,"a",@progbits
.LLSDA0:
	.byte	0xff
	.byte	0xff
	.byte	0x3
	.byte	0x1a
	.4byte	.LEHB0-.LFB0
	.4byte	.LEHE0-.LEHB0
	.4byte	.L6-.LFB0
	.byte	0
	.4byte	.LEHB1-.LFB0
	.4byte	.LEHE1-.LEHB1
	.4byte	0
	.byte	0
	.text
	.size	foo, .-foo
	.hidden	DW.ref.__gcc_personality_v0
	.weak	DW.ref.__gcc_personality_v0
	.section	.sdata.DW.ref.__gcc_personality_v0,"awG",@progbits,DW.ref.__gcc_personality_v0,comdat
	.align	2
	.type	DW.ref.__gcc_personality_v0, @object
	.size	DW.ref.__gcc_personality_v0, 4
DW.ref.__gcc_personality_v0:
	.word	__gcc_personality_v0
	.ident	"GCC: (GNU) 5.1.0"
```

a0u, 2015-06-28:
The same autoconf test also fails, albeit with different assembler errors, when building riscv-gnu-toolchain configured with `--with-xlen=32`.

From `build-gcc-newlib/riscv32-unknown-elf/libgcc/config.log`:
```
configure:4556: checking whether to use setjmp/longjmp exceptions
configure:: /scratch/a_ou/src/riscv-gnu-toolchain/build-gcc-newlib/./gcc/xgcc -B/scratch/a_ou/src/riscv-gnu-toolchain/build-gcc-newlib/./gcc/ -nostdinc -B/scratch/a_ou/src/riscv-gnu-toolchain/build-gcc-newlib/riscv32-unknown-elf/newlib/ -isystem /scratch/a_ou/src/riscv-gnu-toolchain/build-gcc-newlib/riscv32-unknown-elf/newlib/targ-include -isystem /scratch/a_ou/src/riscv-gnu-toolchain/src/newlib-gcc/newlib/libc/include -B/scratch/a_ou/src/riscv-gnu-toolchain/build-gcc-newlib/riscv32-unknown-elf/libgloss/riscv32 -L/scratch/a_ou/src/riscv-gnu-toolchain/build-gcc-newlib/riscv32-unknown-elf/libgloss/libnosys -L/scratch/a_ou/src/riscv-gnu-toolchain/src/newlib-gcc/libgloss/riscv32 -B/scratch/a_ou/tools/riscv32-unknown-elf/bin/ -B/scratch/a_ou/tools/riscv32-unknown-elf/lib/ -isystem /scratch/a_ou/tools/riscv32-unknown-elf/include -isystem /scratch/a_ou/tools/riscv32-unknown-elf/sys-include    -c --save-temps -fexceptions  conftest.c >&5
conftest.s: Assembler messages:
conftest.s:13: Error: unrecognized opcode `sd ra,40(sp)'
conftest.s:14: Error: unrecognized opcode `sd s0,32(sp)'
conftest.s:15: Error: unrecognized opcode `sd s1,24(sp)'
conftest.s:16: Error: unrecognized opcode `sd s2,16(sp)'
conftest.s:45: Error: unrecognized opcode `ld ra,40(sp)'
conftest.s:46: Error: unrecognized opcode `ld s0,32(sp)'
conftest.s:47: Error: unrecognized opcode `ld s1,24(sp)'
conftest.s:48: Error: unrecognized opcode `ld s2,16(sp)'
configure:: $? = 1
configure: failed program was:
| /* confdefs.h */
| #define PACKAGE_NAME "GNU C Runtime Library"
| #define PACKAGE_TARNAME "libgcc"
| #define PACKAGE_VERSION "1.0"
| #define PACKAGE_STRING "GNU C Runtime Library 1.0"
| #define PACKAGE_BUGREPORT ""
| #define PACKAGE_URL "http://www.gnu.org/software/libgcc/"
| #define SIZEOF_DOUBLE 8
| #define SIZEOF_LONG_DOUBLE 8
| #define HAVE_INTTYPES_H 1
| #define HAVE_STDINT_H 1
| #define HAVE_STDLIB_H 1
| #define HAVE_UNISTD_H 1
| #define HAVE_SYS_STAT_H 1
| #define HAVE_SYS_TYPES_H 1
| #define HAVE_STRING_H 1
| #define STDC_HEADERS 1
| #define HAVE_GETIPINFO 1
| /* end confdefs.h.  */
| 
| void bar ();
| void clean (int *);
| void foo ()
| {
|   int i __attribute__ ((cleanup (clean)));
|   bar();
| }
| 
configure:4587: result: unknown
configure:4603: error: unable to detect exception model
```
It appears that `xgcc` is invoked without `-m32` and thus generates RV64 assembly which includes  `ld` and `sd` instructions.

ccelio, 2015-06-28:
In my limited amount of testing, it appears that this is the first commit that this issue shows up for me: 

https://github.com/riscv/riscv-gnu-toolchain/commit/7baf1c7c44026da0ac388960258e4802abd3c769

aswaterman, 2015-06-28:
> 7baf1c7

Thanks everyone, turns out this was a binutils regression that caused the
effects of the -m32/-m64 flags to be suppressed.  Resolved by
https://github.com/riscv/riscv-gnu-toolchain/commit/8ec335a7382c0fdbd6e5d9148f32180719361b2c

closed 2015-06-28 by aswaterman