riscv-collab/riscv-gnu-toolchain #60: The toolchain's configure --with-arch flag is not symmetric with gcc's --with-arch

ccelio, 2015-07-05:
Hey guys,

Currently, to build an RV32I toolchain we do:

    ../configure --with-xlen=32 --with-arch=I --prefix=$RISCV 

That's not symmetric with the other --with-arch flags, which expect "--with-arch=RV32I". 

Ideally, the toolchain configure would be performed as:

        ../configure --with-arch=RV32I --prefix=$RISCV 

(Conceptually) the "32" would be passed to "--with-xlen", and the entire "RV32I" string would be passed to the other "--with-arch" flags.

Configure is not my forte, so it's not obvious to me how to set the @XLEN@ variable based on the @ARCH@ variable, and whether you can perform that action if the --with-xlen flag was used or not.

Although perhaps we no longer need the --with-xlen flag, in which case XLEN is just ${ARCH:2:2}?

palmer-dabbelt, 2015-07-06:
I'm working on this, it should get fixed kind of soon.

palmer-dabbelt, 2016-02-28:
I'm not working on this.

palmer-dabbelt, 2016-05-30:
This is fixed by https://github.com/riscv/riscv-gnu-toolchain/pull/146

closed 2016-05-30 by palmer-dabbelt