riscv-collab/riscv-gnu-toolchain #62: 32-Bit Linux compilation broken

s-macke, 2015-07-13:
One of the latest patches break the compilation of the Linux kernel for the 32-Bit toolchain. 

The problems occurs somewhere after patch 3ae3d5b0bb1d3ee3de600b1440db980d636480b2

The output of the Linux kernel compilation is: 

CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  CHK     include/generated/bounds.h
  CHK     include/generated/asm-offsets.h
  CALL    scripts/checksyscalls.sh
  CHK     include/generated/compile.h
  LINK    vmlinux
  LD      vmlinux.o
  MODPOST vmlinux.o
  GEN     .version
  CHK     include/generated/compile.h
  UPD     include/generated/compile.h
  CC      init/version.o
  LD      init/built-in.o
init/built-in.o: In function `run_init_process':
main.c:(.text+0x0): undefined reference to `__riscv_save_2'
main.c:(.text+0x24): undefined reference to `__riscv_restore_2'
main.c:(.text+0x2c): undefined reference to `__riscv_save_1'
init/built-in.o: In function `try_to_run_init_process':
main.c:(.text+0x70): undefined reference to `__riscv_restore_1'
init/built-in.o: In function `match_dev_by_uuid':
do_mounts.c:(.text+0x84): undefined reference to `__riscv_save_0'
do_mounts.c:(.text+0x9c): undefined reference to `__riscv_restore_0'
init/built-in.o: In function `rootfs_mount':
do_mounts.c:(.text+0xec): undefined reference to `__riscv_save_2'
init/built-in.o: In function `.L97':
do_mounts.c:(.text+0x474): undefined reference to `__riscv_restore_2'
do_mounts.c:(.text+0x488): undefined reference to `__riscv_save_12'
init/built-in.o: In function `.L35':
calibrate.c:(.text+0x860): undefined reference to `__riscv_restore_12'
init/built-in.o: In function `.L9':
main.c:(.init.text+0x8c): undefined reference to `__riscv_save_7'
init/built-in.o: In function `.L14':
main.c:(.init.text+0x128): undefined reference to `__riscv_restore_7'

palmer-dabbelt, 2015-07-13:
I think something like this

https://github.com/palmer-dabbelt/riscv-linux/tree/save-restore

should fix your problem.  Do you mind giving it a shot?  If so I'll merge it.  I'd try, but I'm messing around with my toolchain and don't want to build one from scratch right now.

s-macke, 2015-07-13:
Thanks, will give it a try.

s-macke, 2015-07-13:
It works with the -mno-save-restore. I guess you can close the issue when the patch is in the official repository.

s-macke, 2015-07-19:
I saw, that the official repository was patched.

closed 2015-07-19 by s-macke