riscv-collab/riscv-gnu-toolchain #63: shared libraries crash for at least for 32-Bit Toolchain

s-macke, 2015-07-14:
Hi,

I managed after some time and patches to run Linux on the RV32 architecture. However, shared libraries don't seem to work. They all crash with the same segmentation fault at the same address.
The same code compiled with --static works without problems.

Several month ago I have tried the 64-Bit toolchain with success.

Debug output:

/ # LD_DEBUG=all nano                                                           
        29:                                                                     
        29:      file=libz.so.1 [0];  needed by nano [0]                        
        29:      find library=libz.so.1 [0]; searching                          
        29:       search cache=/etc/ld.so.cache                                 
        29:       search path=/lib32/tls:/lib32:/usr/lib32/tls:/usr/lib32       
         (system search path)                                                   
        29:        trying file=/lib32/tls/libz.so.1                             
        29:        trying file=/lib32/libz.so.1                                 
        29:        trying file=/usr/lib32/tls/libz.so.1                         
[   33.810000] random: nonblocking pool is initialized                          
        29:        trying file=/usr/lib32/libz.so.1                             
        29:                                                                     
        29:      file=libz.so.1 [0];  generating link map                       
        29:        dynamic: 0x5c03a3b4  base: 0x5c023000   size: 0x00017594     
        29:          entry: 0x5c024880  phdr: 0x5c023034  phnum:          4     
        29:                                                                     
        29:                                                                     
        29:      file=libncurses.so.5 [0];  needed by nano [0]                  
        29:      find library=libncurses.so.5 [0]; searching                    
        29:       search cache=/etc/ld.so.cache                                 
        29:       search path=/lib32:/usr/lib32          (system search path)   
        29:        trying file=/lib32/libncurses.so.5                           
        29:        trying file=/usr/lib32/libncurses.so.5                       
        29:                                                                     
        29:      file=libncurses.so.5 [0];  generating link map                 
        29:        dynamic: 0x5c07c8b4  base: 0x5c03b000   size: 0x000448f4     
        29:          entry: 0x5c047ce0  phdr: 0x5c03b034  phnum:          4     
        29:                                                                     
        29:                                                                     
        29:      file=libc.so.6 [0];  needed by nano [0]                        
        29:      find library=libc.so.6 [0]; searching                          
        29:       search cache=/etc/ld.so.cache                                 
        29:       search path=/lib32:/usr/lib32          (system search path)   
        29:        trying file=/lib32/libc.so.6                                 
        29:                                                                     
        29:      file=libc.so.6 [0];  generating link map                       
        29:        dynamic: 0x5c1bff20  base: 0x5c080000   size: 0x001433f0     
        29:          entry: 0x5c097648  phdr: 0x5c080034  phnum:          9     
        29:                                                                     
        29:      checking for version `GLIBC_2.20' in file /lib32/libc.so.6 [0] 
required by file nano [0]                                                       
        29:      checking for version `GLIBC_2.20' in file /lib32/libc.so.6 [0] 
required by file /usr/lib32/libz.so.1 [0]                                       
        29:      checking for version `GLIBC_2.20' in file /lib32/libc.so.6 [0] 
required by file /usr/lib32/libncurses.so.5 [0]                                 
        29:      checking for version `GLIBC_2.20' in file /lib32/ld.so.1 [0] re
quired by file /lib32/libc.so.6 [0]                                             
        29:      checking for version `GLIBC_PRIVATE' in file /lib32/ld.so.1 [0]
 required by file /lib32/libc.so.6 [0]                                          
        29:                                                                     
        29:      Initial object scopes                                          
        29:      object=nano [0]                                                
        29:       scope 0: nano /usr/lib32/libz.so.1 /usr/lib32/libncurses.so.5 
/lib32/libc.so.6 /lib32/ld.so.1                                                 
        29:                                                                     
        29:      object=linux-vdso.so.1 [0]                                     
        29:       scope 0: nano /usr/lib32/libz.so.1 /usr/lib32/libncurses.so.5 
/lib32/libc.so.6 /lib32/ld.so.1                                                 
        29:       scope 1: linux-vdso.so.1                                      
        29:                                                                     
[... 128 lines cut ...]
        29:      symbol=__libc_start_main;  lookup in file=nano [0]             
        29:      symbol=__libc_start_main;  lookup in file=/usr/lib32/libz.so.1 
[0]                                                                             
        29:      symbol=__libc_start_main;  lookup in file=/usr/lib32/libncurses
.so.5 [0]                                                                       
        29:      symbol=__libc_start_main;  lookup in file=/lib32/libc.so.6 [0] 
        29:      binding file nano [0] to /lib32/libc.so.6 [0]: normal symbol `_
_libc_start_main' [GLIBC_2.20]                                                  
[  127.820000] nano[29]: unhandled signal 11 code 0x30001 at 0x02c79c62 in ld-2.
21.so[5c000000+1e000]                                                           
[  127.820000] CPU: 0 PID: 29 Comm: nano Not tainted 4.1.2-g9eb60e4-dirty #4    
[  127.820000] task: c1492060 ti: c1584000 task.ti: c1584000                    
[  127.820000] sepc: 02c79c62 ra : 5c097544 sp : b7ddddc0                       
[  127.820000]  gp : 00822b50 tp : 5c1c44a0 t0 : 5c097444                       
[  127.820000]  t1 : 5c0accb8 t2 : 00000020 s0 : 0081c84c                       
[  127.830000]  s1 : 00000002 a0 : 5c192088 a1 : b7dddf89                       
[  127.830000]  a2 : 00000001 a3 : 00000001 a4 : b7dddeb4                       
[  127.830000]  a5 : 02c79c63 a6 : 5c1c0f6c a7 : 00000000                       
[  127.830000]  s2 : 009cf9ad s3 : 009962bc s4 : 009e955c                       
[  127.830000]  s5 : 009e84b8 s6 : 00000008 s7 : 00000014                       
[  127.830000]  s8 : 009e7000 s9 : 00000000 s10: 0099f8b4                       
[  127.830000]  s11: ffffffff t3 : ffffffff t4 : b7dddad4                       
[  127.840000]  t5 : 00000040 t6 : 5c021018                                     
[  127.840000] sstatus: 00003008 sbadaddr: 02c79c62 scause: 00000001            
Segmentation fault

s-macke, 2015-07-14:
The crash happens in libc.so in __libc_start_main at 0x17540

```
00017444 <__libc_start_main>:
   17444:       f1010113                addi    sp,sp,-240
   17448:       fffe9717                auipc   a4,0xfffe9
   1744c:       bb870713                addi    a4,a4,-1096 # 0 <GLIBC_2.20>
   17450:       0e812423                sw      s0,232(sp)
   17454:       0e112623                sw      ra,236(sp)
   17458:       0e912223                sw      s1,228(sp)
   1745c:       0f212023                sw      s2,224(sp)
   17460:       0d312e23                sw      s3,220(sp)
   17464:       0d412c23                sw      s4,216(sp)
   17468:       00a12623                sw      a0,12(sp)
   1746c:       00b12423                sw      a1,8(sp)
   17470:       00c12223                sw      a2,4(sp)
   17474:       00068413                mv      s0,a3
   17478:       10070863                beqz    a4,17588 <__libc_start_main+0x144>
   1747c:       00072703                lw      a4,0(a4)
   17480:       00173713                seqz    a4,a4
   17484:       00129697                auipc   a3,0x129
   17488:       b7c68693                addi    a3,a3,-1156 # 140000 <__libc_multiple_libcs>
   1748c:       00e6a023                sw      a4,0(a3)
   17490:       00078a63                beqz    a5,174a4 <__libc_start_main+0x60>
   17494:       00000613                li      a2,0
   17498:       00000593                li      a1,0
   1749c:       00078513                mv      a0,a5
   174a0:       235150ef                jal     2ced4 <__cxa_atexit>
   174a4:       00000797                auipc   a5,0x0
   174a8:       b5c78793                addi    a5,a5,-1188 # 17000 <set_regs+0x2b4>
   174ac:       0007a483                lw      s1,0(a5)
   174b0:       0024f493                andi    s1,s1,2
   174b4:       06049c63                bnez    s1,1752c <__libc_start_main+0xe8>
   174b8:       00040e63                beqz    s0,174d4 <__libc_start_main+0x90>
   174bc:       0012a797                auipc   a5,0x12a
   174c0:       57878793                addi    a5,a5,1400 # 141a34 <__environ>
   174c4:       0007a603                lw      a2,0(a5)
   174c8:       00412583                lw      a1,4(sp)
   174cc:       00812503                lw      a0,8(sp)
   174d0:       000400e7                jalr    s0
   174d4:       00000797                auipc   a5,0x0
   174d8:       b2c78793                addi    a5,a5,-1236 # 17000 <set_regs+0x2b4>
   174dc:       0b47aa03                lw      s4,180(a5)
   174e0:       060a1663                bnez    s4,1754c <__libc_start_main+0x108>
   174e4:       0e049263                bnez    s1,175c8 <__libc_start_main+0x184>
   174e8:       01810513                addi    a0,sp,24
   174ec:       335120ef                jal     2a020 <_setjmp>
   174f0:       0a051063                bnez    a0,17590 <__libc_start_main+0x14c>
   174f4:       be022703                lw      a4,-1056(tp) # fffffbe0 <_end+0xffebc7f0>
   174f8:       0012a697                auipc   a3,0x12a
   174fc:       53c68693                addi    a3,a3,1340 # 141a34 <__environ>
   17500:       0006a603                lw      a2,0(a3)
   17504:       0ce12023                sw      a4,192(sp)
   17508:       bdc22703                lw      a4,-1060(tp) # fffffbdc <_end+0xffebc7ec>
   1750c:       00412583                lw      a1,4(sp)
   17510:       00812503                lw      a0,8(sp)
   17514:       00c12783                lw      a5,12(sp)
   17518:       0ce12223                sw      a4,196(sp)
   1751c:       01810713                addi    a4,sp,24
   17520:       bee22023                sw      a4,-1056(tp) # fffffbe0 <_end+0xffebc7f0>
   17524:       000780e7                jalr    a5
   17528:       718150ef                jal     2cc40 <exit>
   1752c:       00412703                lw      a4,4(sp)
   17530:       0887a783                lw      a5,136(a5)
   17534:       000fb517                auipc   a0,0xfb
   17538:       b5450513                addi    a0,a0,-1196 # 112088 <xdr_zero+0x24>
   1753c:       00072583                lw      a1,0(a4)
   17540:       000780e7                jalr    a5                <------ crash
   17544:       f6041ce3                bnez    s0,174bc <__libc_start_main+0x78>
   17548:       f8dff06f                j       174d4 <__libc_start_main+0x90>
   1754c:       00000717                auipc   a4,0x0
   17550:       ab470713                addi    a4,a4,-1356 # 17000 <set_regs+0x2b4>
   17554:       00072403                lw      s0,0(a4)
   17558:       0b07a983                lw      s3,176(a5)
   1755c:       00000913                li      s2,0
   17560:       25840413                addi    s0,s0,600
   17564:       00c9a783                lw      a5,12(s3)
   17568:       00078663                beqz    a5,17574 <__libc_start_main+0x130>
   1756c:       00040513                mv      a0,s0
   17570:       000780e7                jalr    a5   174fc:       53c68693                addi    a3,a3,1340 # 141a34 <__environ>
```

aswaterman, 2015-07-15:
We haven't debugged dynamic linking on RV32 yet because we didn't have a
working Linux distribution.  Aside from the issue Palmer recently helped
with, what did you have to do to get RV32 Linux working?

s-macke, 2015-07-15:
You can find my current hacks for pk and linux here:
http://pastie.org/10273402
http://pastie.org/10273399
I changed the htif interface to use two registers. But the diffs also contain patches to support Linux 4.1 and add a few debug messages.

You can test yourself by using the link http://jor1k.com/jor1k/demos/riscv.html
Wait until you can see the first output in the terminal. During that time it loads the block device image. Then you choose bbl in the dropdown menu and see the (slow but safe) version of the emulator booting Linux.

Unfortunately someone abandoned the support for mmio devices :) . So I can't give you access to download/upload files to test yourself via virtio/9p.

a0u, 2015-07-17:
This problem is not exclusive to RV32.  See issue #66.

s-macke, 2015-07-17:
Good to know. I am glad, that the problem is reproducible for you and more recent. My test of the 64-Bit toolchain was prior to the patch.

closed 2015-07-17 by aswaterman

aswaterman, 2015-07-17:
Resolved by 0f680c80ddaddaff24b2af85e78fa069d040f5f2

s-macke, 2015-07-17:
This specific problem is solved. However it still doesn't start. It just hangs somewhere forever or I get an unaligned memory access.