riscv-collab/riscv-gnu-toolchain #64: Support for RoCC (Xcustom) by default

seldridge, 2015-07-14:
I recently updated rocket-chip to master, but the RV toolchain it references (b85637e5d20e84c263a134f329e09d081714b983) no longer supports RoCC (Xcustom) instructions (e.g., `custom0`) by default. It seems like there may be a temporary fix in there with d8367667ea2cd6e3de1d07548ee5d4f0f603b5d6 (I'm building HEAD to test this now using `--with-arch=RV64IMAFDXcustom`).

Is there an existing way to enable Xcustom support? Should this be turned on by default? I can't quite tell what the ISA Spec intends regarding the latter question as far as `custom0` and `custom1` are concerned.

palmer-dabbelt, 2015-07-14:
You can pass "-march=RV64IMAFDXcustom" to GCC at compile time to enable
RoCC's custom{0,1}.  You don't need to recompile the toolchain to get this
support, the only change was to the default compiler settings.

The RISC-V spec says that custom{0,1,2,3} are reserved for non-standard ISA
extensions.  RoCC is just what we're using them for at Berkeley.

Since these are non-standard ISA extensions, I would be in favor of leaving
them off by default.  Note that this behavior is compatible with x86's GCC
- new ISA extensions are disabled by default unless a -m switch is passed
(either something like -mavx or -march).

seldridge, 2015-07-15:
This works! Thanks. I was only surprised that the version bump on the rocket-chip toolchain reference (which I do appreciate!) defaulted to something (which I now realize is RV64IMAFD) which no longer includes Xcustom.

After thinking about it, you may be right about whether or not these should be on by default. They are non-standard extensions. Similarly, even though RoCC is standardized and may be used by a large percentage of designs, they're not an "on by default" kind of thing with respect to the default rocket-chip configurations.

closed 2015-07-15 by seldridge