riscv-collab/riscv-gnu-toolchain #67: relocation truncated to fit: R_RISCV_JAL when compiling linux

s-macke, 2015-07-17:
Hi, 

when I compile my kernel (32-Bit toolchain) I get the following error.

```
  LINK    vmlinux
  LD      vmlinux.o
  MODPOST vmlinux.o
  GEN     .version
  CHK     include/generated/compile.h
  UPD     include/generated/compile.h
  CC      init/version.o
  LD      init/built-in.o
  KSYM    .tmp_kallsyms1.o
net/built-in.o: In function `netlink_proto_init':
af_netlink.c:(.init.text+0x924): relocation truncated to fit: R_RISCV_JAL against symbol `rhashtable_init' defined in .text section in lib/built-in.o
Makefile:932: recipe for target 'vmlinux' failed
```

Looks like I hit the 20 bit limit of the instruction. It works when I remove certain features from the kernel. 

I get an similar error "unresolvable R_RISCV_PCREL_HI20 relocation against symbol" when I try to compile DirectFB.

s-macke, 2015-07-18:
The unlinked code contains

```
     924:       00000297                auipc   t0,0x0
     928:       000280e7                jalr    t0
```

So, that should be enough to reach the whole address space.

aswaterman, 2015-07-18:
Sounds like a linker relaxation bug. Can you give me whatever I need to
replicate it (e.g. a tarball of your linux build dir, just before the
failed linking step)?

s-macke, 2015-07-18:
Try jor1k.com/linux.tar.xz and compile with "make ARCH=riscv V=1" The error should pop up after a few seconds. 

I have realized, that the reloc id of objdump is different from the one assumed by ld.

```
riscv32-unknown-elf-objdump -r net/built-in.o
...
00000924 R_RISCV_CALL      proto_register
00000944 R_RISCV_CALL      __kmalloc
0000094c R_RISCV_HI20      nl_table
00000950 R_RISCV_LO12_S    nl_table
00000960 R_RISCV_HI20      .LANCHOR3
0000096c R_RISCV_LO12_I    nl_table
00000970 R_RISCV_LO12_I    .LANCHOR3
00000974 R_RISCV_LO12_I    .LANCHOR3
00000978 R_RISCV_HI20      nl_table
00000980 R_RISCV_CALL      rhashtable_init    <-    assumed to be R_RISCV_JAL by ld and position is assumed to be .init.text+0x924 and not .init.text+0x980 !
....
```

My best guess would be memory corruption, but valgrind does not give an error. So it might be indeed a relaxation problem.

s-macke, 2015-07-18:
I understand. The relaxation removes one instruction changing the offset of the relocation too.

aswaterman, 2015-07-19:
The relaxation code didn't take into account that PC-relative offsets can
actually increase after relaxation, in the case that the caller and callee
are in different sections.  I pushed a hackish fix.  Still not sure about
the DirectFB thing.

s-macke, 2015-07-19:
Thanks. it is fixed. You patch also fixed the DirectFB problem.

jkrh, 2017-09-26:
Looks like I get the same (?) thing when building Zephyr kernel:
irq_offload.c:(.vectors.vinit+0x0): relocation truncated to fit: R_RISCV_JAL against `do_reset'

(gdb) disas _irq_do_offload
Dump of assembler code for function _irq_do_offload:
   0x00000003 <+0>:	auipc	a5,0x0
   0x00000007 <+4>:	lw	a5,0(a5) # 0x3 <_irq_do_offload>
   0x0000000b <+8>:	beqz	a5,0x2b <_irq_do_offload+40>
   0x0000000f <+12>:	auipc	a5,0x0
   0x00000013 <+16>:	lw	t1,0(a5) # 0xf <_irq_do_offload+12>
   0x00000017 <+20>:	auipc	a5,0x0
   0x0000001b <+24>:	sw	zero,0(a5) # 0x17 <_irq_do_offload+20>
   0x0000001f <+28>:	auipc	a5,0x0
   0x00000023 <+32>:	lw	a0,0(a5) # 0x1f <_irq_do_offload+28>
   0x00000027 <+36>:	jr	t1
   0x0000002b <+40>:	ret
End of assembler dump.

Configured with: /home/jmk/work/riscv/riscv-tools/riscv-gnu-toolchain/build/../riscv-gcc/configure --target=riscv32-unknown-elf --prefix=/opt/riscv/toolchain --disable-shared --disable-threads --enable-languages=c,c++ --with-system-zlib --enable-tls --with-newlib --with-headers=/opt/riscv/toolchain/riscv32-unknown-elf/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --enable-checking=yes --disable-multilib --with-abi=ilp32 --with-arch=rv32i 'CFLAGS_FOR_TARGET=-Os  -mcmodel=medlow'
Thread model: single
gcc version 7.1.1 20170509 (GCC)

jkrh, 2017-09-27:
Quick write-up on how to reproduce with Zephyr is here: https://github.com/fractalclone/zephyr-riscv

6.1 compiler is happy, 7.1.n barfs as above.

karimhasebou, 2017-10-10:
@jkrh am facing the same problem, have you been able to fix the error in gcc 7.1?

jkrh, 2017-10-10:
Hi,

I think Andrew is working on it, see
d70ac59256fedc192799acb73513e45550669e91 in
riscv-next. It does seem to fix the case for me.

palmer-dabbelt, 2017-11-16:
Did this ever end up getting fixed?

jkrh, 2017-11-16:
Hi,

I used d70ac59256fedc192799acb73513e45550669e91 in riscv-next and worked
fine since.

palmer-dabbelt, 2017-11-16:
Great!

closed 2017-11-16 by palmer-dabbelt

Athavaloshan, 2019-01-08:
> Hi, I used d70ac59256fedc192799acb73513e45550669e91 in riscv-next and worked fine since.
> […](#)
> -- Janne

could you provide the link where you obtain riscv-next

jim-wilson, 2019-01-08:
riscv-next is a branch that exists in multiple github.com riscv trees, including riscv-linux and riscv-binutils-gdb, but you are responding to a post over a year old, and the riscv-binutils-gdb riscv-next branch is no longer used.  There are a great many ways to get a relocation truncated error, and this issue is probably not related to the problem you hit.  New issues for new problems please.