riscv-collab/riscv-gnu-toolchain #68: Error during multilib build

kkelkar, 2015-07-20:
Hi,
 I am trying to build the toolchain with 32-bit and 64-bit support by following the instructions from README file. However, the build failed with the following error:  
```
make[10]: Nothing to be done for `install-data-am'.
make[10]: Leaving directory `/home/kelkar/tools/riscv-gnu-toolchain/build-gcc-linux-stage2/riscv-unknown-linux-gnu/32/soft-float/no-atomic/libstdc++-v3'
make[9]: Leaving directory `/home/kelkar/tools/riscv-gnu-toolchain/build-gcc-linux-stage2/riscv-unknown-linux-gnu/32/soft-float/no-atomic/libstdc++-v3'
make[8]: Leaving directory `/home/kelkar/tools/riscv-gnu-toolchain/build-gcc-linux-stage2/riscv-unknown-linux-gnu/32/soft-float/no-atomic/libstdc++-v3'
make[7]: Leaving directory `/home/kelkar/tools/riscv-gnu-toolchain/build-gcc-linux-stage2/riscv-unknown-linux-gnu/libstdc++-v3'
make[6]: Leaving directory `/home/kelkar/tools/riscv-gnu-toolchain/build-gcc-linux-stage2/riscv-unknown-linux-gnu/libstdc++-v3'
make[5]: Leaving directory `/home/kelkar/tools/riscv-gnu-toolchain/build-gcc-linux-stage2/riscv-unknown-linux-gnu/libstdc++-v3'
make[4]: Leaving directory `/home/kelkar/tools/riscv-gnu-toolchain/build-gcc-linux-stage2/riscv-unknown-linux-gnu/libstdc++-v3'
make[3]: Leaving directory `/home/kelkar/tools/riscv-gnu-toolchain/build-gcc-linux-stage2'
make[2]: *** [install] Error 2
make[2]: Leaving directory `/home/kelkar/tools/riscv-gnu-toolchain/build-gcc-linux-stage2'
make[1]: *** [stamps/build-gcc-linux-stage2] Error 2
make[1]: Leaving directory `/home/kelkar/tools/riscv-gnu-toolchain'
make: *** [linux] Error 2  
```
Has it something to do with #46? Can anyone provide some insights?

tommythorn, 2015-07-20:
Are you sure there weren't an relevant error message somewhere before this snippet?

kkelkar, 2015-07-21:
I am not sure if there were any relevant messages, but here's a part of the build log just before it broke: [link](https://gist.github.com/kkelkar/7a9c45259b37a2713df7)

Thanks,
Kaustubh

tommythorn, 2015-07-21:
Nope, that's still just the tail of the log.

kkelkar, 2015-07-21:
Here we go, the [complete build output](https://gist.github.com/kkelkar/1c034c52bb6bf06dda11).

tommythorn, 2015-07-21:
Is this from the same run?  The error in this one is very visible:
```
libtool: install: error: cannot install `libgomp-plugin-host_nonshm.la' to a directory not ending in /home/kelkar/tools/opt/riscv/riscv-unknown-linux-gnu/lib/../lib/
```

kkelkar, 2015-07-21:
Well, I had to run the build command again to capture all the output.

palmer-dabbelt, 2015-07-21:
I looked in to this a while ago.  The error comes from

````
riscv-gnu-toolchain/build/build-gcc-linux-stage2/riscv-unknown-linux-gnu/soft-float/libgomp/libtool
          # Don't allow the user to place us outside of our expected
          # location b/c this prevents finding dependent libraries that
          # are installed to the same prefix.
          # At present, this check doesn't affect windows .dll's that
          # are installed into $libdir/../bin (currently, that works fine)
          # but it's something to keep an eye on.
          test "$inst_prefix_dir" = "$destdir" && \  
            func_fatal_error "error: cannot install \`$file' to a directory not ending in $libdir"
````

This kind of feels like a bug to me: shouldn't that be an "||", not a "&&"?  IIRC, when I tracked this down in the libtool sources it looked like it had been there for way to long to not have been noticed, so I gave up.

Does anyone here understand libtool?  I'd love to get this fixed, but I understand none of this.

lluixhi, 2015-09-04:
I found this patch from yoctoproject, and last I built the RISCV GNU Toolchain, it worked.
https://github.com/openembedded/oe-core-do-not-use/blob/master/meta/recipes-devtools/gcc/gcc-5.2/0035-Dont-link-the-plugins-with-libgomp-explicitly.patch

palmer-dabbelt, 2017-03-30:
This should be fixed.

closed 2017-03-30 by palmer-dabbelt