riscv-collab/riscv-gnu-toolchain #72: Linker does not produce raw binary files, even though it says it can.

sam-falvo, 2015-07-30:
Please see my gist here.  https://gist.github.com/sam-falvo/a3aba3db13b382224cfc

After compiling the hw.c program and creating the linker script, nothing I do seems to convince the GNU ld command to produce a real, honest-to-goodness binary file.  It *always* produces an ELF, no matter what.

I'm at a loss on how to proceed from here.  This is either a very subtle user-error on my part, or it's a bug in ld.

Here's the help output for ld:

```
... lots of text elided ...
/opt/riscv/bin/riscv64-unknown-elf-ld: supported targets: elf64-littleriscv elf32-littleriscv elf64-little elf64-big elf32-little elf32-big plugin srec symbolsrec verilog tekhex binary ihex
/opt/riscv/bin/riscv64-unknown-elf-ld: supported emulations: elf64lriscv elf32lriscv
/opt/riscv/bin/riscv64-unknown-elf-ld: emulation specific options:
... lots more text elided ...
```

Any ideas?

samkho, 2016-01-12:
If you're only interested in the .text portion, the following has worked for me.

1.  Build the elf first (e.g. hello5.c -> hello5, where hello5 is in elf format).

2.  Then use objcopy:

		riscv64-unknown-elf-objcopy -O binary hello5 hello5.bin

3.  You can verify with:

		hexdump -C hello5.bin

It may also work with .data and .rodata if you specify where they go.  .bss might need some extra work (perhaps in your runtime code).

palmer-dabbelt, 2017-03-30:
I'm going to close this, as I commented on the gist it looks to me like this matches upstream.

closed 2017-03-30 by palmer-dabbelt