riscv-collab/riscv-gnu-toolchain #79: stimecmp removal

heshamelmatary, 2015-08-19:
Hi,

Before https://github.com/riscv/riscv-gnu-toolchain/commit/b85637e5d20e84c263a134f329e09d081714b983 
stimecmp was recognized. What's the reason of its removal? It breaks the seL4 port build (and the privileged manual). This register is written in seL4 while executing in S-Mode and mtimecmp isn't writeable there. What's the reason of this change? Any suggestions for alternative implementations? 

Thanks,
Hesham

aswaterman, 2015-08-19:
> Before b85637e

This is a change that will show up in v1.8 of the privileged ISA.  Timer
interrupts will be registered via an SBI call, which will ultimately cause
mtimecmp to be set.  See
https://github.com/riscv/riscv-linux/commit/9eb60e47496bcc593ad6a4d12b633a390aa66ee3
for an example.

We're trying to get a new priv spec and SBI spec draft out the door soon.

ssumpf, 2015-08-23:
@aswaterman: Strange question from an outsider, I am currently porting Genode (https://github.com/ssumpf/genode/tree/riscv) to your architecture. Can one program this SBI timer interrupt in a way so that it is only delivered to machine mode when interrupts in supervisor/user mode are enabled, i.e. IE is not zero in 'mstatus' . If not this becomes a problem, at least for our kernel in supervisor mode, because it is non-preemptive, @heshamelmatary SeL4 port faces the same problem. Another question, why is there no MMU support in machine mode? Is it supposed to become some sort of "system management mode" or "secure mode"? What is the rational behind that?

Another suggestion: If you change your ISA from 1.7 to 1.8. Wouldn't it be a good idea to have GIT-branches for that? Let's say have a spike/gcc/fevsrv for each ISA and a development branch. The same suggestion goes for gcc updates, we are using 4.9.2 and it is a lot of hassle to remove the 5.x.x commits.

Please don't get me wrong: This is not an attempt to criticize your very awesome project, but the view of an operating system programmer.

aswaterman, 2015-09-02:
@ssumpf M-mode interrupts are always enabled when running in S-mode, so even if the timer didn't have this property, other interrupts would.  However, from the perspective of S-mode software, no preemption occurs.  That is, if the M-mode software is correct, S-mode software shouldn't be able to determine that an interrupt occurred.

closed 2015-09-02 by aswaterman

ssumpf, 2015-09-02:
@aswaterman: Of course I know that M-mode interrupts are enabled and that M-mode will receive the interrupt. That was the concern in the first place, if the timer was programmed by S-Mode (via SBI) and if the timer interrupt occurs for S-Mode in M-Mode. Let's say S-Mode has interrupts disabled, what should M-Mode do? It must somehow determine that S-Mode has interrupts disabled and post-pone, for example, the `mrts` calls (which in my small experience deliver the trap, even when interrupts are disabled). Since I assume that a kernel shall run in S-Mode this becomes way more complicated than simply programming `stimecmp`, which does the right thing. So, I still don't see the rational behind getting rid of `stimecmp`. Is M-Mode supposed to become some sort of system-mangement mode and be delivered as firmware on a real SOC?

aswaterman, 2015-09-02:
I understand your question now.  See lines 174-186 of https://github.com/riscv/riscv-pk/blob/master/pk/mentry.S to see how I handled this.

The rationale behind getting rid of stimecmp is to reduce HW cost.  The SBI call approach is also more virtualization-friendly (though of course it could be made to work either way).

ssumpf, 2015-09-09:
@aswaterman: Makes sense, thanks again!