riscv-collab/riscv-gnu-toolchain #81: linux gcc

sbeamer, 2015-08-20:
Linux gcc doesn't build within travis, error can be seen at line 9078 https://travis-ci.org/riscv/riscv-tools/builds/76549734#L9078

palmer-dabbelt, 2015-08-21:
On my Ubuntu 12.04.5 container, the latest HEAD from riscv-gnu-toolchain builds fine.  I don't know anything about Travis, do you know how I can duplicate their setup so I can actually reproduce this bug?

In order to get this to work, I had to install a few dependencies, but otherwise it's a clean install of Ubuntu.
 * build-essential
 * git
 * gawk
 * texinfo
 * libgmp-dev
 * libmpfr-dev
 * libmpc-dev

sbeamer, 2015-08-21:
The easiest way has been to just push to the repo (travis will run automatically). Since I have been pushing so often, I have been using branches other than master with the intention of deleting them once this is all cleared up.

Travis is just using a docker container, but I am not sure if we can just get it to experiment. In terms of packages, I believe the current script (.travis.yml) install them all.

palmer-dabbelt, 2015-08-24:
One of the first things that happens is

````
export CXX=g++-4.8 CC=gcc-4.8
````

which will break my glibc build, since this uses the host toolchain to try and compile RISC-V stuff.

sbeamer, 2015-08-24:
Ok, currently trying to same trick we did to fix building riscv-pk within riscv-tools when CC & CXX are set (https://github.com/riscv/riscv-tools/commit/0beaf49b414c8458721e1f9059bbac43794482f2).

sbeamer, 2015-08-24:
It would be great to allow users to set CC and CXX, even if they are ignored by the build process.

palmer-dabbelt, 2016-02-27:
#122 should fix this.

closed 2016-02-27 by palmer-dabbelt