riscv-collab/riscv-gnu-toolchain #87: Broken --with-xlen=32 --with-arch=I

cliffordwolf, 2015-09-12:
The binaries and libraries built with `--with-xlen=32 --with-arch=I` (git 77f7450, current master) do not seem to match the requested configuration, or each other:

Build script:

    git clone https://github.com/riscv/riscv-gnu-toolchain riscv-gnu-toolchain-rv32i
    cd riscv-gnu-toolchain-rv32i/
    mkdir build; cd build
    ../configure --with-xlen=32 --with-arch=I --prefix=/opt/riscv32i
    make -j$(nproc)

Tests:

    $ cat > test.c << EOT
    int main() {
      return 0;
    }
    EOT

    $ riscv32-unknown-elf-gcc -c test.c
    $ riscv32-unknown-elf-objdump -a test.o
    test.o: file format elf64-littleriscv

    $ riscv32-unknown-elf-gcc -m32 -c test.c
    $ riscv32-unknown-elf-objdump -a test.o
    test.o: file format elf32-littleriscv

    $ riscv32-unknown-elf-objdump -a /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/crtend.o
    /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/crtend.o: file format elf64-littleriscv

So far it just looks like the configuration did not "stick". But look it this:

    $ riscv32-unknown-elf-gcc test.c
    collect2: fatal error: ld terminated with signal 11 [Segmentation fault]
    compilation terminated.
    /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/lib/crt0.o: ABI is incompatible with that of the selected emulation
    /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/lib/crt0.o
    /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/crtbegin.o: ABI is incompatible with that of the selected emulation
    /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/crtbegin.o
    /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: /tmp/ccAktjEO.o: ABI is incompatible with that of the selected emulation
    /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /tmp/ccAktjEO.o
    /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/crtend.o: ABI is incompatible with that of the selected emulation
    /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/crtend.o
    /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/lib/libc.a(lib_a-atexit.o): ABI is incompatible with that of the selected emulation
    /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/lib/libc.a(lib_a-atexit.o)
    /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/lib/libc.a(lib_a-exit.o): ABI is incompatible with that of the selected emulation
    /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/lib/libc.a(lib_a-exit.o)
    /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/lib/libc.a(lib_a-fini.o): ABI is incompatible with that of the selected emulation
    /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/lib/libc.a(lib_a-fini.o)
    /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/lib/libc.a(lib_a-impure.o): ABI is incompatible with that of the selected emulation
    /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/lib/libc.a(lib_a-impure.o)
    /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/lib/libc.a(lib_a-init.o): ABI is incompatible with that of the selected emulation
    /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/lib/libc.a(lib_a-init.o)
    /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/lib/libc.a(lib_a-__atexit.o): ABI is incompatible with that of the selected emulation
    /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/lib/libc.a(lib_a-__atexit.o)
    /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/lib/libc.a(lib_a-__call_atexit.o): ABI is incompatible with that of the selected emulation
    /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/lib/libc.a(lib_a-__call_atexit.o)
    /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/lib/libgloss.a(syscalls.o): ABI is incompatible with that of the selected emulation
    /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/lib/libgloss.a(syscalls.o)
    /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/libgcc.a(mul.o): ABI is incompatible with that of the selected emulation
    /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/libgcc.a(mul.o)
    /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/lib/libc.a(lib_a-errno.o): ABI is incompatible with that of the selected emulation
    /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv32i/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/lib/libc.a(lib_a-errno.o)

aswaterman, 2015-09-14:
Commit 572033bd23f2c8e7812103c21a3fb3c33f6b7204 should fix this.

closed 2015-09-14 by aswaterman

cliffordwolf, 2015-09-14:
Thanks! I can confirm that the fix works. I have now updated [my toolchain build instructions](https://github.com/cliffordwolf/picorv32#building-a-pure-rv32i-toolchain) to this git rev.