riscv-collab/riscv-gnu-toolchain #89: signed char gcc bug

jmcddn, 2015-09-23:
Hello, 
I've found what I believe to be a bug in the risc-v gcc. As shown in the code below, I define a singed char, however, it is interpreted as unsigned. 

```c
void chartest(char *x, char *y) {
  if (*x < *y)
    printf("c: %hhd < %hhd\n", *x, *y);
  else
    printf("c: %hhd > %hhd\n", *x, *y);
}

void uchartest(unsigned char *x, unsigned char *y) {
  if (*x < *y)
    printf("uc: %hhu < %hhu\n", *x, *y);
  else
    printf("uc: %hhu > %hhu\n", *x, *y);
}

int main(int argc, char **argv) {
  {
    char a = -1;
    char b = 1;
    chartest(&a, &b);
  }
  {
    unsigned char a = -1;
    unsigned char b = 1;
    uchartest(&a, &b);
  }
  return 0;
}
```

The output of the above code compiled on Ubuntu 14.4 w/ gcc 4.9.2 : 
```
$ ./bugtest-native 
c: -1 < 1
uc: 255 > 1
```
The output when compiled on riscv-gcc and run on spike:
```
$ spike pk bugtest-spike
c: 255 > 1
uc: 255 > 1
```

I looked into the assembly produced by riscv-gcc and see that chartest() and uchartest() both treat the arguments as unsigned.

```
0000000000010140 <chartest>:
   10140:       00054783                lbu     a5,0(a0)
   10144:       0005c603                lbu     a2,0(a1)
   10148:       00078593                mv      a1,a5
   1014c:       00c7e863                bltu    a5,a2,1015c <chartest+0x1c>
   10150:       0001b537                lui     a0,0x1b
   10154:       ca050513                addi    a0,a0,-864 # 1aca0 <__clz_tab+0x110>
   10158:       2ac0006f                j       10404 <printf>
   1015c:       0001b537                lui     a0,0x1b
   10160:       c9050513                addi    a0,a0,-880 # 1ac90 <__clz_tab+0x100>
   10164:       2a00006f                j       10404 <printf>

0000000000010168 <uchartest>:
   10168:       00054783                lbu     a5,0(a0)
   1016c:       0005c603                lbu     a2,0(a1)
   10170:       00078593                mv      a1,a5
   10174:       00c7e863                bltu    a5,a2,10184 <uchartest+0x1c>
   10178:       0001b537                lui     a0,0x1b
   1017c:       cc850513                addi    a0,a0,-824 # 1acc8 <__clz_tab+0x138>
   10180:       2840006f                j       10404 <printf>
   10184:       0001b537                lui     a0,0x1b
   10188:       cb050513                addi    a0,a0,-848 # 1acb0 <__clz_tab+0x120>
   1018c:       2780006f                j       10404 <printf>
```

If the compiler had generated "lb" instructions on lines 10140 and 10144 instead of "lbu" I would expect the correct behavior.

jmcddn, 2015-09-23:
I expanded my test to demonstrate that the bug exists only with signed chars.

```c
#include <stdio.h>

void chartest(char *x, char *y) {
  if (*x < *y)
    printf("c: %hhd < %hhd\n", *x, *y);
  else
    printf("c: %hhd > %hhd\n", *x, *y);
}

void uchartest(unsigned char *x, unsigned char *y) {
  if (*x < *y)
    printf("uc: %hhu < %hhu\n", *x, *y);
  else
    printf("uc: %hhu > %hhu\n", *x, *y);
}

void shorttest(short *x, short *y) {
  if (*x < *y)
    printf("s: %hd < %hd\n", *x, *y);
  else
    printf("s: %hd > %hd\n", *x, *y);
}

void ushorttest(unsigned short *x, unsigned short *y) {
  if (*x < *y)
    printf("us: %hu < %hu\n", *x, *y);
  else
    printf("us: %hu > %hu\n", *x, *y);
}

void inttest(int *x, int *y) {
  if (*x < *y)
    printf("i: %d < %d\n", *x, *y);
  else
    printf("i: %d > %d\n", *x, *y);
}
void uinttest(unsigned int *x, unsigned int *y) {
  if (*x < *y)
    printf("ui: %u < %u\n", *x, *y);
  else
    printf("ui: %u > %u\n", *x, *y);
}

void lltest(long long int *x, long long int *y) {
  if (*x < *y)
    printf("ll: %lld < %lld\n", *x, *y);
  else
    printf("ll: %lld > %lld\n", *x, *y);
}

void ulltest(unsigned long long int *x, unsigned long long int *y) {
  if (*x < *y)
    printf("ull: %llu < %llu\n", *x, *y);
  else
    printf("ull: %llu > %llu\n", *x, *y);
}
int main(int argc, char **argv) {
  {
    char a = -1;
    char b = 1;
    chartest(&a, &b);
  }
  {
    unsigned char a = -1;
    unsigned char b = 1;
    uchartest(&a, &b);
  }
  {
    short a = -1;
    short b = 1;
    shorttest(&a, &b);
  }
  {
    unsigned short a = -1;
    unsigned short b = 1;
    ushorttest(&a, &b);
  }
  {
    int a = -1;
    int b = 1;
    inttest(&a, &b);
  }
  {
    unsigned int a = -1;
    unsigned int b = 1;
    uinttest(&a, &b);
  }
  {
    long long int a = -1;
    long long int b = 1;
    lltest(&a, &b);
  }
  {
    unsigned long long int a = -1;
    unsigned long long int b = 1;
    ulltest(&a, &b);
  }
  return 0;
}
```

The output of the above code compiled on Ubuntu 14.4 w/ gcc 4.9.2 :
```
$ ./bugtest-native 
c: -1 < 1
uc: 255 > 1
s: -1 < 1
us: 65535 > 1
i: -1 < 1
ui: 4294967295 > 1
ll: -1 < 1
ull: 18446744073709551615 > 1
```
The output when compiled w/ riscv-gcc and run on spike:
```
$ spike pk bugtest-spike
c: 255 > 1
uc: 255 > 1
s: -1 < 1
us: 65535 > 1
i: -1 < 1
ui: 4294967295 > 1
ll: -1 < 1
ull: 18446744073709551615 > 1
```

palmer-dabbelt, 2015-09-28:
C does not define if "char" is signed or unsigned:

http://www.open-std.org/JTC1/SC22/WG14/www/docs/n1256.pdf
"
The three types char, signed char, and unsigned char are collectively called
the character types. The implementation shall define char to have the same range,
representation, and behavior as either signed char or unsigned char.
"

Our GCC port looks internally consistent to me:

````
$ riscv64-unknown-linux-gnu-gcc -dM -E -c /usr/riscv64-unknown-linux-gnu/usr/include/limits.h | grep CHAR_MIN
#define SCHAR_MIN (-SCHAR_MAX - 1)
#define __WCHAR_MIN__ (-__WCHAR_MAX__ - 1)
#define CHAR_MIN 0
````

````
$ cat -n riscv-gnu-toolchain/gcc/gcc/config/riscv/riscv.h  | grep DEFAULT_SIGNED_CHAR
   859	#define DEFAULT_SIGNED_CHAR 0
````

Unless I've missed something and it's explicitly stated somewhere that "char" is signed on RISC-V, I think this isn't actually a bug.

jmcddn, 2015-10-01:
Yes, you're absolutely right. Looks like my head was stuck in x86_64!

closed 2015-10-01 by jmcddn