riscv-collab/riscv-gnu-toolchain #92: Unnecessary instructions generated for 16/8 bit integer calculations

svenstucki, 2015-09-27:
Consider this code:

    int16_t i16_ret(const int16_t *arr, unsigned int arr_len) {
      int16_t sum = 0;
      for (unsigned int i = 0; i < arr_len; i++) {
        sum += arr[i];
      }
      return sum;
    }

The current riscv-gcc with -O3 generates the following:

    00000000 <i16_ret>:
       0: 02058863            beqz  a1,30 <.L4>
       4: 00159593            slli  a1,a1,0x1
       8: 00050793            mv  a5,a0
       c: 00b505b3            add a1,a0,a1
      10: 00000513            li  a0,0

    00000014 <.L3>:
      14: 0007d703            lhu a4,0(a5)
      18: 00278793            addi  a5,a5,2
      1c: 00e50533            add a0,a0,a4
      20: 01051513            slli  a0,a0,0x10
      24: 41055513            srai  a0,a0,0x10
      28: fef596e3            bne a1,a5,14 <.L3>
      2c: 00008067            ret

    00000030 <.L4>:
      30: 00000513            li  a0,0
      34: 00008067            ret

Note the slli/srai pair before the branch. It looks like the compiler is trying to truncate the upper 16 bit of the result or sign-extend it. It's not too bad in that case, but being part of the loop hurts performance

.

However the compiler does the same even in cases where it wouldn't be needed:

    void i16_mem(const int16_t *arr, unsigned int arr_len, int16_t *out) {
      int16_t sum = 0;
      for (unsigned int i = 0; i < arr_len; i++) {
        sum += arr[i];
      }
      *out = sum;
    }
 
Compiles to (-O3 again):

    00000038 <i16_mem>:
      38: 02058863            beqz  a1,68 <.L10>
      3c: 00159593            slli  a1,a1,0x1
      40: 00b505b3            add a1,a0,a1
      44: 00000793            li  a5,0

    00000048 <.L9>:
      48: 00055703            lhu a4,0(a0)
      4c: 00250513            addi  a0,a0,2
      50: 00e787b3            add a5,a5,a4
      54: 01079793            slli  a5,a5,0x10
      58: 4107d793            srai  a5,a5,0x10
      5c: fea596e3            bne a1,a0,48 <.L9>

    00000060 <.L8>:
      60: 00f61023            sh  a5,0(a2)
      64: 00008067            ret

    00000068 <.L10>:
      68: 00000793            li  a5,0
      6c: ff5ff06f            j 60 <.L8>

Again there's a slli/srai pair. Since the result is stored using a sh, it is not needed though.

When using unsigned ints, the compiler generates a slli/srli pair, but only with -O3. Using -O0 unsigned ints work fine, signed ints still have the problem though.

I also tried with 8 bit integers and got similar results, see below. The problem persists since quite some time, the dumps are from a recent compiler version though (https://github.com/riscv/riscv-gnu-toolchain/commit/781d50af05dc75522e799c962489ff4e4723240a). Compilers for other targets can optimize this btw., for example or1k llvm/gcc doesn't have this problem.

    #include <stdint.h>

    // O0:  slli/srli lh/lhu/sh
    // O3:  slli/srai lhu
    int16_t i16_ret(const int16_t *arr, unsigned int arr_len) {
      int16_t sum = 0;
      for (unsigned int i = 0; i < arr_len; i++) {
        sum += arr[i];
      }
      return sum;
    }

    // O0:  slli/srli lh/lhu/sh
    // O3:  slli/srai lhu/sh
    void i16_mem(const int16_t *arr, unsigned int arr_len, int16_t *out) {
      int16_t sum = 0;
      for (unsigned int i = 0; i < arr_len; i++) {
        sum += arr[i];
      }
      *out = sum;
    }

    // O0:  OK        lhu/lhu/sh
    // O3:  slli/srli lhu
    uint16_t u16_ret(const uint16_t *arr, unsigned int arr_len) {
      uint16_t sum = 0;
      for (unsigned int i = 0; i < arr_len; i++) {
        sum += arr[i];
      }
      return sum;
    }

    // O0:  OK        lhu/lhu/sh
    // O3:  slli/srli lhu/sh
    void u16_mem(const uint16_t *arr, unsigned int arr_len, uint16_t *out) {
      uint16_t sum = 0;
      for (unsigned int i = 0; i < arr_len; i++) {
        sum += arr[i];
      }
      *out = sum;
    }

    // O0:  OK        lb/lbu/sb
    // O3:  slli/srai lbu
    int8_t i8_ret(const int8_t *arr, unsigned int arr_len) {
      int8_t sum = 0;
      for (unsigned int i = 0; i < arr_len; i++) {
        sum += arr[i];
      }
      return sum;
    }

    // O0:  OK        lbu/lbu/sb
    // O3:  OK        lw/lbu      (heavily unrolled, 8 bit additions vectorized)
    uint8_t u8_ret(const uint8_t *arr, unsigned int arr_len) {
      uint8_t sum = 0;
      for (unsigned int i = 0; i < arr_len; i++) {
        sum += arr[i];
      }
      return sum;
    }

aswaterman, 2015-09-27:
> compiler version though (781d50a

In both cases, it would be best if the summation were computed using the
entire register's width, then truncated at the end. In the case of the
store, of course, the truncation is free.

I don't know why gcc insists on using a HImode register for the entire
thing. Maybe something in the cost model is messed up.

aswaterman, 2015-09-27:
FWIW, the ARMv8 GCC port produces similarly suboptimal code.  This leads me
to believe the culprit is in the middle-end of the compiler.

svenstucki, 2015-09-28:
Thanks for looking into it. Indeed, I was mistaken. Both of our or1k gcc have the same problem - but llvm is smart enough to only do the sign-extension/truncation at the end of the loop or not at all for stores.

Seems like it's not directly a RISC-V problem then. It adds quite a bit of overhead in small loops though (compared to llvm generated code).

aswaterman, 2015-09-28:
My guess is that the main reason it hasn't been fixed is that it doesn't
matter much on x86, which allows the lower 1/2 bytes of the Integrr
registers to be addressed directly.

closed 2015-09-28 by aswaterman