riscv-collab/riscv-gnu-toolchain #96: R_RISCV_COPY relocs and functions

bukinr, 2015-10-26:
It looks like gcc or ld is no idea about type of some symbols (see linker warnings below) and it emits R_RISCV_COPY relocations for it. But the symbols are functions (!). We don't expect that functions body can be relocated.

$ pwd
/home/rb743/obj/riscv.riscv/home/rb743/dev/freebsd-riscv/bin/ls
$ 
$ /home/rb743/opt/sptbr/bin/riscv64-unknown-freebsd11.0-gcc -isystem /home/rb743/obj//riscv.riscv/home/rb743/dev/freebsd-riscv/tmp/usr/include -L/home/rb743/obj//riscv.riscv/home/rb743/dev/freebsd-riscv/tmp/usr/lib --sysroot=/home/rb743/obj//riscv.riscv/home/rb743/dev/freebsd-riscv/tmp  -O2 -pipe   -DCOLORLS  -std=gnu99 -fstack-protector-strong -Wsystem-headers -Werror -Wall -Wno-format-y2k -W -Wno-unused-parameter -Wstrict-prototypes -Wmissing-prototypes -Wpointer-arith -Wreturn-type -Wcast-qual -Wwrite-strings -Wswitch -Wshadow -Wunused-parameter -Wcast-align -Wchar-subscripts -Winline -Wnested-externs -Wredundant-decls -Wold-style-definition -Wno-pointer-sign -Wno-error=unused-function -Wno-error=enum-compare -Wno-error=logical-not-parentheses -Wno-error=bool-compare -Wno-error=uninitialized -Wno-error=array-bounds -Wno-error=clobbered -Wno-error=cast-align -Wno-error=extra -Wno-error=attributes -Wno-error=inline -Wno-error=unused-but-set-variable -Wno-error=unused-value -Wno-error=strict-aliasing -Wno-error=address    -o ls cmp.o ls.o print.o util.o  -lxo  -lutil  -ltermcapw
/home/rb743/opt/sptbr/lib/gcc/riscv64-unknown-freebsd11.0/5.2.0/../../../../riscv64-unknown-freebsd11.0/bin/ld: warning: type and size of dynamic symbol `_kill@@FBSDprivate_1.0' are not defined
/home/rb743/opt/sptbr/lib/gcc/riscv64-unknown-freebsd11.0/5.2.0/../../../../riscv64-unknown-freebsd11.0/bin/ld: warning: type and size of dynamic symbol `_getpid@@FBSDprivate_1.0' are not defined
/home/rb743/opt/sptbr/lib/gcc/riscv64-unknown-freebsd11.0/5.2.0/../../../../riscv64-unknown-freebsd11.0/bin/ld: warning: type and size of dynamic symbol `_getuid@@FBSDprivate_1.0' are not defined
/home/rb743/opt/sptbr/lib/gcc/riscv64-unknown-freebsd11.0/5.2.0/../../../../riscv64-unknown-freebsd11.0/bin/ld: warning: type and size of dynamic symbol `_readlink@@FBSDprivate_1.0' are not defined
/home/rb743/opt/sptbr/lib/gcc/riscv64-unknown-freebsd11.0/5.2.0/../../../../riscv64-unknown-freebsd11.0/bin/ld: warning: type and size of dynamic symbol `ioctl@@FBSD_1.0' are not defined
/home/rb743/opt/sptbr/lib/gcc/riscv64-unknown-freebsd11.0/5.2.0/../../../../riscv64-unknown-freebsd11.0/bin/ld: warning: type and size of dynamic symbol `kill@@FBSD_1.0' are not defined
/home/rb743/opt/sptbr/lib/gcc/riscv64-unknown-freebsd11.0/5.2.0/../../../../riscv64-unknown-freebsd11.0/bin/ld: warning: type and size of dynamic symbol `lpathconf@@FBSD_1.1' are not defined
/home/rb743/opt/sptbr/lib/gcc/riscv64-unknown-freebsd11.0/5.2.0/../../../../riscv64-unknown-freebsd11.0/bin/ld: warning: type and size of dynamic symbol `getuid@@FBSD_1.0' are not defined
/home/rb743/opt/sptbr/lib/gcc/riscv64-unknown-freebsd11.0/5.2.0/../../../../riscv64-unknown-freebsd11.0/bin/ld: warning: type and size of dynamic symbol `readlink@@FBSD_1.0' are not defined
/home/rb743/opt/sptbr/lib/gcc/riscv64-unknown-freebsd11.0/5.2.0/../../../../riscv64-unknown-freebsd11.0/bin/ld: warning: type and size of dynamic symbol `_ioctl@@FBSDprivate_1.0' are not defined
/home/rb743/opt/sptbr/lib/gcc/riscv64-unknown-freebsd11.0/5.2.0/../../../../riscv64-unknown-freebsd11.0/bin/ld: warning: type and size of dynamic symbol `getpid@@FBSD_1.0' are not defined
$ 
$ 
$ /home/rb743/opt/sptbr/bin/riscv64-unknown-freebsd11.0-gcc -isystem /home/rb743/obj//riscv.riscv/home/rb743/dev/freebsd-riscv/tmp/usr/include -L/home/rb743/obj//riscv.riscv/home/rb743/dev/freebsd-riscv/tmp/usr/lib --sysroot=/home/rb743/obj//riscv.riscv/home/rb743/dev/freebsd-riscv/tmp  -O2 -pipe   -DCOLORLS  -std=gnu99 -fstack-protector-strong -Wsystem-headers -Werror -Wall -Wno-format-y2k -W -Wno-unused-parameter -Wstrict-prototypes -Wmissing-prototypes -Wpointer-arith -Wreturn-type -Wcast-qual -Wwrite-strings -Wswitch -Wshadow -Wunused-parameter -Wcast-align -Wchar-subscripts -Winline -Wnested-externs -Wredundant-decls -Wold-style-definition -Wno-pointer-sign -Wno-error=unused-function -Wno-error=enum-compare -Wno-error=logical-not-parentheses -Wno-error=bool-compare -Wno-error=uninitialized -Wno-error=array-bounds -Wno-error=clobbered -Wno-error=cast-align -Wno-error=extra -Wno-error=attributes -Wno-error=inline -Wno-error=unused-but-set-variable -Wno-error=unused-value -Wno-error=strict-aliasing -Wno-error=address    -o ls cmp.o ls.o print.o util.o  -lxo  -lutil  -ltermcapw -###
Using built-in specs.
COLLECT_GCC=/home/rb743/opt/sptbr/bin/riscv64-unknown-freebsd11.0-gcc
COLLECT_LTO_WRAPPER=/home/rb743/opt/sptbr/libexec/gcc/riscv64-unknown-freebsd11.0/5.2.0/lto-wrapper
Target: riscv64-unknown-freebsd11.0
Configured with: /home/rb743/dev/freebsd-riscv-gnu-toolchain-sptbr/src/gcc/configure --target=riscv64-unknown-freebsd11.0 --prefix=/home/rb743/opt/sptbr --with-sysroot=/home/rb743/opt/sptbr/sysroot --without-newlib --without-headers --disable-shared --disable-threads --enable-tls --enable-languages=c,c++ --with-gmp=/usr/local --with-mpfr=/usr/local --with-mpc=/usr/local --disable-libatomic --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-multilib --disable-bootstrap
Thread model: single
gcc version 5.2.0 (GCC) 
COMPILER_PATH=/home/rb743/opt/sptbr/libexec/gcc/riscv64-unknown-freebsd11.0/5.2.0/:/home/rb743/opt/sptbr/libexec/gcc/riscv64-unknown-freebsd11.0/5.2.0/:/home/rb743/opt/sptbr/libexec/gcc/riscv64-unknown-freebsd11.0/:/home/rb743/opt/sptbr/lib/gcc/riscv64-unknown-freebsd11.0/5.2.0/:/home/rb743/opt/sptbr/lib/gcc/riscv64-unknown-freebsd11.0/:/home/rb743/opt/sptbr/lib/gcc/riscv64-unknown-freebsd11.0/5.2.0/../../../../riscv64-unknown-freebsd11.0/bin/
LIBRARY_PATH=/home/rb743/opt/sptbr/lib/gcc/riscv64-unknown-freebsd11.0/5.2.0/:/home/rb743/opt/sptbr/lib/gcc/riscv64-unknown-freebsd11.0/5.2.0/../../../../riscv64-unknown-freebsd11.0/lib/../lib/:/home/rb743/obj//riscv.riscv/home/rb743/dev/freebsd-riscv/tmp/lib/../lib/:/home/rb743/obj//riscv.riscv/home/rb743/dev/freebsd-riscv/tmp/usr/lib/../lib/:/home/rb743/opt/sptbr/lib/gcc/riscv64-unknown-freebsd11.0/5.2.0/../../../../riscv64-unknown-freebsd11.0/lib/:/home/rb743/obj//riscv.riscv/home/rb743/dev/freebsd-riscv/tmp/lib/:/home/rb743/obj//riscv.riscv/home/rb743/dev/freebsd-riscv/tmp/usr/lib/
COLLECT_GCC_OPTIONS='-isystem' '/home/rb743/obj//riscv.riscv/home/rb743/dev/freebsd-riscv/tmp/usr/include' '-L/home/rb743/obj//riscv.riscv/home/rb743/dev/freebsd-riscv/tmp/usr/lib' '-O2' '-pipe' '-D' 'COLORLS' '-std=gnu99' '-fstack-protector-strong' '-Wsystem-headers' '-Werror' '-Wall' '-Wno-format-y2k' '-Wextra' '-Wstrict-prototypes' '-Wmissing-prototypes' '-Wpointer-arith' '-Wreturn-type' '-Wcast-qual' '-Wwrite-strings' '-Wswitch' '-Wshadow' '-Wunused-parameter' '-Wcast-align' '-Wchar-subscripts' '-Winline' '-Wnested-externs' '-Wredundant-decls' '-Wold-style-definition' '-Wno-pointer-sign' '-Wno-error=unused-function' '-Wno-error=enum-compare' '-Wno-error=logical-not-parentheses' '-Wno-error=bool-compare' '-Wno-error=uninitialized' '-Wno-error=array-bounds' '-Wno-error=clobbered' '-Wno-error=cast-align' '-Wno-error=extra' '-Wno-error=attributes' '-Wno-error=inline' '-Wno-error=unused-but-set-variable' '-Wno-error=unused-value' '-Wno-error=strict-aliasing' '-Wno-error=address' '-o' 'ls'
 /home/rb743/opt/sptbr/libexec/gcc/riscv64-unknown-freebsd11.0/5.2.0/collect2 -plugin /home/rb743/opt/sptbr/libexec/gcc/riscv64-unknown-freebsd11.0/5.2.0/liblto_plugin.so "-plugin-opt=/home/rb743/opt/sptbr/libexec/gcc/riscv64-unknown-freebsd11.0/5.2.0/lto-wrapper" "-plugin-opt=-fresolution=/tmp//ccK9vr17.res" "-plugin-opt=-pass-through=-lgcc" "-plugin-opt=-pass-through=-lc" "-plugin-opt=-pass-through=-lgcc" "--sysroot=/home/rb743/obj//riscv.riscv/home/rb743/dev/freebsd-riscv/tmp" --eh-frame-hdr -dynamic-linker /libexec/ld-elf.so.1 -melf64lriscv -o ls /home/rb743/obj//riscv.riscv/home/rb743/dev/freebsd-riscv/tmp/usr/lib/../lib/crt1.o /home/rb743/obj//riscv.riscv/home/rb743/dev/freebsd-riscv/tmp/usr/lib/../lib/crti.o /home/rb743/obj//riscv.riscv/home/rb743/dev/freebsd-riscv/tmp/usr/lib/../lib/crtbegin.o -L/home/rb743/obj//riscv.riscv/home/rb743/dev/freebsd-riscv/tmp/usr/lib -L/home/rb743/opt/sptbr/lib/gcc/riscv64-unknown-freebsd11.0/5.2.0 -L/home/rb743/opt/sptbr/lib/gcc/riscv64-unknown-freebsd11.0/5.2.0/../../../../riscv64-unknown-fr [... cut ...]
 Addr: 0x0000000000801348  Offset: 0x001348  Link: 5 (.dynstr)
  000000: Version: 1  File: libc.so.7  Cnt: 4
  0x0010:   Name: FBSD_1.3  Flags: none  Version: 5
  0x0020:   Name: FBSDprivate_1.0  Flags: none  Version: 4
  0x0030:   Name: FBSD_1.0  Flags: none  Version: 3
  0x0040:   Name: FBSD_1.1  Flags: none  Version: 2

Displaying notes found at file offset 0x000001e0 with length 0x00000030:
  Owner                 Data size	Description
  FreeBSD              0x00000004	NT_ARCH (architecture)
  FreeBSD              0x00000004	NT_VERSION (version)
$

aswaterman, 2015-10-26:
It is interesting that all of the examples are against syscalls, i.e.,
functions that are likely written in assembly.  This doesn't happen for
strlen, for example.  I suspect the assembly code for the syscalls is to
blame.

bukinr, 2015-10-26:
yeah, I noticed.

I think this should help:
.type   symbol,@function;

I just found it in yours glibc implementation

bukinr, 2015-10-26:
yeah, that helps!
thanks

closed 2015-10-26 by bukinr

aswaterman, 2015-10-26:
Glad to have helped.