riscv-collab/riscv-gnu-toolchain #98: No libgcc.a for -m32 -march=RV32I

cliffordwolf, 2015-10-30:
I used to be able to create RV32I binaries that use symbols such as `__mulsi3` from libgcc like this:

    riscv64-unknown-elf-gcc -c -m32 -march=RV32I -ffreestanding -nostdlib test.c
    riscv64-unknown-elf-gcc -m32 -march=RV32I -ffreestanding -nostdlib -o test.elf test.o -lgcc

Here is an example test.c that want to use `__mulsi3`:

    #include <stdint.h>
    
    void _start(uint32_t a, uint32_t b)
    {
        uint64_t au = a, bu = b;
        int64_t as = (int32_t)a, bs = (int32_t)b;
        uint32_t s_mul, s_mulh, s_mulhsu, s_mulhu;
    
        s_mul = a * b;
        s_mulh = (as * bs) >> 32;
        s_mulhsu = (as * bu) >> 32;
        s_mulhu = (au * bu) >> 32;
    
        if (s_mul == s_mulh || s_mulhsu == s_mulhu)
            __asm__ volatile ("sbreak");
    }

With git head of riscv-gnu-toolchain (1ad4bfc) I can't do that anymore and get the following error:

    $ riscv64-unknown-elf-gcc -m32 -march=RV32I -ffreestanding -nostdlib -o test.elf test.o -lgcc
    /opt/riscv/lib/gcc/riscv64-unknown-elf/5.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv/lib/gcc/riscv64-unknown-elf/5.2.0/libgcc.a(mul.o): ABI is incompatible with that of the selected emulation
    /opt/riscv/lib/gcc/riscv64-unknown-elf/5.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv/lib/gcc/riscv64-unknown-elf/5.2.0/libgcc.a(mul.o)
    test.o: In function `_start':
    test.c:(.text+0x70): undefined reference to `__mulsi3'
    collect2: error: ld returned 1 exit status

There used to be a `/opt/riscv/lib/gcc/riscv64-unknown-elf/5.2.0/32/libgcc.a`, but it's not there anymore. Do I need to build multilib support for this, even when I don't use any libraries but just want full C language support for integer ops? This seems a little bit odd to me..

aswaterman, 2015-10-30:
@palmer-dabbelt any idea if this is the result of the with-arch stuff you did a while ago?

heshamelmatary, 2015-12-09:
Any updates regarding this issue? 

I am also trying to build 32-bit version of gcc-newlib, but I am getting some error when compiling gcc directly from the src subdirectory (not from the riscv-gnu-toolchain top-level):

"../../../../../newlib/libc/stdlib/dtoa.c: In function '_dtoa_r':
../../../../../newlib/libc/stdlib/dtoa.c:862:1: error: insn does not satisfy its constraints:
 }
 ^
(insn 69 68 215 2 (set (reg/v:DF 28 t3 [orig:396 _d ] [396])
        (reg:DF 44 fa2 [ _d ])) ../../../../../newlib/libc/stdlib/dtoa.c:189 142 {*movdf_hardfloat_rv32}
     (nil))
../../../../../newlib/libc/stdlib/dtoa.c:862:1: internal compiler error: in extract_constrain_insn, at recog.c:2246"

aswaterman, 2015-12-09:
Turns out these are separate issues.  @cliffordwolf's results from the fact that multilibs is no longer the default.  I was able to build his example by configuring a compiler with --with-xlen=32 --with-arch=I, and I expect --enable-multilib would also do the trick.

@heshamelmatary: yours is a bug.  I'm working on it.

heshamelmatary, 2015-12-09:
@aswaterman Thanks. I am adding support to build riscv*-rtems* toolchain; binutils and gcc build fine for riscv64, but I am having this error when trying to build for riscv32. Please ping me when it's fixed.

heshamelmatary, 2015-12-10:
Building --with-arch=IMA and --with-xlen=32 goes fine BTW, thanks for the hint.

closed 2015-12-10 by aswaterman

AnttiLukats, 2018-10-18:
the issues is still (or again) there

jim-wilson, 2018-10-18:
@AnttiLukats which issue is still there?  If you are asking about the 32-bit libgcc for a 64-bit target, then you must use --enable-multilibs when configuring to get it.  Multilib is not the default.  Or alternatively configure a compiler for the arch/abi that you want.

martinKindall, 2023-02-10:
This worked for me, had to manually add the path to the libgcc.a:
`riscv32-unknown-elf-ld -o exampleProg2.elf -T bram.ld -m elf32lriscv -nostdlib --no-relax /(RVTOOLCHAIN_GCC_LIB_DIR)/libgcc.a`

Just be careful of choosing the right libgcc.a, there may be multiple in your installation, depending of your multilib seutp.