riscv-collab/riscv-gnu-toolchain #102: (bad codegen): Extra shifting with array indexing on RV64

ccelio, 2015-11-17:
This is a common bad-code generation for RV64, when using an "unsigned int" as an index into an array. This problem does not show up with a signed int index.

````
int test32ArrayAccess(uint32_t* arr, unsigned int idx) {  
   return arr[idx];
}
````
Turns into:
````
slli  a1,a1,0x20  
srli  a1,a1,0x20  
slli  a1,a1,0x2   
add   a0,a0,a1    
lw a0,0(a0)       
ret         
````     

At a minimum, the last two shifts should be combined into a single shift.

aswaterman, 2015-11-17:
I'll see if we can peephole this into just two shifts.  Why can't people
just use size_t :-(

aswaterman, 2015-11-17:
I wrote a peephole optimizer for this case, but it only succeeds some of
the time (e.g. it works for your example, but not some others).  This needs
to be fixed earlier on in the compiler and I don't have the time to figure
out how to do that.

On Mon, Nov 16, 2015 at 5:19 PM, Andrew Waterman <waterman@eecs.berkeley.edu

ccelio, 2015-11-17:
Why does the "signed int" version not require clearing the top 32 bits? Is the compiler able to trust that the top 32-bits will all be either 0's or 1's? 

````
slli  a1,a1,0x2
add   a0,a0,a1
lw a0,0(a0)
ret
````

tommythorn, 2015-11-17:
It's the ABI. 32-bit signed ints are ALWAYS sign extended into sign extended 64-bits. It's works well in general (Alpha did it too IIRC), but it's not the only choice.

The real tragedy is that K&R intended int to be the natural size (here, 64-bit), but too much sw would break if we did that.

(Soapbox: C should never had added "unsigned", it's the source of so many problems & bugs.)  

Tommy

jim-wilson, 2018-01-25:
Current FSF gcc sources generate
test32ArrayAccess:
	slli	a1,a1,32
	srli	a1,a1,30
	add	a1,a0,a1
	lw	a0,0(a1)
	ret

This was fixed by
    https://gcc.gnu.org/ml/gcc-patches/2017-11/msg02522.html

Andrew's patch handled 1 case, I extended it to handle 3 cases.  There are still some less common cases we don't handle well, so more work can be done here.  But the most common cases should be fixed.

To fix this earlier in the compiler, you need to split apart the sign/zero extend operations into shifts earlier, and if you do that, then you lose the ability to optimize away unnecessary sign/zero extensions.  This is worth trying at some point, but it isn't obvious if it will make things better or worse.

closed 2018-01-25 by jim-wilson