riscv-collab/riscv-gnu-toolchain #103: Relocation truncated to fit: R_RISCV_HI20 despite of XLEN

aka-sps, 2015-11-18:
# Synopsis
References to symbols with address higher than 2GiB are unavailable: linker fails with message `relocation truncated to fit: R_RISCV_HI20` against symbols with value above or equal to `0x7ffff800` despite of `XLEN=32` or `XLEN=64`.

# Simple test
### relocation_test.c
```C
/* no access to void variable, only address taking
skip warning: taking address of expression of type ‘void’ */
extern void feedbeef;
int main(int argc, char* argv[]){
  return (unsigned long)&feedbeef != 0x7ffff800UL;
}
```
### abs_def.S
```asm
.globl feedbeef
.set feedbeef, 0x7ffff800
```
## Intermediate files
### relocation_test.s
```
	.file	"relocation_test.c"
	.section	.text.startup,"ax",@progbits
	.align	2
	.globl	main
	.type	main, @function
main:
	li	a5,-2147483648
	lui	a4,%hi(feedbeef)
	xor	a5,a5,-2048
	add	a4,a4,%lo(feedbeef)
	li	a0,1
	beq	a4,a5,.L4
	ret
.L4:
	li	a0,0
	ret
	.size	main, .-main
	.ident	"GCC: (GNU) 5.2.0"
```
## objdump
```
relocation_test.o:     file format elf32-littleriscv

SYMBOL TABLE:
00000000 l    df *ABS*	00000000 relocation_test.c
00000000 l    d  .text	00000000 .text
00000000 l    d  .data	00000000 .data
00000000 l    d  .bss	00000000 .bss
00000000 l    d  .text.startup	00000000 .text.startup
0000001c l       .text.startup	00000000 .L4
00000000 l    d  .comment	00000000 .comment
00000000 g     F .text.startup	00000024 main
00000000         *UND*	00000000 feedbeef

Disassembly of section .text.startup:

00000000 <main>:
   0:	800007b7          	lui	a5,0x80000
   4:	00000737          	lui	a4,0x0
			4: R_RISCV_HI20	feedbeef
   8:	8007c793          	xori	a5,a5,-2048
   c:	00070713          	mv	a4,a4
			c: R_RISCV_LO12_I	feedbeef
  10:	00100513          	li	a0,1
  14:	00f70463          	beq	a4,a5,1c <.L4>
			14: R_RISCV_BRANCH	.L4
  18:	00008067          	ret

0000001c <.L4>:
  1c:	00000513          	li	a0,0
  20:	00008067          	ret
```
```
abs_def.o:     file format elf32-littleriscv

SYMBOL TABLE:
00000000 l    d  .text	00000000 .text
00000000 l    d  .data	00000000 .data
00000000 l    d  .bss	00000000 .bss
7ffff800 g       *ABS*	00000000 feedbeef

```
## ld error
```
relocation_test.o: In function `main':
relocation_test.c:(.text.startup+0x4): relocation truncated to fit: R_RISCV_HI20 against symbol `feedbeef' defined in *ABS* section in a.out
```
# Root cause
Expression `VALID_UTYPE_IMM (RISCV_CONST_HIGH_PART (value))` is **false** if `value >= 0x7ffff800U` in binutils/bfd/elfnn-riscv.c@11c69401b1225965d9929f0ba4018deaca5ea0b9 (1485)

```c++
/* Emplace a static relocation.  */

static bfd_reloc_status_type
perform_relocation (const reloc_howto_type *howto,
		    const Elf_Internal_Rela *rel,
		    bfd_vma value,
		    asection *input_section,
		    bfd *input_bfd,
		    bfd_byte *contents)
{
  ...
  switch (ELFNN_R_TYPE (rel->r_info))
    {
    case R_RISCV_HI20:
    case R_RISCV_TPREL_HI20:
    case R_RISCV_PCREL_HI20:
    case R_RISCV_GOT_HI20:
    case R_RISCV_TLS_GOT_HI20:
    case R_RISCV_TLS_GD_HI20:
      if (!VALID_UTYPE_IMM (RISCV_CONST_HIGH_PART (value)))
        return bfd_reloc_overflow;
      value = ENCODE_UTYPE_IMM (RISCV_CONST_HIGH_PART (value));
      break;
    ...
   }
  ...
}
```
Need to check similar cases.

aswaterman, 2015-11-18:
This is the intended behavior for RV64, where the default code model (-mcmodel=medlow) imposes that static addresses lie below 0x7ffff800 because of the limited range of LUI + ADDI.  For RV32, this restriction is artificial; we shouldn't perform the check.

aswaterman, 2015-11-18:
resolved by a36a4fba58a3f04e9fa439c16e8d6e453ef6067d

closed 2015-11-18 by aswaterman