riscv-collab/riscv-gnu-toolchain #104: Missing riscv64-unknown-linux-gnu-cc, and not symlinks

manuelafm, 2015-11-22:
Using the master branch updated today, I get these binaries installed:
```
-rwxr-xr-x 1 root root 4046136 Nov 22 12:18 riscv64-unknown-linux-gnu-addr2line
-rwxr-xr-x 2 root root 4207904 Nov 22 12:18 riscv64-unknown-linux-gnu-ar
-rwxr-xr-x 2 root root 5502984 Nov 22 12:18 riscv64-unknown-linux-gnu-as
-rwxr-xr-x 2 root root 3005360 Nov 22 12:30 riscv64-unknown-linux-gnu-c++
-rwxr-xr-x 1 root root 3999880 Nov 22 12:18 riscv64-unknown-linux-gnu-c++filt
-rwxr-xr-x 1 root root 2999312 Nov 22 12:30 riscv64-unknown-linux-gnu-cpp
-rwxr-xr-x 1 root root   92488 Nov 22 12:18 riscv64-unknown-linux-gnu-elfedit
-rwxr-xr-x 2 root root 3005360 Nov 22 12:30 riscv64-unknown-linux-gnu-g++
-rwxr-xr-x 2 root root 2995552 Nov 22 12:30 riscv64-unknown-linux-gnu-gcc
-rwxr-xr-x 2 root root 2995552 Nov 22 12:30 riscv64-unknown-linux-gnu-gcc-5.2.0
-rwxr-xr-x 1 root root  139880 Nov 22 12:30 riscv64-unknown-linux-gnu-gcc-ar
-rwxr-xr-x 1 root root  139744 Nov 22 12:30 riscv64-unknown-linux-gnu-gcc-nm
-rwxr-xr-x 1 root root  139760 Nov 22 12:30 riscv64-unknown-linux-gnu-gcc-ranlib
-rwxr-xr-x 1 root root 1995680 Nov 22 12:30 riscv64-unknown-linux-gnu-gcov
-rwxr-xr-x 1 root root 1878096 Nov 22 12:30 riscv64-unknown-linux-gnu-gcov-tool
-rwxr-xr-x 1 root root 3006232 Nov 22 12:30 riscv64-unknown-linux-gnu-gfortran
-rwxr-xr-x 1 root root 4568016 Nov 22 12:18 riscv64-unknown-linux-gnu-gprof
-rwxr-xr-x 4 root root 5588184 Nov 22 12:18 riscv64-unknown-linux-gnu-ld
-rwxr-xr-x 4 root root 5588184 Nov 22 12:18 riscv64-unknown-linux-gnu-ld.bfd
-rwxr-xr-x 2 root root 4078192 Nov 22 12:18 riscv64-unknown-linux-gnu-nm
-rwxr-xr-x 2 root root 4985680 Nov 22 12:18 riscv64-unknown-linux-gnu-objcopy
-rwxr-xr-x 2 root root 5601312 Nov 22 12:18 riscv64-unknown-linux-gnu-objdump
-rwxr-xr-x 2 root root 4207904 Nov 22 12:18 riscv64-unknown-linux-gnu-ranlib
-rwxr-xr-x 1 root root 1269120 Nov 22 12:18 riscv64-unknown-linux-gnu-readelf
-rwxr-xr-x 1 root root 4035424 Nov 22 12:18 riscv64-unknown-linux-gnu-size
-rwxr-xr-x 1 root root 4035320 Nov 22 12:18 riscv64-unknown-linux-gnu-strings
-rwxr-xr-x 2 root root 4985704 Nov 22 12:18 riscv64-unknown-linux-gnu-strip
```

At least "riscv64-unknown-linux-gnu-cc" is missing, but "-c++" for example is there.  I noticed because it is the one tried to be used by default in a project that I am trying to compile.

Also, a very minor issue -- even if some of them could be just symlinks none of the above are, all of them are "hard copies".

aswaterman, 2015-11-22:
None of the cross compilers on my machine (arm, armv8, riscv, mips) have a
cc symlink, but they all have a c++.  So this doesn't seem like a RISC-V
port issue.  Perhaps it is your distro that installed the cc symlink?

manuelafm, 2015-11-23:
I reported this issue because some project wanted to automatically use ${cross_prefix}-cc during configure/compilation and it failed (it used the native compiler instead).  Cannot pintpoint which one now, but it was some well known piece of software such as "vorbis" mentioned in the other report.

In Debian, for my regular (non-cross) compilers, cc points to gcc.  The cross-toolchain packages have -cc as symlink, but also -c++ (and -c89, -c99), and ld is a symlink to ld.bfd.  So I assumed that it was somehow standard practice to ship a binary/symlink with that name.

I reported this in the case that it was an oversight, I am not particularly troubled by this issue -- so feel free to close it if you don't think that it's a problem.

aswaterman, 2015-11-23:
Looking at the main gcc Makefile, there's no rule to install -cc; it's probably done externally by whatever tool prepares the debian package.

closed 2015-11-23 by aswaterman