riscv-collab/riscv-gnu-toolchain #105: Wrong syntax in usr/include/bits/mathcalls.h ?

manuelafm, 2015-11-22:
Hi,

Trying to compile libvorbis fails with the following errors.  Given that it's a file shipped by you I think that it's a problem with the toolchain port, so reporting just in case:
```
In file included from /usr/include/math.h:69:0,
                 from mdct.c:43:
/usr/local/sysroot/usr/include/bits/mathcalls.h:63:21: error: expected ')' before ',' token
 __MATHCALL_VEC (cos,, (_Mdouble_ __x));
                     ^
/usr/local/sysroot/usr/include/bits/mathcalls.h:65:21: error: expected ')' before ',' token
 __MATHCALL_VEC (sin,, (_Mdouble_ __x));
                     ^
/usr/local/sysroot/usr/include/bits/mathcalls.h:100:21: error: expected ')' before ',' token
 __MATHCALL_VEC (exp,, (_Mdouble_ __x));
                     ^
/usr/local/sysroot/usr/include/bits/mathcalls.h:109:21: error: expected ')' before ',' token
 __MATHCALL_VEC (log,, (_Mdouble_ __x));
                     ^
/usr/local/sysroot/usr/include/bits/mathcalls.h:153:21: error: expected ')' before ',' token
 __MATHCALL_VEC (pow,, (_Mdouble_ __x, _Mdouble_ __y));
                     ^
```

aswaterman, 2015-11-22:
That is a file that's part of mainline glibc, not the RISC-V port.
Something else is surely afoot, because I was able to build libogg and
libvorbis using --host=riscv64-unknown-linux

manuelafm, 2015-11-23:
Hmmm, something must be borked in my system then.  Sorry for the false alarm.

aswaterman, 2015-11-23:
good luck figuring it out -- I'll close this issue for now.

closed 2015-11-23 by aswaterman

manuelafm, 2015-11-27:
In the case that somebody arrives here because of searching the web for people with the same problem: I could compile those after I uninstalled the native glibc (libc6-dev).

machinekoder, 2017-06-12:
@manuelafm This solution sucks a lot

machinekoder, 2017-06-12:
To add another sucking solution: temporarily copy the math.h to a another place - then you don't need install all your dev toolchains at least