riscv-collab/riscv-gnu-toolchain #106: Assertion failure in md_apply_fix

timsifive, 2015-12-02:
The following source file (bug.S):
```
         sw     x9, 4, x1
```

And command: `/opt/riscv/bin/riscv64-unknown-elf-gcc -c bug.S`

Leads to this crash:
```
/tmp/ccbUoWbX.s: Assembler messages:
/tmp/ccbUoWbX.s: Internal error!
Assertion failure in md_apply_fix at /data/home/tnewsome/riscv-gnu-toolchain/src/binutils/gas/config/tc-riscv.c line 1889.
Please report this bug.
```

palmer-dabbelt, 2015-12-02:
@timsifive This is actually an invalid instruction format.  I added a check to avoid the gas assertion.  You're only meant to use symbols in this format, not numeric addresses.

Thanks for the bug report!

timsifive, 2015-12-02:
Great! Thank you.

closed 2015-12-03 by palmer-dabbelt