riscv-collab/riscv-gnu-toolchain #108: error: initializer element is not constant (building with gcc 4.9.2)

cliffordwolf, 2015-12-13:
I'm trying to build riscv-gnu-toolchain (commit c85720a) on raspbian jessie. This is my build script:

        sudo apt-get install autoconf automake autotools-dev curl libmpc-dev libmpfr-dev \
                             libgmp-dev gawk build-essential bison flex texinfo gperf

        sudo mkdir /opt/riscv32i
        sudo chown $USER /opt/riscv32i

        git clone https://github.com/riscv/riscv-gnu-toolchain riscv-gnu-toolchain-rv32i
        cd riscv-gnu-toolchain-rv32i

        mkdir build; cd build
        ../configure --with-xlen=32 --with-arch=I --prefix=/opt/riscv32i
        make

This works fine on Ubuntu 14.04.3 LTS with `gcc (Ubuntu 4.8.4-2ubuntu1~14.04) 4.8.4`, but on raspbian jessie with `gcc (Raspbian 4.9.2-10) 4.9.2` I get the following error:

    /bin/bash ./libtool --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I. -I/home/pi/riscv-gnu-toolchain-rv32i/build/src/binutils/bfd -I. -I/home/pi/riscv-gnu-toolchain-rv32i/build/src/binutils/bfd -I/home/pi/riscv-gnu-toolchain-rv32i/build/src/binutils/bfd/../include  -DHAVE_riscv_elf32_vec -DHAVE_elf32_le_vec -DHAVE_elf32_be_vec -DHAVE_plugin_vec  -DBINDIR='"/opt/riscv32i/bin"'  -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -g -O2 -MT elfxx-riscv.lo -MD -MP -MF .deps/elfxx-riscv.Tpo -c -o elfxx-riscv.lo /home/pi/riscv-gnu-toolchain-rv32i/build/src/binutils/bfd/elfxx-riscv.c
    libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I/home/pi/riscv-gnu-toolchain-rv32i/build/src/binutils/bfd -I. -I/home/pi/riscv-gnu-toolchain-rv32i/build/src/binutils/bfd -I/home/pi/riscv-gnu-toolchain-rv32i/build/src/binutils/bfd/../include -DHAVE_riscv_elf32_vec -DHAVE_elf32_le_vec -DHAVE_elf32_be_vec -DHAVE_plugin_vec -DBINDIR=\"/opt/riscv32i/bin\" -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -g -O2 -MT elfxx-riscv.lo -MD -MP -MF .deps/elfxx-riscv.Tpo -c /home/pi/riscv-gnu-toolchain-rv32i/build/src/binutils/bfd/elfxx-riscv.c -o elfxx-riscv.o
    /home/pi/riscv-gnu-toolchain-rv32i/build/src/binutils/bfd/elfxx-riscv.c:252:3: warning: left shift count >= width of type
       HOWTO (R_RISCV_CALL,   /* type */
       ^
    /home/pi/riscv-gnu-toolchain-rv32i/build/src/binutils/bfd/elfxx-riscv.c:252:3: error: initializer element is not constant
    /home/pi/riscv-gnu-toolchain-rv32i/build/src/binutils/bfd/elfxx-riscv.c:252:3: error: (near initialization for ‘howto_table[18].dst_mask’)
    /home/pi/riscv-gnu-toolchain-rv32i/build/src/binutils/bfd/elfxx-riscv.c:268:3: warning: left shift count >= width of type
       HOWTO (R_RISCV_CALL_PLT,  /* type */
       ^
    /home/pi/riscv-gnu-toolchain-rv32i/build/src/binutils/bfd/elfxx-riscv.c:268:3: error: initializer element is not constant
    /home/pi/riscv-gnu-toolchain-rv32i/build/src/binutils/bfd/elfxx-riscv.c:268:3: error: (near initialization for ‘howto_table[19].dst_mask’)
    Makefile:1607: recipe for target 'elfxx-riscv.lo' failed
    make[5]: *** [elfxx-riscv.lo] Error 1

cliffordwolf, 2015-12-13:
Quick update: I've now tried downgrading GCC to `gcc (Raspbian 4.8.4-1) 4.8.4` on my raspbian. I get the same error message. So the problem does not seem to be the gcc version. Maybe its because the raspberry pi is a 32 bit platform?

aswaterman, 2015-12-13:
Yes, I believe that's the issue.  Can you see if this pull request fixes
it?  I can't test it because I don't have access to a 32-bit machine at the
moment.

https://github.com/riscv/riscv-gnu-toolchain/pull/109

cliffordwolf, 2015-12-14:
Yes. I can now confirm that PR #109 fixes the problem. Thank you!

cliffordwolf, 2015-12-18:
closing as PR #109 has been merged.

closed 2015-12-18 by cliffordwolf