riscv-collab/riscv-gnu-toolchain #113: Unverified downloads in Makefile.in

neuschaefer, 2016-01-10:
Hi,
I noticed that the package fetching code in `Makefile.in` doesn't verify downloads, but immediately pipes them into `tar zxf -`. This creates an opportunity for [man in the middle attacks](https://en.wikipedia.org/wiki/Man_in_the_middle_attack), where an attacker intercepts the HTTP/FTP connections and modifies the tarballs to contain malicious files.

A simple solution would be to check every package against its known checksum (e.g. SHA256) and abort with an error message on mismatch.

palmer-dabbelt, 2016-01-12:
I'd be happy to accept a patch that does this, but I don't care enough to implement it myself right now.  As we move to getting all our stuff upstream and in Linux distributions this will be solved by using a proper package manager to get the RISC-V tools, which you can already do to some degree (Gentoo, homebrew, OpenEmbedded, for example).

neuschaefer, 2016-01-12:
Maybe I'll send a patch, I don't promise it, though :)

palmer-dabbelt, 2016-01-12:
OK.  I'm going to leave this open: it is a bug, just not one I'm going fix.

palmer-dabbelt, 2017-05-04:
I'm closing this: we're using git to track everything now, which seems to be close enough :)

closed 2017-05-04 by palmer-dabbelt