riscv-collab/riscv-gnu-toolchain #115: adding new opode extensions

vanjoe, 2016-01-15:
at the bottom of riscv-opc.c there is ca  comment 
```
/* Removed const from the following to allow for dynamic extensions to the
   built-in instruction set.  */
struct riscv_opcode *riscv_opcodes =
  (struct riscv_opcode *) riscv_builtin_opcodes;
int bfd_riscv_num_opcodes = RISCV_NUM_OPCODES;
```
This seems to indicate that there is a [planned] way to easily add instructions. Is there currently a clean way of doing this?
I've currently hacked the `riscv_buildin_opcodes` data structure to add my extensions, but I would love to know if there is a better way of doing this.

palmer-dabbelt, 2017-03-30:
There's no way to do this.  I'm closing this in favor of https://github.com/riscv/riscv-binutils-gdb/issues/38

closed 2017-03-30 by palmer-dabbelt

TommyMurphyTM1234, 2022-11-10:
This is a list of all issues logged that I can find that relate to adding new instructions to the toolchain:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1095
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1055
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/837
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/782
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1003
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/795
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/660
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/543
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/341
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/115

At this stage in the evolution of RISC-V,the majority of instruction addition will be done under the auspices of the relevant groups working on standardising, prototyping, and implementing/enhancing RISC-V extensions and their corresponding instructions. As such, there should be less of a demand for others to do so.

However, where somebody still wants to implement a non stadard custom extension/instruction here are some useful resources:

* GCC internals documentation: https://gcc.gnu.org/onlinedocs/gccint/
* https://hsandid.github.io/posts/risc-v-custom-instruction/
* https://ossg.bcs.org/blog/2020/10/29/adding-an-instruction-to-the-gnu-assembler/
* https://embdev.net/topic/228773
* https://groups.google.com/u/0/a/groups.riscv.org/g/sw-dev/c/sL_OHXYj3LY/m/NOm5Fwa3BAAJ
* https://forums.sifive.com/t/adding-custom-instruction-to-riscv-toolchain/4044/2