riscv-collab/riscv-gnu-toolchain #121: --enable-atomic --disable-float is broken

palmer-dabbelt, 2016-02-26:
Colin discovered the buildbot has been broken for a while.  Just last night we started building all the toolchain configurations again, and "--enable-atomic --disable-float" is broken.

````
/vm/home/buildbot/slave/riscv-gnu-toolchain_riscv32-unknown-linux-gnu_disable-multilib_enable-atomic_disable-float/build/install/lib/gcc/riscv32-unknown-linux-gnu/5.3.0/../../../../riscv32-unknown-linux-gnu/bin/ld: /vm/home/buildbot/slave/riscv-gnu-toolchain_riscv32-unknown-linux-gnu_disable-multilib_enable-atomic_disable-float/build/install/lib/gcc/riscv32-unknown-linux-gnu/5.3.0/libgcc.a(_ffssi2.o): can't link hard-float modules with soft-float modules
/vm/home/buildbot/slave/riscv-gnu-toolchain_riscv32-unknown-linux-gnu_disable-multilib_enable-atomic_disable-float/build/install/lib/gcc/riscv32-unknown-linux-gnu/5.3.0/../../../../riscv32-unknown-linux-gnu/bin/ld: failed to merge target specific data of file /vm/home/buildbot/slave/riscv-gnu-toolchain_riscv32-unknown-linux-gnu_disable-multilib_enable-atomic_disable-float/build/install/lib/gcc/riscv32-unknown-linux-gnu/5.3.0/libgcc.a(_ffssi2.o)
/vm/home/buildbot/slave/riscv-gnu-toolchain_riscv32-unknown-linux-gnu_disable-multilib_enable-atomic_disable-float/build/install/lib/gcc/riscv32-unknown-linux-gnu/5.3.0/../../../../riscv32-unknown-linux-gnu/bin/ld: /vm/home/buildbot/slave/riscv-gnu-toolchain_riscv32-unknown-linux-gnu_disable-multilib_enable-atomic_disable-float/build/install/lib/gcc/riscv32-unknown-linux-gnu/5.3.0/libgcc.a(_bswapsi2.o): can't link hard-float modules with soft-float modules
/vm/home/buildbot/slave/riscv-gnu-toolchain_riscv32-unknown-linux-gnu_disable-multilib_enable-atomic_disable-float/build/install/lib/gcc/riscv32-unknown-linux-gnu/5.3.0/../../../../riscv32-unknown-linux-gnu/bin/ld: failed to merge target specific data of file /vm/home/buildbot/slave/riscv-gnu-toolchain_riscv32-unknown-linux-gnu_disable-multilib_enable-atomic_disable-float/build/install/lib/gcc/riscv32-unknown-linux-gnu/5.3.0/libgcc.a(_bswapsi2.o)
/vm/home/buildbot/slave/riscv-gnu-toolchain_riscv32-unknown-linux-gnu_disable-multilib_enable-atomic_disable-float/build/install/lib/gcc/riscv32-unknown-linux-gnu/5.3.0/../../../../riscv32-unknown-linux-gnu/bin/ld: /vm/home/buildbot/slave/riscv-gnu-toolchain_riscv32-unknown-linux-gnu_disable-multilib_enable-atomic_disable-float/build/install/lib/gcc/riscv32-unknown-linux-gnu/5.3.0/libgcc.a(fp-bit.o): can't link hard-float modules with soft-float modules
/vm/home/buildbot/slave/riscv-gnu-toolchain_riscv32-unknown-linux-gnu_disable-multilib_enable-atomic_disable-float/build/install/lib/gcc/riscv32-unknown-linux-gnu/5.3.0/../../../../riscv32-unknown-linux-gnu/bin/ld: failed to merge target specific data of file /vm/home/buildbot/slave/riscv-gnu-toolchain_riscv32-unknown-linux-gnu_disable-multilib_enable-atomic_disable-float/build/install/lib/gcc/riscv32-unknown-linux-gnu/5.3.0/libgcc.a(fp-bit.o)
/vm/home/buildbot/slave/riscv-gnu-toolchain_riscv32-unknown-linux-gnu_disable-multilib_enable-atomic_disable-float/build/install/lib/gcc/riscv32-unknown-linux-gnu/5.3.0/../../../../riscv32-unknown-linux-gnu/bin/ld: /vm/home/buildbot/slave/riscv-gnu-toolchain_riscv32-unknown-linux-gnu_disable-multilib_enable-atomic_disable-float/build/install/lib/gcc/riscv32-unknown-linux-gnu/5.3.0/libgcc.a(dp-bit.o): can't link hard-float modules with soft-float modules
/vm/home/buildbot/slave/riscv-gnu-toolchain_riscv32-unknown-linux-gnu_disable-multilib_enable-atomic_disable-float/build/install/lib/gcc/riscv32-unknown-linux-gnu/5.3.0/../../../../riscv32-unknown-linux-gnu/bin/ld: failed to merge target specific data of file /vm/home/buildbot/slave/riscv-gnu-toolchain_riscv32-unknown-linux-gnu_disable-multilib_enable-atomic_disable-float/build/install/lib/gcc/riscv32-unknown-linux-gnu/5.3.0/libgcc.a(dp-bit.o)
/vm/home/buildbot/slave/riscv-gnu-toolchain_riscv32-unknown-linux-gnu_disable-multilib_enable-atomic_disable-float/build/install/lib/gcc/riscv32-unknown-linux-gnu/5.3.0/../../../../riscv32-unknown-linux-gnu/bin/ld: /vm/home/buildbot/slave/riscv-gnu-toolchain_riscv32-unknown-linux-gnu_disable-multilib_enable-atomic_disable-float/build/install/lib/gcc/riscv32-unknown-linux-gnu/5.3.0/libgcc.a(_clzsi2.o): can't link hard-float modules with soft-float modules
/vm/home/buildbot/slave/riscv-gnu-toolchain_riscv32-unknown-linux-gnu_disable-multilib_enable-atomic_disable-float/build/install/lib/gcc/riscv32-unknown-linux-gnu/5.3.0/../../../../riscv32-unknown-linux-gnu/bin/ld: failed to merge target specific data of file /vm/home/buildbot/slave/riscv-gnu-toolchain_riscv32-unknown-linux-gnu_disable-multilib_enable-atomic_disable-float/build/install/lib/gcc/riscv32-unknown-linux-gnu/5.3.0/libgcc.a(_clzsi2.o)
collect2: error: ld returned 1 exit status
````

aswaterman, 2016-03-20:
Where is this going wrong? Is libgcc the problem or something else?

palmer-dabbelt, 2016-04-04:
I think the problem is that fp-bit.o (the floating-point emulation routines) is being built with hardfloat enabled.  You can see "flw", "fsw" floating around in the binary.  I don't see softfloat getting passed.

````
/scratch/palmer.dabbelt/work/rgt-atomic_float/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/xgcc -B/scratch/palmer.dabbelt/work/rgt-atomic_float/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/ -B/home/eecs/palmer.dabbelt/work/rgt-atomic_float/install/riscv64-unknown-linux-gnu/bin/ -B/home/eecs/palmer.dabbelt/work/rgt-atomic_float/install/riscv64-unknown-linux-gnu/lib/ -isystem /home/eecs/palmer.dabbelt/work/rgt-atomic_float/install/riscv64-unknown-linux-gnu/include -isystem /home/eecs/palmer.dabbelt/work/rgt-atomic_float/install/riscv64-unknown-linux-gnu/sys-include    -g -O2 -O2  -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wstrict-prototypes -Wmissing-prototypes -Wold-style-definition  -isystem ./include   -fPIC -g -DIN_LIBGCC2 -fbuilding-libgcc -fno-stack-protector -Dinhibit_libc  -fPIC -I. -I. -I../.././gcc -I/scratch/palmer.dabbelt/work/rgt-atomic_float/riscv-gnu-toolchain/src/gcc/libgcc -I/scratch/palmer.dabbelt/work/rgt-atomic_float/riscv-gnu-toolchain/src/gcc/libgcc/. -I/scratch/palmer.dabbelt/work/rgt-atomic_float/riscv-gnu-toolchain/src/gcc/libgcc/../gcc -I/scratch/palmer.dabbelt/work/rgt-atomic_float/riscv-gnu-toolchain/src/gcc/libgcc/../include  -DHAVE_CC_TLS  -o fp-bit.o -MT fp-bit.o -MD -MP -MF fp-bit.dep  -c fp-bit.c -fvisibility=hidden -DHIDE_EXPORTS
````

aswaterman, 2016-04-04:
If you compile with --disable-float shouldn't -msoft-float become the default?

palmer-dabbelt, 2016-04-04:
That's what I thought, but I can only assume something has gone awry.  I
thought it still passed softfloat in a bunch of places when building the
libraries, but maybe that's just libc.

PR #133 (2016-04-06): soft/hard float argument fixes
This should fix #121, and possibly other brokenness we're seeing related to floating-point stuff.
closed 2016-04-06 by palmer-dabbelt