riscv-collab/riscv-gnu-toolchain #124: sizeof(long double) == 8

sorear, 2016-03-23:
```
$$ echo 'int foo() { return sizeof(long double); }' > x.c && riscv64-unknown-elf-gcc -save-temps -O2 -c x.c
        .file   "x.c"
        .text
        .align  2
        .globl  foo
        .type   foo, @function
foo:
        li      a0,8
        ret
        .size   foo, .-foo
        .ident  "GCC: (GNU) 5.3.0"
```

from a fresh source build on OSX 10.10.5 using the homebrew file.  https://blog.riscv.org/wp-content/uploads/2015/01/riscv-calling.pdf says that the result should be 16.

aswaterman, 2016-03-23:
This is a known ABI issue that must be resolved before upstreaming.
Fortunately, code using long double is still correct (it just has long
double == double).

kito-cheng, 2016-09-09:
I have implemented long double in my branch, I will send pull request  recently.

https://github.com/kito-cheng/riscv-gcc/commit/bbb3db7d720076fe6676d1db35158aa47e5944ac

aswaterman, 2016-09-09:
> kito-cheng/riscv-gcc@bbb3db7

Cool. We also need glibc support for 128b long double before we can merge,
but I assume that is easier.

Your long double ABI passes/returns arguments by reference, right? Using
pairs of floating point registers, like MIPS does, doesn't make sense for
us.

kito-cheng, 2016-09-13:
ISA spec 20.2:

...
Arguments smaller than a pointer-word are passed in the least-significant bits of argument registers.
Correspondingly, sub-pointer-word arguments passed on the stack appear in the lower addresses of
a pointer-word, since RISC-V has a little-endian memory system.
...

so according spec long double will pass by reference in RV32G, but pass in register pair in RV64G, 
and our implement it's seem follow this rule,

and for example, a function declare as `void foo(long double f)` and f pass by follow way:

RV32I (soft float): pass by ref
RV64I (soft float): pass in a0/a1
RV32G (hard float): pass by ref
RV64G (hard float): pass in fa0/fa1

is my understanding right?

btw, after test more arch,
our long double is work for RV32I, RV64I and RV64G, but still some bug in RV32G.

aswaterman, 2016-09-13:
It's OK if we change the ABI: it's already broken here.

It is never a good idea to pass in pairs of FP registers.  MIPS did that because their original ISA treated two 32b FP regs as a double-precision reg.  For RISC-V, it will always be low performance.

It's also probably a bad idea to pass in pairs of integer registers.  Here's why: systems that implement the Q extension will still use the G ABI.  If the values are passed in a0/a1, then RV64IMAFDQ systems will have to do SD+SD+FLQ to process the value.  If passed by reference, it's just FLQ.

So my proposal is to _always_ pass/return long doubles by reference.  Likewise, I would propose changing the RV32I ABI to pass/return doubles by reference.

There is also a desire to minimize the number of ABIs.  The following should give good performance and avoid a calling convention proliferation:

- RV32I, RV32IF, RV64IF: if FP arg size <= XLEN, pass in integer register; otherwise, by reference.  This means systems with F but not D use the soft-float calling convention, but can still use FP instructions on caller-saved FP registers.
- RV64IFD, RV64IFDQ: if FP arg size <= FLEN, pass in FP register; otherwise, by reference.

Does this proposal make sense?

aswaterman, 2016-09-13:
To simplify that logic:

- Use soft-float calling convention if FLEN < 64
- If using soft-float calling convention, pass in integer register if arg size <= XLEN, else by reference
- If using hard-float calling convention, pass in float-point register if arg size <= FLEN, else by reference

kito-cheng, 2016-09-13:
ABI change is look good to me, it's more simple,

but I am not sure long double pass by ref even with Q extension will impact how many performance? and dose it's worth for simplify ABI. 

and I am curious about what arch do your guys (Berkeley and SiFive) are tested now?

We are build RV32I/RV32IFD/RV64I/RV64IFD x newlib/glibc and only RV32I/RV64I x newlib do the full gcc testsuite now (will add RV32IFD/RV64IFD x newlib when FD extension for gdb sim is porting done).

aswaterman, 2016-09-13:
Glad we agree, and thanks for your help.  One detail I forgot: we should still use the fa0+fa1 pair to return _Complex numbers, or structs of two floats.

At SiFive and UCB, most of our focus/testing has been on RV32I and RV64G.

We do not need to implement the GQ ABI now (I don't expect HW implementations of the Q extension for at least a year), so we can punt on this decision.  I see two choices:

- Make the GQ ABI the same as G, as I originally proposed.  Q values cannot be callee-saved.  This means you can take full advantage of the Q instructions within a function, but can maintain compatibility with G code.  (This is also how the V-extension ABI will work.)
- Make a new, incompatible GQ ABI.  This will only slightly improve performance (passing/returning long doubles is dynamically infrequent) and will generally mean GQ systems need two sets of libraries.

I think it's pretty clear which option I favor :)

kito-cheng, 2016-09-14:
OK, so struct {long double v;} pass by ref too, struct {double a; double b;} pass by fa0+fa1 right?

sorear, 2016-09-14:
@aswaterman I'd be surprised if the second approach was actually more efficient at a system level.  Double-sized spills are more common in C code I've seen than long double anything, and you're now paying for larger stack frames and more stack memory traffic...

aswaterman, 2016-09-14:
@sorear yeah, I should've made clear that I was thinking about systems optimized for Q performance (e.g., boutique HPC applications).  This is the vastly uncommon case.  In the general case, the memory traffic consequent to larger stack frames is likely to dominate.

sorear, 2016-10-14:
FWIW now that I'm working on the Fedora port this is a somewhat higher priority for me; this bug makes gcc ABI-incompatible with the spec, and we need to recompile any package that uses long double after this bug is fixed; if we're going to have an ABI break flag day, the sooner the better.

aswaterman, 2016-10-14:
@kito-cheng can you pull-request your long double branch?

@palmer-dabbelt can you handle merging?  I'm going to be afk for several days.

kito-cheng, 2016-10-14:
@aswaterman I got new build problem after merge this https://github.com/riscv/riscv-gcc/commit/14fdbdae12f3b195ec6bbc9cb1c337ec1cc6265 for RV32* target, and  some regression for RV64G.

@sorear As I remember Fedora use RV64G, so I will make RV64* work first, and remain RV32* still use 
sizeof(long double) == 8, it's ok for you?

sorear, 2016-10-14:
@kito-cheng yeah that's fine.  Does this use libquadmath?

palmer-dabbelt, 2016-10-14:
I can look at a PR.

kito-cheng, 2016-10-19:
@sorear it's contain in libgcc.

kito-cheng, 2016-10-25:
@sorear RV64 part is merged :)

aswaterman, 2016-11-14:
This is now resolved for RV32, too.

closed 2016-11-14 by aswaterman