riscv-collab/riscv-gnu-toolchain #125: libgcc.a is not(?) optimized

sorear, 2016-03-26:
After installing with `brew install riscv-tools` and poking around for a bit, I found this:

```
$$ riscv64-unknown-elf-objdump -d /usr/local/lib/gcc/riscv64-unknown-elf/5.3.0/libgcc.a | head -n 32
In archive /usr/local/lib/gcc/riscv64-unknown-elf/5.3.0/libgcc.a:

_negdi2.o:     file format elf64-littleriscv

Disassembly of section .text:

0000000000000000 <__negti2>:
   0:   40b007b3                neg     a5,a1
   4:   00a036b3                snez    a3,a0

0000000000000008 <.LM3>:
   8:   fe010113                addi    sp,sp,-32

000000000000000c <.LM4>:
   c:   40a00733                neg     a4,a0
  10:   40d787b3                sub     a5,a5,a3
  14:   00e13823                sd      a4,16(sp)
  18:   00f13c23                sd      a5,24(sp)

000000000000001c <.LM5>:
  1c:   00b13423                sd      a1,8(sp)
  20:   00a13023                sd      a0,0(sp)

0000000000000024 <.LM6>:
  24:   01813583                ld      a1,24(sp)
  28:   01013503                ld      a0,16(sp)

000000000000002c <.LVL1>:
  2c:   02010113                addi    sp,sp,32
  30:   00008067                ret
```

I can only assume that's the output of a non-optimizing compile, because there's a lot of store forwarding opportunities.

aswaterman, 2016-03-27:
It's actually just spectacularly bad code (presumably specific to double-word arithmetic).  It is indeed compiled with -O2.

jim-wilson, 2018-01-25:
Current sources basically get this right.  I see

```
0000000000000000 <__negti2>:
   0:	87aa                	mv	a5,a0
   2:	1141                	addi	sp,sp,-16

0000000000000004 <.L0 >:
   4:	40b005b3          	neg	a1,a1
   8:	00f037b3          	snez	a5,a5

000000000000000c <.L0 >:
   c:	40a00533          	neg	a0,a0

0000000000000010 <.L0 >:
  10:	8d9d                	sub	a1,a1,a5
  12:	0141                	addi	sp,sp,16
  14:	8082                	ret
```

The main problem here is that we have an empty frame created.  This is because a local variable gets allocated to the stack, and then gcc is able to optimize away all of the stores and loads to that local, but we can't easily optimize away a frame once it is allocated.  Investigating, I see that we are missing a definition for MAX_FIXED_MODE_SIZE.  By default, only 64-bit and smaller variables can be allocated to registers.  64-bit targets usually increase the limit to 128-bits when in 64-bit mode.  With that change, the two addi instructions disappear.

There is also an issue with an unnecessary move instruction, but this is due to an interaction between the instruction scheduler and the register allocator, failing to notice that there is an instruction sort that requires one less register until after register allocation has already been performed.  We have neg, neg, snez, sub when register allocation is performed, which requires the extra move.  But then the second scheduling pass changes this to neg, snez, neg, sub which makes the extra move unnecessary, but we don't have an easy way to remove it at this point.  Even if we could fix it, minor changes to the scheduler may cause the instruction order to change again, and the extra regsiter may be required again.  These kinds of problems are very difficult to fix, and I'm not even going to try.

jim-wilson, 2018-01-27:
I added a patch upstream to fix the stack frame problem.  With that patch, I get the same code for a 32-bit negdi2 as I do for a 64-bit negti2.  The patch makes a 64-bit libstdc++ 3KB smaller.
    https://gcc.gnu.org/ml/gcc-patches/2018-01/msg02217.html

closed 2018-01-27 by jim-wilson