riscv-collab/riscv-gnu-toolchain #126: Slightly suboptimal codegen for `int` switches on RV64

sorear, 2016-03-26:
Found this while experimenting.  Minor, but reporting in case it's an easy fix.

```
$$ riscv64-unknown-elf-gcc -S -O3 asdf.c ; head -n50 asdf.[cs]
==> /tmp/asdf.c <==
extern void asdf(int);
void foo(int x) {
    switch (x) {
        case 0: asdf(10); break;
        case 1: asdf(11); break;
        case 2: asdf(12); break;
        case 3: asdf(13); break;
        case 4: asdf(14); break;
    }
}

==> /tmp/asdf.s <==
        .file   "asdf.c"
        .text
        .align  2
        .globl  foo
        .type   foo, @function
foo:
        li      a5,4
        bgtu    a0,a5,.L1
        sll     a0,a0,32
        lui     a5,%hi(.L4)
        add     a5,a5,%lo(.L4)
        srl     a0,a0,30
        add     a0,a0,a5
        lw      a5,0(a0)
        jr      a5
        .section        .rodata
        .align  2
        .align  2
.L4:
        .word   .L3
        .word   .L5
        .word   .L6
        .word   .L7
        .word   .L8
        .text
.L7:
        li      a0,13
        tail    asdf
.L8:
        li      a0,14
        tail    asdf
.L3:
        li      a0,10
        tail    asdf
.L5:
        li      a0,11
        tail    asdf
.L6:
        li      a0,12
        tail    asdf
.L1:
        ret
        .size   foo, .-foo
        .ident  "GCC: (GNU) 5.3.0"
```

The `sll` / `srl` is unneeded because after the `bgtu` we know that the upper half of `a0` is clear.  The compiler generates better code if the argument is typed as `long`, even though the behavior is identical.

aswaterman, 2016-03-27:
Also, the %lo relocation could go on the lw rather than requiring a second add.

I don't have time to look into this now, but it is certainly suboptimal.

jim-wilson, 2018-01-27:
I looked at the switch issue.  The compiler does know that parameters are sign-extended.  I was able to write a patch to use this info to eliminate the extra sign-extend.  When I tried changing the parameter type to char, I noticed an unnecessary zero-extend before the bgtu, and wrote a similar patch for that.  These patches need a lot more testing, and aren't reasonable to submit until after the gcc-8 branch is made, so this will have to wait 2 months or so. 

I haven't tried looking at the %lo optimization yet.  That is a different issue.

[subreg-promoted.txt](https://github.com/riscv/riscv-gnu-toolchain/files/1669572/subreg-promoted.txt)

jim-wilson, 2018-05-11:
My two subreg promoted patches have been posted to the FSF GCC list.  The first one requires approval and is waiting for review.  The second one I can self approve, but the testcases in it won't work without the first patch, so I'm waiting for the review.

Meanwhile, I took a look at the problem with the extra add instruction that Andrew mentioned.  Given
    char array[10];
    int sub (int i) { return array[i]; }
The compiler generates
	lui	a5,%hi(array)
	addi	a5,a5,%lo(array)
	add	a0,a5,a0
	lbu	a0,0(a0)
I wrote a patch for riscv_legitimize_address to fix this.  I now get
	lui	a5,%hi(array)
	add	a0,a5,a0
	lbu	a0,%lo(array)(a0)
However, during testing, I found that this fails when linker relaxation converts the address to a gp-relative access.  I get after linking
   101ac:       00a78533                add     a0,a5,a0
   101b0:       9981c503                lbu     a0,-1640(gp) # 11ba8 <array>
which is wrong.  The linker assumes that a %hi always directly feeds into a %lo.  I don't have any good ideas on how to fix this at the moment.  Since the linker doesn't do code analysis, we probably need a new relocation to represent this case, which means new assembly syntax, an official ABI change, etc.  That means this is no longer an easy gcc problem to fix.
[extra-array-add-patch.txt](https://github.com/riscv/riscv-gnu-toolchain/files/1996812/extra-array-add-patch.txt)
[extra-array-add-testcase.txt](https://github.com/riscv/riscv-gnu-toolchain/files/1996813/extra-array-add-testcase.txt)

aswaterman, 2018-05-12:
I believe we would need to define new relaxations to handle this, kind of
like TLS LE works. That’s a lot of work, so it probably makes sense to
deprioritize this one.

lazyparser, 2021-03-03:
This issue had not been updated for 2 years. Is this issue still relevant? or can we close it safely?

jim-wilson, 2021-03-04:
This issue with the extra add instruction hasn't been fixed  yet.  But this is listed on the code speed SIG gcc optimization project page so we don't need this issue to track it.  The orther issues have already been fixed.
https://github.com/riscv/riscv-code-speed-optimization/blob/main/projects/gcc-optimizations.adoc

TommyMurphyTM1234, 2022-10-06:
Seems to be fixed now?
Using the latest nightly build available (https://github.com/riscv-collab/riscv-gnu-toolchain/releases/tag/2022.09.30) I get this:
```
==> asdf.c <==
extern void asdf(int);
void foo(int x) {
    switch (x) {
        case 0: asdf(10); break;
        case 1: asdf(11); break;
        case 2: asdf(12); break;
        case 3: asdf(13); break;
        case 4: asdf(14); break;
    }
}

==> asdf.s <==
	.file	"asdf.c"
	.option nopic
	.attribute arch, "rv64i2p0_m2p0_a2p0_f2p0_d2p0_c2p0"
	.attribute unaligned_access, 0
	.attribute stack_align, 16
	.text
	.align	1
	.globl	foo
	.type	foo, @function
foo:
	li	a5,4
	bgtu	a0,a5,.L1
	lui	a5,%hi(.L4)
	addi	a5,a5,%lo(.L4)
	slli	a0,a0,2
	add	a0,a0,a5
	lw	a5,0(a0)
	jr	a5
	.section	.rodata
	.align	2
	.align	2
.L4:
	.word	.L8
	.word	.L7
	.word	.L6
	.word	.L5
	.word	.L3
	.text
.L5:
	li	a0,13
	tail	asdf
.L3:
	li	a0,14
	tail	asdf
.L8:
	li	a0,10
	tail	asdf
.L7:
	li	a0,11
	tail	asdf
.L6:
	li	a0,12
	tail	asdf
.L1:
	ret
	.size	foo, .-foo
	.ident	"GCC: (g2ee5e430018) 12.2.0"
```

closed 2022-10-06 by TommyMurphyTM1234