riscv-collab/riscv-gnu-toolchain #127: Newlib printf for floats/doubles looks broken

seldridge, 2016-03-28:
The following program is giving me garbage outputs.
```c
#include <stdio.h>

int main() {
  printf("%f\n", 0.7);
}
```
Output:
```
> riscv64-unknown-elf-gcc -Wall float-test.c -o float-test.o
> spike pk float-test.o
190359837124021820742322352196969651504809874340871176408181720254371418237508877811893858511874258431232803779565493232680859306993996370319781741109529245266007018994166819642276839424.000000
```
This is from rocket-chip master, or, specifically:
```
34a245b943bcae60b6ccece8aba86670cd0f2631 riscv-tools (homebrew-66-g34a245b)
d5278834830bdd2cb8586f25fe05ae917b0eb949 riscv-pk (remotes/origin/priv-1.9)
11fb8bbdb07e80a5b02c9e8753482fb739152bef riscv-gnu-toolchain (remotes/origin/HEAD)
```
I have not had a chance to dig around and see what's going on here.

aswaterman, 2016-03-28:
It seems specific to the newlib port; let me know if you figure out what
the cause is.

seldridge, 2016-03-28:
@aswaterman -- I bisected just now and it looks like 4f0411fecc9c862ffb2c7f4501b30a895c979784 is doing it for newlib. 

I blindly changed the newlib patch to use `RISCVEL` and not the old `_RISCVEL` and it works for that simple program above, but I don't have your scope to verify that this is the correct way to go about it. I'll throw a pull request your way for this quick fix.

Note, that `_RISCVEL` is still used by:
* gcc/libgcc/config/riscv/t-tpbit
And `__RISCVEL__` is used by
* linux-headers/include/asm/bytorder.h

aswaterman, 2016-03-28:
Thanks for doing the detective work.  I fixed the problem upstream before noticing you had also submitted a pull request.

commit 4a4918ab (2016-03-28): newlib: use correct macro to detect endianness

Closes #127.  h/t @seldridge
closed 2016-03-28 by aswaterman