riscv-collab/riscv-gnu-toolchain #134: Build fails with "configure: error: cannot compute suffix of object files: cannot compile"

cliffordwolf, 2016-04-08:
I am using the following build script:

```
mkdir build; cd build
../configure --with-xlen=32 --with-arch=I --prefix=/opt/riscv32i_test
```

With git head of `riscv-gnu-toolchain` I get the following build error:

```
checking for riscv32-unknown-elf-gcc... /home/clifford/Temp/riscv-gnu-toolchain-rv32i/build/build-gcc-newlib/./gcc/xgcc -B/home/clifford/Temp/riscv-gnu-toolchain-rv32i/build/build-gcc-newlib/./gcc/ -nostdinc -B/home/clifford/Temp/riscv-gnu-toolchain-rv32i/build/build-gcc-newlib/riscv32-unknown-elf/newlib/ -isystem /home/clifford/Temp/riscv-gnu-toolchain-rv32i/build/build-gcc-newlib/riscv32-unknown-elf/newlib/targ-include -isystem /home/clifford/Temp/riscv-gnu-toolchain-rv32i/build/src/newlib-gcc/newlib/libc/include -B/home/clifford/Temp/riscv-gnu-toolchain-rv32i/build/build-gcc-newlib/riscv32-unknown-elf/libgloss/riscv32 -L/home/clifford/Temp/riscv-gnu-toolchain-rv32i/build/build-gcc-newlib/riscv32-unknown-elf/libgloss/libnosys -L/home/clifford/Temp/riscv-gnu-toolchain-rv32i/build/src/newlib-gcc/libgloss/riscv32 -B/opt/riscv32i_test/riscv32-unknown-elf/bin/ -B/opt/riscv32i_test/riscv32-unknown-elf/lib/ -isystem /opt/riscv32i_test/riscv32-unknown-elf/include -isystem /opt/riscv32i_test/riscv32-unknown-elf/sys-include   
checking for suffix of object files... configure: error: in `/home/clifford/Temp/riscv-gnu-toolchain-rv32i/build/build-gcc-newlib/riscv32-unknown-elf/libgcc':
configure: error: cannot compute suffix of object files: cannot compile
See `config.log' for more details.
make[2]: *** [configure-target-libgcc] Error 1
make[2]: Leaving directory `/home/clifford/Temp/riscv-gnu-toolchain-rv32i/build/build-gcc-newlib'
make[1]: *** [all] Error 2
make[1]: Leaving directory `/home/clifford/Temp/riscv-gnu-toolchain-rv32i/build/build-gcc-newlib'
make: *** [stamps/build-gcc-newlib] Error 2
```

I do not get this with acfa480 (from a few days ago).

palmer-dabbelt, 2016-04-08:
The problem is that "--enable-float" and "--enable-atomic" are on by default, but the "I" ISA doesn't support either of those.  The linker was recently changed to respect the "--with-isa" argument and now should actually be setting the ELF flags correctly for your ISA (it was ignored before), but the top-level riscv-gnu-toolchain's "--with-isa" argument doesn't set "--disable-float" or "--disable-atomic" accordingly when you request an ISA that can't support them.

If you configure with

````
./configure --with-xlen=32 --with-arch=I --prefix=... --disable-float --disable-atomic
````

your toolchain should build again.  I'm in the middle of making ISA strings work correctly in the top-level build stuff, to try and get multilib working again -- see #46.

cliffordwolf, 2016-04-08:
Thanks! Adding `--disable-float --disable-atomic` solved the problem.

palmer-dabbelt, 2016-05-30:
This is fixed by https://github.com/riscv/riscv-gnu-toolchain/pull/146

closed 2016-05-30 by palmer-dabbelt

LakshmiKanth-77, 2021-03-27:
thank me later:
./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d--disable-float--disable-atomic