riscv-collab/riscv-gnu-toolchain #135: "addi x2, x2, 0" incorrectly assembled to 0x6101 (jalr sp) on RV32IC

cliffordwolf, 2016-04-08:
The bogus (nop) instruction `addi x2, x2, 0` is encoded as 0x6101 by `riscv32-unknown-elf-as`, which also decodes as `addi x2, x2, 0` with `riscv32-unknown-elf-objdump -d`:

```
$ echo "addi x2, x2, 0" > test.s
$ echo "SECTIONS { .memory : { . = 0x000000; *(.text); *(*); end = .; } }" > test.lds
$ riscv32-unknown-elf-gcc -m32 -march=RV32IC -nostdlib -Wl,-Bstatic,-T,test.lds -o test.elf test.s
$ riscv32-unknown-elf-objdump -d test.elf 

test.elf:     file format elf32-littleriscv

Disassembly of section .memory:

00000000 <end-0x2>:
   0:	6101                	addi	sp,sp,0
```

However, `spike` decodes 0x6101 as `jalr sp`:

```
$ spike -l test.elf
core   0: 0x0000000000000000 (0x00006101) jalr    sp
core   0: exception trap_illegal_instruction, epc 0x0000000000000000
core   0: 0x0000000000000008 (0x00000000) addi    s0, sp, 0
core   0: exception trap_illegal_instruction, epc 0x0000000000000008
core   0: 0x0000000000000008 (0x00000000) addi    s0, sp, 0
```

I'm having troubles accessing riscv.org at the moment, so I can't check with the compressed ISA spec what the correct interpretation of opcode 0x6101 is, but if I remember correctly, the 16 bit encoding for ALU ops with immediate are only valid for nonzero immediates. So I would assume this is a bug in 
riscv-gnu-toolchain, not spike.

aswaterman, 2016-04-08:
GAS is correct and Spike's disassembler is wrong: 011x00010xxxxx01 is, indeed, C.ADDI16SP.  I'll go fix Spike.

aswaterman, 2016-04-08:
My bad.  Both are wrong.  The immediate is required to be nonzero, so GAS should've selected C.MV instead of C.ADDI16SP.

(Spike itself is correct in detecting the illegal instruction.)

closed 2016-04-08 by aswaterman

aswaterman, 2016-04-08:
the assembler is fixed by af92ce8

cliffordwolf, 2016-04-09:
Trying to build af92ce80cff334c7eca3ba6b194dc3371336d7a5 I get the following error:

```
/tmp/ccXP3tmx.s: Assembler messages:
/tmp/ccXP3tmx.s:101: Error: bad expression
make[4]: *** [unwind-c.o] Error 1
make[4]: *** Waiting for unfinished jobs....
/tmp/cchXIiQu.s: Assembler messages:
/tmp/cchXIiQu.s:331: Error: bad expression
/tmp/cchXIiQu.s:1941: Error: bad expression
/tmp/cchXIiQu.s:2245: Error: bad expression
/tmp/cchXIiQu.s:2324: Error: bad expression
/tmp/cchXIiQu.s:2348: Error: bad expression
/tmp/cchXIiQu.s:2349: Error: bad expression
/tmp/cchXIiQu.s:2605: Error: bad expression
/tmp/cchXIiQu.s:3075: Error: bad expression
/tmp/cchXIiQu.s:3168: Error: bad expression
/tmp/cchXIiQu.s:3416: Error: bad expression
make[4]: *** [unwind-dw2-fde.o] Error 1
/tmp/ccPlF8St.s: Assembler messages:
/tmp/ccPlF8St.s:24: Error: bad expression
/tmp/ccPlF8St.s:431: Error: bad expression
/tmp/ccPlF8St.s:2852: Error: bad expression
/tmp/ccPlF8St.s:2855: Error: bad expression
/tmp/ccPlF8St.s:2860: Error: bad expression
/tmp/ccPlF8St.s:4672: Error: bad expression
/tmp/ccPlF8St.s:4691: Error: bad expression
/tmp/ccPlF8St.s:5085: Error: bad expression
/tmp/ccPlF8St.s:5249: Error: bad expression
/tmp/ccPlF8St.s:5444: Error: bad expression
/tmp/ccPlF8St.s:5848: Error: bad expression
/tmp/ccPlF8St.s:5988: Error: bad expression
/tmp/ccPlF8St.s:6095: Error: bad expression
/tmp/ccPlF8St.s:6182: Error: bad expression
make[4]: *** [unwind-dw2.o] Error 1
make[4]: Leaving directory `/home/clifford/Work/picorv32/riscv-gnu-toolchain-riscv32i/build/build-gcc-newlib/riscv32-unknown-elf/libgcc'
make[3]: *** [all-target-libgcc] Error 2
make[3]: Leaving directory `/home/clifford/Work/picorv32/riscv-gnu-toolchain-riscv32i/build/build-gcc-newlib'
make[2]: *** [all] Error 2
make[2]: Leaving directory `/home/clifford/Work/picorv32/riscv-gnu-toolchain-riscv32i/build/build-gcc-newlib'
make[1]: *** [stamps/build-gcc-newlib] Error 2
make[1]: Leaving directory `/home/clifford/Work/picorv32/riscv-gnu-toolchain-riscv32i/build'
make: *** [build-riscv32i-tools] Error 2
```

The build was configured with `--with-xlen=32 --with-arch=I --prefix=/opt/riscv32i --disable-float --disable-atomic`. The same build script works fine with 3134bf43d81720863c8b8bc8fd4acd1017bca3fc.

Some of the lines with "bad expressions" in them:

        add     a4,a4,%lo(__clz_tab)
        add     a4,a4,%lo(.L43)
        add     a5,a5,%lo(terminator.4183)
        add     s1,s1,%lo(fde_unencoded_compare)
        add     s1,s1,%lo(fde_mixed_encoding_compare)

aswaterman, 2016-04-09:
Sorry about that, Cliff.  I think I rebuilt gcc without the modified
binutils in my path so failed to test the change properly. I'll fix later
today.

aswaterman, 2016-04-09:
fixed, sorry about that.

The Spike disassembler fix will come later, as the whole approach needs to be rethought (but I'd of course accept a quick fix in the short term).

commit 888ae43f (2016-04-09): binutils: unfuck previous commit

The addi -> c.mv optimization was incorrect.

Resolves #135.