riscv-collab/riscv-gnu-toolchain #142: How to specify .text address

colin4124, 2016-05-13:
I use MIPS simulator SPIM, I can use `.text 0x8000180` to specify `.text` segment address to `0x80000180` .

But I use `riscv64-unknown-elf-as`, and then `riscv64-unknown-elf-objcopy` to verilog , it does not work.

So, how I can do that?

gkelly, 2016-05-13:
I believe `.org 0x8000180`, or use a linker script.

colin4124, 2016-05-13:
@gkelly Yes, it works.

>  addi x1, x0, 32
  addi x1, x0, 32
    .org 0x10
  addi x1, x0, 32

I use riscv64-unknown-elf-as, and then riscv64-unknown-elf-objcopy to verilog

> @00000000
93 00 00 02 93 00 00 02 00 00 00 00 00 00 00 00 
93 00 00 02 

Is there way to generate that:
> @00000000
93 00 00 02 93 00 00 02 
@00000010 
93 00 00 02

gkelly, 2016-05-13:
I played with objcopy a bit and I don't see a way to get it to skip in-section padding for the verilog output option.  So here's the slightly longer way of getting this done:

`test.S`:

    .section .first, "xa"
    addi x1, x0, 32
    addi x1, x0, 32
    
    .section .second, "xa"
    addi x1, x0, 32

`test.ld`:

    SECTIONS {
      . = 0;
      .first : { *(.first) }
      . = 0x10;
      .second : { *(.second) }
    }

`riscv64-unknown-elf-as test.S -o test.o`
`riscv64-unknown-elf-ld --script test.ld test.o -o test.elf`
`riscv64-unknown-elf-objcopy -O verilog test.elf test.v`

`test.v`:

    @00000000
    93 00 00 02 93 00 00 02 
    @00000010
    93 00 00 02

colin4124, 2016-05-13:
Thx very much!

closed 2016-05-13 by colin4124

colin4124, 2016-05-15:
Hi, @gkelly 

I have never use  linker script before, Would you recommend me some learning source?

Think very much!

AngelTerrones, 2016-05-15:
Could be useful: http://www.math.utah.edu/docs/info/ld_3.html

colin4124, 2016-05-15:
@AngelTerrones Thinks very much!