riscv-collab/riscv-gnu-toolchain #143: About riscv64-unknown-elf-as generate BEQ and BNQ instructions

colin4124, 2016-05-14:
The file `tmp.s` as follow:

>  beq  x16, x10, 9
bne x16, x10, 9

After I do that: `riscv64-unknown-elf-as -m32 tmp.s` and `riscv64-unknown-elf-objcopy a.out -O verilog tmp.dat`, the `tmp.dat` file as follow:

> @00000000
63 14 A8 00 6F 00 00 00 63 04 A8 00 6F 00 00 00 

So, `riscv64-unknown-elf-as` generate that

`beq  x16, x10, 9` to `0x00A81463`,  that is 0000000 01010 10000 **001** 01000 1100011 .
`bne  x16, x10, 9` to `0x00A80463`, that is 0000000 01010 10000 **000** 01000 1100011

But in _RISC-V Instruction Set Manual Volume I_, Chapter 8 , RV32I Base Instruction Set Table show that:

BEQ `funct3` field is **000**, not the  **001**;  BNE `funct3` field is **001**, not the  **000**.

There is another question, why `6F 00 00 00` would additional to `beq` and `bne` instructions?

aswaterman, 2016-05-14:
Instead of

      beq x16, x10, label

the assembler emits something like

      bne x16, x10, 1f
      j label
    1:

because it can't prove that `label` is within range.  So everyone is correct :p

colin4124, 2016-05-14:
@aswaterman 

Why BEQ funct3 field is 001, and BNE funct3 field is 000? It does not match the _RISC-V Instruction Set Manual_

aswaterman, 2016-05-14:
BEQ funct3 is 000, as you said.

What I'm trying to say is that the assembler converts BEQ into BNE+J if the branch is far away.  Since you aren't branching to a defined label, it's performing this transformation.

If you define a label, then branch to that label, you'll see that BEQ is assembled as you'd expect.

closed 2016-05-14 by aswaterman

colin4124, 2016-05-14:
@aswaterman thx very much!

colin4124, 2016-05-14:
@aswaterman Are there any documents about  how to write assembly language and the `riscv64-unknown-elf-as` converts convention( as you told me above).

I just find one [The GNU Assembler](http://tigcc.ticalc.org/doc/gnuasm.html). Any others recommend ?

Think very much.

aswaterman, 2016-05-14:
There's actually no assembly programmer's manual yet.  The syntax is generally similar to that of MIPS.  Looking at GCC's compiled output (using the -S flag) is a good way to learn by example.

colin4124, 2016-05-15:
Hi, @aswaterman 

Sorry disturbing you again.

`beq x16, x10, 9` and `beq x16, x10, 10` both all assemble to `0x00A81463 0x0000006F` two instructions.  Whether target address using number(such as 9) is invalid ? Dose target address only use label ? Is there any way to use  absolute address?

Thinks again!

aswaterman, 2016-05-18:
Assembled objects are inherently incomplete; they need to be linked to have all the addresses resolved.

I'd suggest using gcc as a guide for how to write complete programs that correctly link.