riscv-collab/riscv-gnu-toolchain #149: Adding Linker command file

susi655, 2016-05-23:
For adding linker command file,
$riscv64-unknown-elf-gcc -T ldStart.ld check2.c
It gave error like the following..
/home/suseela/Desktop/riscv/lib/gcc/riscv64-unknown-elf/5.3.0/../../../../riscv64-unknown-elf/lib/crt0.o: In function `.L11':
(.text+0x0): undefined reference to `_gp'
/home/suseela/Desktop/riscv/lib/gcc/riscv64-unknown-elf/5.3.0/../../../../riscv64-unknown-elf/lib/crt0.o: In function `.L0 ':
(.text+0x8): undefined reference to `_fbss'
/home/suseela/Desktop/riscv/lib/gcc/riscv64-unknown-elf/5.3.0/../../../../riscv64-unknown-elf/lib/crt0.o: In function `.L0 ':
(.text+0x10): undefined reference to `_end'
/home/suseela/Desktop/riscv/lib/gcc/riscv64-unknown-elf/5.3.0/../../../../riscv64-unknown-elf/lib/crt0.o: In function `.L0 ':
(.text+0x24): undefined reference to `__libc_fini_array'
/home/suseela/Desktop/riscv/lib/gcc/riscv64-unknown-elf/5.3.0/../../../../riscv64-unknown-elf/lib/crt0.o: In function `.L0 ':
(.text+0x2c): undefined reference to `atexit'
/home/suseela/Desktop/riscv/lib/gcc/riscv64-unknown-elf/5.3.0/../../../../riscv64-unknown-elf/lib/crt0.o: In function `.L0 ':
(.text+0x34): undefined reference to `__libc_init_array'
/home/suseela/Desktop/riscv/lib/gcc/riscv64-unknown-elf/5.3.0/../../../../riscv64-unknown-elf/lib/crt0.o: In function `.L0 ':
(.text+0x4c): undefined reference to `exit'
collect2: error: ld returned 1 exit status

Out of these i was able to find _gp and _end in ldScript, i added them in my ld file. Remaining I was unable to. But i don't know this will be correct or not. Please suggest the solution. My aim is to make the tool chain suitable for my target board.

gkelly, 2016-05-23:
Those symbols are all provided by the default linker script (or by libraries which the default linker script pulls in), which is located at `$RISCV/riscv64-unknown-elf/lib/riscv.ld`, assuming your `$RISCV` is set to your `--prefix` and that your build arch was riscv64.  Consider starting from `riscv.ld` and making modifications instead of creating a new script from scratch.

If you truly don't depend on the standard C library, then you can use `-nostdlib` when compiling.  However, this will require you to write your own `crt0` routines, which include initializing machine state.

palmer-dabbelt, 2017-05-04:
I'm going to close this, as I don't think it's a RISC-V problem.

closed 2017-05-04 by palmer-dabbelt