riscv-collab/riscv-gnu-toolchain #154: (bad codegen): if-convert(?) gone wrong in -O3

ccelio, 2016-05-28:
Here is a snippet of code where a single if-condition guards an assignment to two variables (a and b), and it turns into TWO branches and 4 extra moves.

It looks like an attempt at if-conversion gone wrong (as RISC-V lacks cmoves).

``` c
int testme(size_t a, size_t b, size_t* queue, size_t len) {

   for (size_t k = 0; k < len; k++) {
      if (queue[k] < a) {
         a = queue[k];
         b = k; 
      }
   }

   return a + b;
}
```

Compiled with:

     riscv64-unknown-elf-gcc -c -S -O3 -o hello.s hello.c

```asm
    .file "hello.c"             
    .text                       
    .align   2                  
    .globl   testme             
    .type testme, @function     
 testme:                        
    beqz  a3,.L7                
    li a5,0                     
 .L6:                           
    ld a4,0(a2)                 
    mv a7,a5                   ???
    mv a6,a4                   ???
    bleu  a4,a0,.L3             
    mv a6,a0                    
 .L3:                           
    bltu  a4,a0,.L5            a second branch?!
    mv a7,a1                    
 .L5:                           
    add   a5,a5,1               
    mv a1,a7                   :(
    add   a2,a2,8               
    mv a0,a6                   :(
    bne   a3,a5,.L6             
 .L2:                           
    addw  a0,a1,a6              
    ret                         
 .L7:                           
    mv a6,a0                    
    j  .L2                      
    .size testme, .-testme      
    .align   2                  
    .globl   func               
    .type func, @function       
 func:                          
    ret                         
    .size func, .-func          
    .ident   "GCC: (GNU) 6.1.0" 
````

---------------

This may have to sit quietly until RISC-V is mainlined. I don't have an ISA without cmove lying around for me to try this on, and specifying via the "-fno" flags to stop if-conversion doesn't seem to do anything.

ccelio, 2016-05-28:
Thanks to @aswaterman, this problem goes away when turning off the "tree-loop-if-convert" pass:

     riscv64-unknown-elf-gcc -c -S -O3 -fno-tree-loop-if-convert -o hello.s hello.c

As this code pattern shows up in SPECINT, I'll try to report back on the performance effect this has.

ccelio, 2016-05-29:
For `445.gobmk`, it saves 1.4% on the instruction count.  One of the critical loops goes down from 17 instructions to 12 instructions.  Until we can get a fix into gcc proper, I think this optimization pass should be disabled since we have neither cmoves nor vector units in RISC-V.

aswaterman, 2016-05-31:
How about overall?  I'm fine with disabling this by default for RISC-V if it doesn't regress on the other benchmarks.

aswaterman, 2016-11-14:
@ccelio is the verdict that disabling this option is better overall for SPEC?

ccelio, 2016-11-14:
It looks to me to be a help on all benchmarks and does not regress on any workload.

lazyparser, 2021-03-04:
Hi @ccelio, do you know whether this bug got fixed or not?

/cc @jim-wilson if it is still open and worth to fix, can it be tagged as a `good-first-bug` for new gcc contributors?

aswaterman, 2021-03-04:
AFAICS this is fixed.  ` -O3` itself doesn't appear to trigger this; only explicitly using `-ftree-loop-if-convert` triggers it, which is OK.

jim-wilson, 2021-03-05:
I am puzzled how this could have been broken in the first place, since tree-loop-if-convert is only enabled by default if the user enables vectorization, or something like a pragma indicates that this loop should be vectorized.  I can't reproduce with gcc-7.3.  Maybe some pre-upstream gcc version was broken.

aswaterman, 2021-03-05:
@jim-wilson Yeah, the bug report was from gcc-6.1, so I'm guessing it was fixed upstream before we upstreamed the port.  I will close the issue.

closed 2021-03-05 by aswaterman