riscv-collab/riscv-gnu-toolchain #159: `make` only works the first time

bbarenblat, 2016-07-06:
Currently, running `make` builds and installs the toolchain. However, if you then delete the installed files and run `make` again, it reports ‘Nothing to be done’. To reinstall the built files, you need to run

    make -C build-binutils-newlib install && make -C build-gcc-newlib install

This is confusing; `make` should always leave the system in the same state.

jim-wilson, 2018-01-25:
This is hard to fix because riscv-gnu-toolchain is a collection of different packages, each with its own conventions.  Plus, there are interdependencies between packages that force things to be built and installed in certain orders, and in some cases, packages need to be built more than once for correct results.  We could maybe add an install rule to the Makefile, or improve the docs to explain how to do this.  But fixing "make" to do the right thing is not practical.

TommyMurphyTM1234, 2022-09-29:
@jim-wilson explains why addressing this is not really feasible especially given the very limited resources available. Obviously, contributions are welcome so if somebody is able to implement this that would be great. However, given that over four years have elapsed with no update/activity here, I think it best to close this issue at this stage.