riscv-collab/riscv-gnu-toolchain #162: spike execution of 32-bit newlib program doesn't terminate

pmundkur, 2016-07-12:
I'm trying to build 32-bit exes via compilation to assembly with a
recent toolchain repo using the master branch.  I'm using the newlib toolchain, compiled with
--enable-multilib.

$ riscv64-unknown-elf-gcc --version
riscv64-unknown-elf-gcc (GCC) 6.1.0
$ riscv64-unknown-elf-gcc -m32 -S -o switch-gcc64-m32.s switch.c
$ riscv64-unknown-elf-gcc -m32 -c -o switch-gcc64-m32.o switch-gcc64-m32.s
$ riscv64-unknown-elf-gcc -m32 -o rv64-m32.exe switch-gcc64-m32.o
$ file rv64-m32.exe
rv64-m32.exe: ELF 32-bit LSB executable, UCB RISC-V, version 1 (SYSV), statically linked, not stripped
$ spike --isa=RV32 pk rv64-m32.exe
<does not terminate>

For sanity checking, I tried a 64-bit exe, which works:

$ riscv64-unknown-elf-gcc -S -o switch-gcc64-m64.s switch.c
$ riscv64-unknown-elf-gcc -c -o switch-gcc64-m64.o switch-gcc64-m64.s
$ riscv64-unknown-elf-gcc -o rv64-m64.exe switch-gcc64-m64.o
$ spike pk rv64-m64.exe
<terminates after approx 370K instructions>

aswaterman, 2016-07-12:
Did you build a 32-bit pk?

pmundkur, 2016-07-12:
I guess not.

$ file ../../install/riscv64-unknown-elf/bin/pk
../../install/riscv64-unknown-elf/bin/pk: ELF 64-bit LSB executable, UCB RISC-V, version 1 (SYSV), statically linked, not stripped

What is the configure magic to build 32-bit?  There's nothing in the
pk readme.

aswaterman, 2016-07-12:
Build a 32-bit compiler, then change the --host string used when you configure pk.  Or, if you don't want to build a 32-bit compiler, manually add -m32 to the CFLAGS and LDFLAGS in the pk's generated Makefile.

pmundkur, 2016-07-12:
I manually added -m32 to CFLAGS and LDFLAGS to the generated Makefile.
I get an internal compiler error:

These are the last lines of the build log:

a - snprintf.o
a - string.o
riscv64-unknown-elf-ranlib libutil.a
riscv64-unknown-elf-gcc -MMD -MP -m32 -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks  -DBBL_PAYLOAD=\"dummy_payload\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../pk/pk.c
../pk/pk.c: In function 'rest_of_boot_loader':
../pk/pk.c:174:1: error: unrecognizable insn:
 }
  ^
  (insn 489 117 503 11 (set (mem/c:SI (const:SI (plus:SI (symbol_ref:SI ("current") [flags 0x80] <var_decl 0x7f7830d83f30 current>)
                      (const_int 48 [0x30]))) [1 current.t0+0 S4 A32])
		              (reg:SI 389 [ uarch_counters_enabled_lsm.41 ])) ../pk/pk.c:52 -1
			           (nil))
../pk/pk.c:174:1: internal compiler error: in extract_insn, at recog.c:2287
0x8f87c3 _fatal_insn(char const*, rtx_def const*, char const*, int, char const*)
    /home/mundkur/src/hw/riscv/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/gcc/rtl-error.c:108
0x8f87f9 _fatal_insn_not_found(rtx_def const*, char const*, int, char const*)
    /home/mundkur/src/hw/riscv/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/gcc/rtl-error.c:116
0x8cd3e7 extract_insn(rtx_insn*)
    /home/mundkur/src/hw/riscv/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/gcc/recog.c:2287
0xd08779 decompose_multiword_subregs
    /home/mundkur/src/hw/riscv/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/gcc/lower-subreg.c:1466
0xd0992d execute
    /home/mundkur/src/hw/riscv/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/gcc/lower-subreg.c:1736
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <http://gcc.gnu.org/bugs.html> for instructions.
Makefile:300: recipe for target 'pk.o' failed
make: *** [pk.o] Error 1

aswaterman, 2016-07-12:
Not sure what's going on in the compiler, but I pushed a work-around to riscv-pk d88671efdfd08f5fcb33b4c4099d79eb82978c62

pmundkur, 2016-07-12:
(I really appreciate the rapid responses, Andrew. Thanks!)

That helped, I now get spike terminating, but not in the expected way:

spike -l --isa=RV32 ../../install/riscv32-unknown-elf/bin/pk  cmpi-m32.exe
...
core   0: 0xffffffff8000324c (0x00042703) lw      a4, 0(s0)
core   0: 0xffffffff80003250 (0x00442783) lw      a5, 4(s0)
core   0: 0xffffffff80003254 (0x00f766b3) or      a3, a4, a5
core   0: 0xffffffff80003258 (0xfe068ae3) beqz    a3, pc - 12
core   0: 0xffffffff8000324c (0x00042703) lw      a4, 0(s0)
terminate called after throwing an instance of 'trap_load_access_fault'

This occurs after approx 115385 instructions (using wc -l on the log file).

aswaterman, 2016-07-12:
np, I think this should be fixed if you update riscv-isa-sim and riscv-pk again.

pmundkur, 2016-07-12:
Excellent, spike terminates normally now, thanks a lot!

closed 2016-07-12 by pmundkur