riscv-collab/riscv-gnu-toolchain #165: ABI failure with RV32I

vanjoe, 2016-08-04:
Hi, I just built a standard build, I ran the following from the riscv-tools repository (priv-v1.9):
```
$ git submodule update --init --recursive
$ export RISCV=/path/to/install/riscv/toolchain
$ ./build.sh
```

then with the following commands
```
riscv64-unknown-elf-gcc -m32  -march=RV32I -msoft-float  -O2 -MD -Wall -std=gnu99 -c main.c -o obj/main.c.o
riscv64-unknown-elf-gcc -m32  -march=RV32I -msoft-float  -O2 -MD -Wall -std=gnu99 -c printf.c -o obj/printf.c.o
riscv64-unknown-elf-gcc -m32  -march=RV32I -msoft-float  -O2 -MD -Wall -std=gnu99 -c crt.S -o obj/crt.S.o
riscv64-unknown-elf-gcc -T link.ld obj/crt.S.o  obj/main.c.o obj/printf.c.o  -o test.elf -m32 -march=RV32I -msoft-float -static -nostartfiles
```
I get the following error:
```
/nfs/scratch/joel/riscv-tools/lib/gcc/riscv64-unknown-elf/6.1.0/../../../../riscv64-unknown-elf/bin/ld: /nfs/scratch/joel/riscv-tools/lib/gcc/riscv64-unknown-elf/6.1.0/libgcc.a(div.o): ABI is incompatible with that of the selected emulation
/nfs/scratch/joel/riscv-tools/lib/gcc/riscv64-unknown-elf/6.1.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /nfs/scratch/joel/riscv-tools/lib/gcc/riscv64-unknown-elf/6.1.0/libgcc.a(div.o)
obj/printf.c.o: In function `.L3':
printf.c:(.text+0x54): undefined reference to `__mulsi3'
collect2: error: ld returned 1 exit status
make: *** [test.elf] Error 1
```

When I compile my project with `-march=RV32IM` I don't get the error. In the past on a toolchain built a while (~6 months) ago I didn't have this issue. I also did not have to use the -msoft-float flag either.

I'm unclear what is going on here. Could someone please enlighten me?

vanjoe, 2016-08-04:
Ok, that will fix my problem I'm sure. I'm still curious Why it built correctly with `-march=RV32IM` but not with `-march=RV32I` Shouldn't  they both be equally incompatible with libgcc built with RV64G? Or does ld not look at libgcc if it doesn't need the __mulsi3 function?

aswaterman, 2016-08-04:
Basically, the latter.  ld doesn't inspect libgcc.a because after linking all the object code there are no unresolved dependences.

closed 2016-08-04 by aswaterman