riscv-collab/riscv-gnu-toolchain #171: ".align 4" always expands to three nops

neuschaefer, 2016-08-13:
Test case:

* Create `align.s` with the following content:
```
/* add or remove instructions to change the number of required nops */
li a0, 5
li a0, 5

.align 4
```
* Run `riscv64-unknown-elf-gcc -c align.s && riscv64-unknown-elf-objdump -d align.o` to check the assembler's work

With commit riscv/riscv-tools@d34419ff390 (and older versions) I observe that `.align 4` always expands to three nops.

palmer-dabbelt, 2016-08-14:
That's a feature. not a bug!

I actually just kind of guessed here, but I figured I would have noticed since I've used .align on RISC-V.  I hate regressions, though, so I figured it'd be worth looking at.  Here's my notes:

````
palmer.dabbelt a2 test $ cat test.s
.text

align_4_0:
        .align 6
        .align 4
        addi x0,x0,32

align_4_1:
        .align 6
        addi x0,x0,1
        .align 4
        addi x0,x0,32

align_4_2:
        .align 6
        addi x0,x0,1
        addi x0,x0,2
        addi x0,x0,3
        .align 4
        addi x0,x0,32

align_4_3:
        .align 6
        addi x0,x0,1
        addi x0,x0,2
        addi x0,x0,3
        .align 4
        addi x0,x0,32
````

without linking I get the surprising result you found

````
palmer.dabbelt a2 test $ riscv64-unknown-linux-gnu-gcc test.s -c -o test.o
palmer.dabbelt a2 test $ riscv64-unknown-linux-gnu-objdump -d test.o

test.o:     file format elf64-littleriscv

Disassembly of section .text:

0000000000000000 <align_4_0>:
   0:   00000013                nop
   4:   00000013                nop
   8:   00000013                nop
   c:   02000013                li      zero,32

0000000000000010 <align_4_1>:
  10:   00100013                li      zero,1
  14:   00000013                nop
  18:   00000013                nop
  1c:   00000013                nop
  20:   02000013                li      zero,32

0000000000000024 <align_4_2>:
  24:   00100013                li      zero,1
  28:   00200013                li      zero,2
  2c:   00300013                li      zero,3
  30:   00000013                nop
  34:   00000013                nop
  38:   00000013                nop
  3c:   02000013                li      zero,32

0000000000000040 <align_4_3>:
  40:   00100013                li      zero,1
[... 197 lines cut ...]
  if (rel->r_addend < nop_bytes)
    return FALSE;

  /* Delete the reloc.  */
  rel->r_info = ELFNN_R_INFO (0, R_RISCV_NONE);

  /* If the number of NOPs is already correct, there's nothing to do.  */
  if (nop_bytes == rel->r_addend)
    return TRUE;

  /* Write as many RISC-V NOPs as we need.  */
  for (pos = 0; pos < (nop_bytes & -4); pos += 4)
    bfd_put_32 (abfd, RISCV_NOP, contents + rel->r_offset + pos);

  /* Write a final RVC NOP if need be.  */
  if (nop_bytes % 4 != 0)
    bfd_put_16 (abfd, RVC_NOP, contents + rel->r_offset + pos);

  /* Delete the excess bytes.  */
  return riscv_relax_delete_bytes (abfd, sec, rel->r_offset + nop_bytes,
                                   rel->r_addend - nop_bytes);
}
````

which seems to be doing what we're expected.

neuschaefer, 2016-08-14:
Thanks for your very detailed analysis! Next time I find such a quirk, I'll check the linked binary.

closed 2016-08-15 by palmer-dabbelt

thomascp, 2019-04-17:
@palmer-dabbelt 

Hi Palmer,
Here is a problem.

I wrote this test codes:

.global main
        .global _start

        .text
        .balign 4

_start:
        j main

        .org 0x40

main:
        nop

First, I compiled it into .o:

a.o:     file format elf32-littleriscv

Disassembly of section .text:

00000000 <_start-0x2>:
   0:   0001                    nop

00000002 <_start>:
   2:   a83d                    j       40 <main>
        ...

00000040 <main>:
  40:   0001                    nop
        ...

Then I linked it to a.out,

"riscv32-unknown-linux-gnu-gcc -nostdlib -Ttext=0x0 a.s"

a.out:     file format elf32-littleriscv

Disassembly of section .text:

00000000 <_start>:
   0:   a83d                    j       3e <main>
        ...

0000003e <main>:
  3e:   0001                    nop
        ...

This is not what we want, right? We want the function main located at 0x40.
But because compiler first inserts extra nops linker remove it, function main moved to 0x3e.
Is this a bug? How to solve it? Thanks!

BR,
Peng