riscv-collab/riscv-gnu-toolchain #173: Bug somewhere in R_RISCV_PCREL_LO1 handling (RV32I)

cliffordwolf, 2016-08-16:
Hi guys!

I'm updating the riscv-gnu-toolchain used for PicoRV32 and I think I found a bug.

Consider the following test case (`demo.S`), derived from `auipc.S` from riscv-tests:

```
.text
.align 3
.global _start
_start:
lla a0, 1f + 10000
jal a1, 1f
1: sub a0, a0, a1
j _start
```

In `/opt/rv32i` I have an old installation of git rev 13f52d2 (2016-05-31) of riscv-gnu-toolchain. In `/opt/riscv32i` I have a new installation of todays git head (0068461). Both are configured with `--with-arch=RV32I`.

Creating object files:

```
$ /opt/rv32i/bin/riscv32-unknown-elf-gcc -c -o demo_old.o demo.S
$ /opt/riscv32i/bin/riscv32-unknown-elf-gcc -c -o demo_new.o demo.S
$ md5sum demo_old.o demo_new.o
abe4ccc6992a2d95e6430a15765ea019  demo_old.o
abe4ccc6992a2d95e6430a15765ea019  demo_new.o
```

So the object files are identical. Now linking:

```
$ /opt/rv32i/bin/riscv32-unknown-elf-gcc -nostdlib -o demo_old.elf demo_old.o
$ /opt/riscv32i/bin/riscv32-unknown-elf-gcc -nostdlib -o demo_new.elf demo_new.o
```

Diffing the output of `riscv32-unknown-elf-objdump -d demo_old.elf` and `riscv32-unknown-elf-objdump -d demo_new.elf`:

```
$ riscv32-unknown-elf-objdump -d demo_old.elf > demo_old.txt
$ riscv32-unknown-elf-objdump -d demo_new.elf > demo_new.txt
$ diff -u demo_old.txt demo_new.txt
--- demo_old.txt	2016-08-16 23:53:20.727397820 +0200
+++ demo_new.txt	2016-08-16 23:53:26.919398025 +0200
@@ -1,12 +1,12 @@
 
-demo_old.elf:     file format elf32-littleriscv
+demo_new.elf:     file format elf32-littleriscv
 
 
 Disassembly of section .text:
 
 00010000 <_ftext>:
    10000:	00002517          	auipc	a0,0x2
-   10004:	71c50513          	addi	a0,a0,1820 # 1271c <_gp+0x1efc>
+   10004:	00050513          	mv	a0,a0
    10008:	004005ef          	jal	a1,1000c <_ftext+0xc>
    1000c:	40b50533          	sub	a0,a0,a1
    10010:	ff1ff06f          	j	10000 <_ftext>
```

So for some reason the addi insn at 10004 does not get patched with an offset with the new toolchain version. This causes the `auipc.S` test case to fail in the PicoRV32 test bench when using the new toolchain.

Looking at the disassembly and relocation table in the (identical) object files:

```
$ riscv32-unknown-elf-readelf -r demo_new.o

Relocation section '.rela.text' at offset 0xe8 contains 5 entries:
 Offset     Info    Type            Sym.Value  Sym. Name + Addend
00000000  0000002b R_RISCV_ALIGN                4
00000004  00000417 R_RISCV_PCREL_HI2 00000010   .L1^B1 + 2710
00000008  00000518 R_RISCV_PCREL_LO1 00000004   .L0  + 0
0000000c  00000411 R_RISCV_JAL       00000010   .L1^B1 + 0
00000014  00000611 R_RISCV_JAL       00000004   _start + 0
```

```
$ riscv32-unknown-elf-objdump -d demo_new.o

demo_new.o:     file format elf32-littleriscv

Disassembly of section .text:

00000000 <_start-0x4>:
   0:	00000013          	nop

00000004 <_start>:
   4:	00000517          	auipc	a0,0x0
   8:	00050513          	mv	a0,a0
   c:	004005ef          	jal	a1,10 <.L1^B1>

00000010 <.L1^B1>:
  10:	40b50533          	sub	a0,a0,a1
  14:	ff1ff06f          	j	4 <_start>
```

To be honest, I don't know what the exact semantics of the `.rela.text` entries is, so I don't know if the `R_RISCV_PCREL_LO1` entry in `.rela.text` is incorrect but previous versions of the linker only used the `R_RISCV_PCREL_HI2` entry, or if the entry is correct and something broke in the linker.

aswaterman, 2016-08-16:
Hey Cliff, I think we fixed this slightly after you encountered it... it was a porting bug when upgrading binutils.  Pull from master and the problem should go away.

cliffordwolf, 2016-08-16:
What a timing! You must have pushed that minutes after I bumped the git rev in my scripts.. :)

Everything is working fine now with 7e48594. Thanks!

closed 2016-08-16 by cliffordwolf