riscv-collab/riscv-gnu-toolchain #174: Installing riscv toolchain

susi655, 2016-08-30:
when i added "configure --with-xlen=32" to get 32-bit libraries, i am getting following error while installing tool chain

Configuring project riscv-pk
Building project riscv-pk
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small
make: *** [mtrap.o] Error 1

palmer-dabbelt, 2016-08-30:
These look like the Intel memory models -- you can verify this by looking at the configure log, which should be unable to find "riscv64-unknown-gnu-gcc" and will have fallen back to just "gcc".  This is an unfortunate autoconf behavior that causes lots of problematic setups to end up producing arcane error messages.

I added a "--with-xlen" argument to riscv-tools' build.sh.  Can you verify it actually works?  All I did was test that it builds.

https://github.com/riscv/riscv-tools/pull/39

susi655, 2016-08-31:
thanq palmer-dabbelt. But some new problems.

Here is a long story. Please bear with me.

I changed the build.sh according to that mentioned in the post "riscv/riscv-tools#39" 

First I installed 64 bit tool-chain using with_xlen=64, then in the same target file i installed 32-bit tool-chain using with_xlen=32. both got successfully installed.
I tested 64-bit with spike it was success. now coming to test of 32-bit toolchain, I ran the following 

$riscv32-unknown-elf-gcc -o hello hello.c
$spike pk test
It's there no response....? waiting .... no use.
 
Then i tried installing 32-bit tool-chain separately without installing 64-bit tool-chain.
following error came...

Configuring project riscv-tests
configure: WARNING: using cross tools not prefixed with host triplet
Building project riscv-tests
make[1]: riscv64-unknown-elf-gcc: Command not found
make[1]: *** [median_main.o] Error 127
... 

So riscv-tests is dependent on 64 bit tool chain. can we change this to 32-bit? 
It failed in riscv-tests.
The order in build.sh is
riscv-fesvr
riscv-isa-sim
riscv-gnu-toolchain
riscv-pk
riscv-tests

then i checked working of spike

$riscv32-unknown-elf-gcc -o hello hello.c
$spike pk hello
terminate called after throwing an instance of 'std::runtime_error'
  what():  could not open pk

susi655, 2016-09-02:
I didn't get any solution.......

palmer-dabbelt, 2016-09-08:
@susi655 It's best if you:

* Avoid having two github issues that report the same bug
* Only report one bug per github issue

Since your new comments in this issue seem unrelated to the original bug report and there's already another issue open that addresses these problems, I'm closing this issue.  See this

  https://github.com/riscv/riscv-pk/issues/35

instead.

closed 2016-09-08 by palmer-dabbelt

ipelagic, 2019-02-21:
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
I met same problem on Ubuntu and solved it by changing default shell (dash to bash).

maheshharidasan2013, 2019-07-05:
@ipelagic
how did you change the dash to bash

jim-wilson, 2019-07-05:
@maheshharidasan2013 changing /bin/sh from dash to bash should not be necessary, I can build riscv-gnu-toolchain on Ubuntu 18 which has dash as /bin/sh

I don't know about riscv-tools, but this isn't the riscv-tools, and riscv-tools doesn't have a compiler anymore

Most of the problems described here are from people accidentally passing RISC-V compiler options to an x86 compiler.

maheshharidasan2013, 2019-07-06:
@jim-wilson 
Sir i am trying to install riscv-tools  based on the instructions in the following link @https://sywtwaosir.wordpress.com/2018/02/01/lesson-0-understanding-the-risc-v-ecosystem-booting-your-first-kernel/
In one step ./build.sh I end up with the above mentioned errors and so far i haven't got a clue as to why this problem happening .Any way to surmount it?

maheshharidasan2013, 2019-07-06:
@jim-wilson 
This is my error
`Configuring project riscv-pk
Building project riscv-pk
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small; did you mean ‘medium’?
make: *** [file.o] Error 1
`

jim-wilson, 2019-07-07:
The instructions you are following are obsolete.

riscv-tools no longer includes a toolchain.  You must build a riscv toolchain before you can build riscv-tools.  The riscv-toolchain is riscv/riscv-gnu-toolchain and has always been riscv/riscv-gnu-toolchain.  riscv-tools contains misc tools primarily for hardware developers.  If you do need or want riscv-tools, then after you have a riscv64-unknown-elf toolchain, put it on your path, and riscv-tools should build.

You can also download a toolchain if you want.  sifive.com/boards has pre-built toolchains freedom studio (eclipse based) and GNU embedded toolchain.

I see another problem with your docs.  It tells you to build freedom-u-sdk and run "make sim" to boot.  That should be "make qemu" instead.  spike is a simulator for hardware testing and has limited features.  qemu is a simulator for software testing and has many features.  So make qemu is far more useful than make spike.  make spike is primarily useful for hardware developers as it is the golden reference for the hardware.  make sim is not supported with current freedom-u-sdk.  If you do want make sim to work, check out the legacy v1_0 branch of freedom-u-sdk.

ipelagic, 2019-07-08:
You can try following command in ubuntu to change shell to bash before
installing riscv-gnu-toolchain
"*sudo dpkg-reconfigure dash" *
Hope that works.

chyhsueh, 2019-07-22:
Hi, buddies:

It seems the symptoms found in case the expected toolchain with prefix "riscv64-unknown-elf-" cannot be found in the host system and hence the gcc on host was applied as fallback toolchain. Currently the riscv-gnu-toolchain must be built/ready before the riscv-tools could be built. Please grab the source code of riscv-gnu-toolchain from the following repo: 

https://github.com/riscv/riscv-gnu-toolchain

The expected toolchain will be built by:

```
./configure --prefix=$RISCV
make
```

Please following the section "Installation (Newlib)" in Readme.md. Once the riscv-gnu-toolchain was successfully setup in $RISCV and the directory $RISCV\bin was set in $PATH, the riscv-tools can be successfully built then.