riscv-collab/riscv-gnu-toolchain #176: building linux toolchain (libc) for 32 bit fails with 'cannot find -lc'

jhotl, 2016-09-03:
with the master checkout of the riscv-gnu-toolchain I get a failure when building for 32bit without floating point

I use the option:
'''
./configure --prefix=/opt/riscv/rv32ima-linux-master/ --disable-float --with-xlen=32
make linux
'''
This results in:
'''
/opt/riscv/riscv-rv32i-linux-master/riscv32-unknown-linux-gnu/bin/ld: cannot find -lc
collect2: error: ld returned 1 exit status
Makefile:945: recipe for target 'libgcc_s.so' failed
make[3]: *** [libgcc_s.so] Error 1
make[3]: Leaving directory '/opt/riscv/build/riscv-rv32i-linux-master/riscv-gnu-toolchain/build-gcc-linux-stage2/riscv32-unknown-linux-gnu/libgcc'
Makefile:11402: recipe for target 'all-target-libgcc' failed
make[2]: *** [all-target-libgcc] Error 2
make[2]: Leaving directory '/opt/riscv/build/riscv-rv32i-linux-master/riscv-gnu-toolchain/build-gcc-linux-stage2'
Makefile:875: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/opt/riscv/build/riscv-rv32i-linux-master/riscv-gnu-toolchain/build-gcc-linux-stage2'
Makefile:206: recipe for target 'stamps/build-gcc-linux-stage2' failed
make: *** [stamps/build-gcc-linux-stage2] Error 2
'''
I already tried some different combinations of options. The only setting --with-xlen=32 works fine.

I will add a full log later on.

palmer-dabbelt, 2016-09-12:
This reproduces, and might be related to <https://github.com/riscv/riscv-gnu-toolchain/pull/136>.  @manuelafm did you look at fixing your patches?

manuelafm, 2016-09-12:
Sorry, I couldn't look into this yet, I was on the move in the last few months and will still be for a while.  I plan to look into if nobody beats me, but cannot promise anything still for a few weeks.

kito-cheng, 2016-09-30:
This patch fix RV32I build, it's cause by the multi-lib library search path problem.

You can try it this patch for gcc: https://github.com/kito-cheng/riscv-gcc/commit/c09f56b48e08d671ae57e7c4d992d869e2b448f7

jhotl, 2016-09-30:
With the patch the toolchain compiles indeed for the --disable-float --with-xlen=32 combination. Great!

palmer-dabbelt, 2016-10-05:
@kito-cheng Is c09f56b upstream?  If it is, then I think this bug is resolved.  If not, is there a PR I missed?

kito-cheng, 2016-10-05:
I create a PR just now :)
https://github.com/riscv/riscv-gcc/pull/4

palmer-dabbelt, 2016-10-05:
I think https://github.com/riscv/riscv-gcc/pull/4 should have fixed this bug, so I'm closing the issue.

closed 2016-10-05 by palmer-dabbelt