riscv-collab/riscv-gnu-toolchain #177: static linking (-pie): unresolvable R_RISCV_PCREL_HI20 relocation against symbol `stdin@@GLIBC_2.20'

davidlt, 2016-09-04:
All the details, code and built packages for Fedora bootstrapping are here: https://github.com/rwmjones/fedora-riscv

Fedora is building hardened packages. Details on what that is are here: https://fedoraproject.org/wiki/Changes/Harden_All_Packages

Simply put, Fedora has custom SPECs:
```
stage3:/rpmbuild/SPECS# cat /usr/lib/rpm/redhat/redhat-hardened-cc1
*cc1_options:
+ %{!r:%{!fpie:%{!fPIE:%{!fpic:%{!fPIC:%{!fno-pic:-fPIE}}}}}}

stage3:/rpmbuild/SPECS# cat /usr/lib/rpm/redhat/redhat-hardened-ld
*self_spec:
+ %{!shared:%{!r:-pie}}
```
Object files for static linking get `-fPIE`
Linking executable gets `-pie -Wl,-z,now`
Object files for shared/dynamic linking get `-fPIC`
Linking shared object gets `-shared -Wl,-z,now`

We quickly noticed that we couldn't build packages producing static binaries and later found that it was due to -pie flag. We fail to link them:
```
[..]
/usr/lib64/gcc/riscv64-unknown-linux-gnu/6.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: bzip2.o(.text+0x1f20): unresolvable R_RISCV_PCREL_HI20 relocation against symbol `stderr@@GLIBC_2.20'
/usr/lib64/gcc/riscv64-unknown-linux-gnu/6.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: bzip2.o(.text+0x1f64): unresolvable R_RISCV_PCREL_HI20 relocation against symbol `stdin@@GLIBC_2.20'
/usr/lib64/gcc/riscv64-unknown-linux-gnu/6.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: bzip2.o(.text+0x1f70): unresolvable R_RISCV_PCREL_HI20 relocation against symbol `stdout@@GLIBC_2.20'
/usr/lib64/gcc/riscv64-unknown-linux-gnu/6.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: bzip2.o(.text+0x1f9c): unresolvable R_RISCV_PCREL_HI20 relocation against symbol `stderr@@GLIBC_2.20'
/usr/lib64/gcc/riscv64-unknown-linux-gnu/6.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: bzip2.o(.text+0x200c): unresolvable R_RISCV_PCREL_HI20 relocation against symbol `stderr@@GLIBC_2.20'
/usr/lib64/gcc/riscv64-unknown-linux-gnu/6.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: bzip2.o(.text+0x20a0): unresolvable R_RISCV_PCREL_HI20 relocation against symbol `stderr@@GLIBC_2.20'
/usr/lib64/gcc/riscv64-unknown-linux-gnu/6.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: bzip2.o(.text+0x20ec): unresolvable R_RISCV_PCREL_HI20 relocation against symbol `stderr@@GLIBC_2.20'
/usr/lib64/gcc/riscv64-unknown-linux-gnu/6.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: bzip2.o(.text+0x2150): unresolvable R_RISCV_PCREL_HI20 relocation against symbol `stderr@@GLIBC_2.20'
/usr/lib64/gcc/riscv64-unknown-linux-gnu/6.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: bzip2.o(.text+0x21c0): unresolvable R_RISCV_PCREL_HI20 relocation against symbol `stderr@@GLIBC_2.20'
/usr/lib64/gcc/riscv64-unknown-linux-gnu/6.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: bzip2.o(.text+0x2214): unresolvable R_RISCV_PCREL_HI20 relocation against symbol `stderr@@GLIBC_2.20'
/usr/lib64/gcc/riscv64-unknown-linux-gnu/6.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: bzip2.o(.text+0x2260): unresolvable R_RISCV_PCREL_HI20 relocation against symbol `stderr@@GLIBC_2.20
[..]
```
Full build log: http://davidlt.web.cern.ch/davidlt/vault/riscv64_issues/bzip2_build.log
Full build directory: http://davidlt.web.cern.ch/davidlt/vault/riscv64_issues/bzip2-1.0.6.tar.gz
Specific object files:
http://davidlt.web.cern.ch/davidlt/vault/riscv64_issues/bzip2.o
http://davidlt.web.cern.ch/davidlt/vault/riscv64_issues/bzip2recover.o
http://davidlt.web.cern.ch/davidlt/vault/riscv64_issues/libbz2.a

At this point we are building packages with hardening disabled.

davidlt, 2016-09-04:
Toolchain info:
```
stage3:/rpmbuild/BUILD/gmp-6.1.1# gcc -dumpversion
6.1.0

stage3:/rpmbuild/BUILD/gmp-6.1.1# gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/riscv64-unknown-linux-gnu/6.1.0/lto-wrapper
Target: riscv64-unknown-linux-gnu
Configured with: ../configure --host=riscv64-unknown-linux-gnu --prefix=/usr --libdir=/usr/lib64 --enable-shared --enable-tls --enable-languages=c,c++ --disable-libmudflap --disable-libssp --disable-libquadmath --disable-nls --disable-multilib
Thread model: posix
gcc version 6.1.0 (GCC)

stage3:/rpmbuild/BUILD/gmp-6.1.1# ld --version
GNU ld (GNU Binutils) 2.26.0.20160516
Copyright (C) 2015 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.

stage3:/rpmbuild/BUILD/gmp-6.1.1# /lib64/libc.so.6
GNU C Library (GNU libc) stable release version 2.22, by Roland McGrath et al.
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.
Compiled by GNU CC version 5.3.0.
Available extensions:
        crypt add-on version 2.1 by Michael Glad and others
        GNU Libidn by Simon Josefsson
        Native POSIX Threads Library by Ulrich Drepper et al
        BIND-8.2.3-T5B
libc ABIs: UNIQUE
For bug reporting instructions, please see:
<http://www.gnu.org/software/libc/bugs.html>.
```

aswaterman, 2016-09-04:
How do I reproduce the problem given the tarball?

davidlt, 2016-09-04:
Assuming that issue is within linker (not GCC generating wrong instruction patterns), maybe this is enough: download `bzip2.o` and `libbz2.a` also make local `redhat-hardened-ld` and `redhat-hardened-cc1`, then re-execute from the log:
```
gcc -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -D_FILE_OFFSET_BITS=64  -Wl,-z,relro -specs=/usr/lib/rpm/redhat/redhat-hardened-ld -o bzip2 bzip2.o -L. -lbz2
```

Alternatively you can grab my last build stage3 rootfs and kernel from: http://davidlt.web.cern.ch/davidlt/riscv/20160823/

Boot it up and try to build a fresh version of bzip2 in it with similar configuration to Fedora, e.g (from log):
```
make -f Makefile-libbz2_so CC=gcc AR=ar RANLIB=ranlib 'CFLAGS=-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -D_FILE_OFFSET_BITS=64 -fpic -fPIC ' 'LDFLAGS=-Wl,-z,relro -specs=/usr/lib/rpm/redhat/redhat-hardened-ld' all
rm -f blocksort.o bzlib.o compress.o crctable.o decompress.o huffman.o randtable.o
make CC=gcc AR=ar RANLIB=ranlib 'CFLAGS=-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -D_FILE_OFFSET_BITS=64 ' 'LDFLAGS=-Wl,-z,relro -specs=/usr/lib/rpm/redhat/redhat-hardened-ld' all
```

I cal also upload a tarball with 4 patches which Fedora applies, but hopefully it fails even without those patches.

You can boot this with with QEMU like this: `qemu-system-riscv -kernel /usr/bin/bbl -append /path/to/vmlinux -drive file=stage3-disk.img,format=raw -nographic -m 2G`

sorear, 2016-09-04:
@aswaterman note that the Fedora team is using priv-1.7 versions of everything, so use a priv-1.7 qemu to boot the kernel David just linked

davidlt, 2016-09-04:
Correct, priv-1.7. I am running everything on Fedora. Info on how to setup is here: https://lists.fedoraproject.org/archives/list/devel@lists.fedoraproject.org/thread/SF36QMKHC5AOI73QLYTKAVQLFT5F3NVK/

Something like that worked for me to re-produce from scratch:

```
wget http://davidlt.web.cern.ch/davidlt/riscv/20160823/stage3-disk.img.xz
wget http://davidlt.web.cern.ch/davidlt/riscv/20160823/vmlinux
unxz stage3-disk.img.xz

cat <<EOF > redhat-hardened-cc1
*cc1_options:
+ %{!r:%{!fpie:%{!fPIE:%{!fpic:%{!fPIC:%{!fno-pic:-fPIE}}}}}}
EOF

cat <<EOF > redhat-hardened-ld
*self_spec:
+ %{!shared:%{!r:-pie}}
EOF

virt-copy-in -a stage3-disk.img redhat-hardened-cc1  redhat-hardened-ld  /

cat <<EOF > compile
#!/bin/bash

set -x

make -f Makefile-libbz2_so CC=gcc AR=ar RANLIB=ranlib CFLAGS='-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/redhat-hardened-cc1 -D_FILE_OFFSET_BITS=64 -fpic -fPIC' LDFLAGS='-Wl,-z,relro -specs=/redhat-hardened-ld' all
rm -f blocksort.o bzlib.o compress.o crctable.o decompress.o huffman.o randtable.o
make CC=gcc AR=ar RANLIB=ranlib CFLAGS='-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/redhat-hardened-cc1 -D_FILE_OFFSET_BITS=64' LDFLAGS='-Wl,-z,relro -specs=/redhat-hardened-ld' all
EOF

chmod +x compile

wget http://www.bzip.org/1.0.6/bzip2-1.0.6.tar.gz

tar xzvf bzip2-1.0.6.tar.gz

virt-copy-in -a stage3-disk.img bzip2-1.0.6  /
virt-copy-in -a stage3-disk.img compile  /bzip2-1.0.6
qemu-system-riscv -kernel /usr/bin/bbl -append vmlinux -drive file=stage3-disk.img,format=raw -nographic -m 2G

# cd /bzip2-1.0.6 && ./compile
```

aswaterman, 2016-09-06:
Thanks, I'm able to reproduce (using your first set of instructions and a cross compiler).

aswaterman, 2016-09-06:
looks like GCC is using the wrong reloc type... probably not a binutils bug.  Looking into it.

aswaterman, 2016-09-06:
Where's the log file? I'd like to see how gcc was invoked on the sub-pieces.

davidlt, 2016-09-06:
The link is in the issue. Here it is: http://davidlt.web.cern.ch/davidlt/vault/riscv64_issues/bzip2_build.log

aswaterman, 2016-09-06:
OK, I'm stuck...

In your huffman.o, I see the a the wrong kind of reloc:

    waterman@a3:~/tmp/bzip2-1.0.6$ riscv64-unknown-linux-gnu-readelf -r huffman.o|grep stack_chk_guard
    00000000000c  01f100000017 R_RISCV_PCREL_HI2 0000000000000000 __stack_chk_guard + 0
    0000000004f4  01f100000017 R_RISCV_PCREL_HI2 0000000000000000 __stack_chk_guard + 0

When I recompile it (using commit 610de0f1e6f11cc3a963c10e7dafcdb51dbbf059), I get the correct reloc:

    waterman@i0:~/tmp/bzip2-1.0.6$ /scratch/waterman/tmp/bin/riscv64-unknown-linux-gnu-gcc -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=./redhat-hardened-cc1 -D_FILE_OFFSET_BITS=64 -fpic -fPIC  -c huffman.c
    waterman@i0:~/tmp/bzip2-1.0.6$ riscv64-unknown-linux-gnu-readelf -r huffman.o|grep stack_chk_guard
    00000000000c  01f800000014 R_RISCV_GOT_HI20  0000000000000000 __stack_chk_guard + 0

My redhat-hardened-cc1 contains:

    *cc1_options:
    + %{!r:%{!fpie:%{!fPIE:%{!fpic:%{!fPIC:%{!fno-pic:-fPIE}}}}}}

Not sure how to proceed since I can't replicate the problem... can you spot anything I'm doing wrong?

davidlt, 2016-09-06:
We are using 728afcddcb0526a0f6560c4032da82805f054d58
```
commit 728afcddcb0526a0f6560c4032da82805f054d58
Merge: 1e71c88 b0e194b
Author: Andrew Waterman <waterman@eecs.berkeley.edu>
Date:   Tue Jan 12 14:43:44 2016 -0800

    Merge pull request #114 from sbeamer/patch-1

    clarify os x requirements
```

This might have been fixed already in master. We need to discuss why we are using 728afcddc

davidlt, 2016-09-06:
Is it safe for us to bump our toolchain commit to the tip of master if we are using priv-1.7? Or what is the last commit working with priv-1.7?

aswaterman, 2016-09-06:
I think so, but I can't say for sure.  The ABI hasn't changed, so it might just work to compile the userland with master.

davidlt, 2016-09-06:
Did you have any instruction changes between priv-1.7 to priv-1.9? That's the main concern.

aswaterman, 2016-09-06:
Not at the user level, but you'll need to use priv-1.7 to build the priv-1.7 kernel.

davidlt, 2016-09-08:
After looking again at what version of GCC and Binutils we have in stage3 image:
```
BINUTILS_X_VERSION = 2.26
GCC_X_VERSION      = 6.1.0
https://github.com/riscv/riscv-gcc/archive/riscv-gcc-$(GCC_X_VERSION).tar.gz
https://github.com/riscv/riscv-binutils-gdb/archive/riscv-binutils-$(BINUTILS_X_VERSION).tar.gz
```

Basically we are using the tip of those branches. So we have the latest GCC, but previous version of Binutils.

rwmjones, 2016-09-08:
I'm afraid I wasn't able to reproduce this issue in our stage3.  However I updated binutils to 2.27 anyway (refer to previous comment).

davidlt, 2016-09-08:
Took fresh stage3 image, installed dependencies:
```
rpm -Uvh --nodeps /rpmbuild/RPMS/noarch/{automake,autoconf,perl}*.rpm /rpmbuild/RPMS/riscv64/{bash,make,perl,dwz}*.rpm
```
Restored `/usr/lib/rpm/redhat/macros` and compiled bzip2.
Failed:
```
gcc -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -D_FILE_OFFSET_BITS=64  -Wl,-z,relro -specs=/usr/lib/rpm/redhat/redh
at-hardened-ld -o bzip2 bzip2.o -L. -lbz2
/usr/lib64/gcc/riscv64-unknown-linux-gnu/6.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: ./libbz2.a(blocksort.o)(.text+0x4): unresolvable R_RISCV_PCREL_HI20 relocation against symbol `__stack_chk_guard@@GLIBC_2.20'
/usr/lib64/gcc/riscv64-unknown-linux-gnu/6.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: ./libbz2.a(blocksort.o)(.text+0x78): unresolvable R_RISCV_PCREL_HI20 relocation against symbol `stderr@@GLIBC_2.20'
/usr/lib64/gcc/riscv64-unknown-linux-gnu/6.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: ./libbz2.a(blocksort.o)(.text+0x2b8): unresolvable R_RISCV_PCREL_HI20 relocation against symbol `stderr@@GLIBC_2.20'
/usr/lib64/gcc/riscv64-unknown-linux-gnu/6.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: ./libbz2.a(blocksort.o)(.text+0x8f0): unresolvable R_RISCV_PCREL_HI20 relocation against symbol `stderr@@GLIBC_2.20'
/usr/lib64/gcc/riscv64-unknown-linux-gnu/6.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: ./libbz2.a(blocksort.o)(.text+0x934): unresolvable R_RISCV_PCREL_HI20 relocation against symbol `stderr@@GLIBC_2.20'
```

Will check with Binutils 2.27.

davidlt, 2016-09-09:
Binutils 2.27 did not change the outcome.

davidlt, 2016-09-09:
I also tried our cross-compiler (GCC 5.3.0) and yes, it did produce different result. Same as what you have reported.

davidlt, 2016-09-09:
@aswaterman I think, you made a wrong comparison. huffman.o was built for static linking, while your command is with -fPIC for shared linking.

aswaterman, 2016-09-09:
good catch... that certainly seems relevant.

davidlt, 2016-09-09:
I changed my `compile` to script to this:
```
#!/bin/bash

set -x

cat <<EOF > /redhat-hardened-cc1
*cc1_options:
+ %{!r:%{!fpie:%{!fPIE:%{!fpic:%{!fPIC:%{!fno-pic:-fPIE}}}}}}
EOF

cat <<EOF > /redhat-hardened-ld
*self_spec:
+ %{!shared:%{!r:-pie}}
EOF

gcc -save-temps -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/redhat-hardened-cc1 -D_FILE_OFFSET_BITS=64  -c blocksort.c
gcc -save-temps -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/redhat-hardened-cc1 -D_FILE_OFFSET_BITS=64  -c huffman.c
gcc -save-temps -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/redhat-hardened-cc1 -D_FILE_OFFSET_BITS=64  -c crctable.c
gcc -save-temps -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/redhat-hardened-cc1 -D_FILE_OFFSET_BITS=64  -c randtable.c
gcc -save-temps -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/redhat-hardened-cc1 -D_FILE_OFFSET_BITS=64  -c compress.c
gcc -save-temps -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/redhat-hardened-cc1 -D_FILE_OFFSET_BITS=64  -c decompress.c
gcc -save-temps -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/redhat-hardened-cc1 -D_FILE_OFFSET_BITS=64  -c bzlib.c
rm -f libbz2.a
ar cq libbz2.a blocksort.o huffman.o crctable.o randtable.o compress.o decompress.o bzlib.o
ranlib libbz2.a
gcc -save-temps  -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/redhat-hardened-cc1 -D_FILE_OFFSET_BITS=64  -c bzip2.c
gcc -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/redhat-hardened-cc1 -D_FILE_OFFSET_BITS=64  -Wl,-z,relro -specs=/redhat-hardened-ld -o bzip2 bzip2.o -L. -lbz2
```
It only builds the static part which fails and also generated all temporaries (pre-processed sources, assembly).

aswaterman, 2016-09-09:
I can replicate it... I appreciate the hand-holding :)

aswaterman, 2016-09-09:
OK... I think I've figured it out.  redhat-hardened-cc1 is causing cc1 to be passed -fPIE, but for whatever reason, AS is not being passed -fpic.  So the code is being assembled wrong.  (Specifically, the "la" macro has different behavior for -fpic and not.)

I am not sure if this is a bug in redhat-hardened-cc1 or a bug in our GCC port.  Can you figure out which?  I'll fix the GCC problem if need be.

davidlt, 2016-09-10:
This works on other architectures (aarch64, x86_64, i*86, ppc64, ppc64le, <whatever else Fedora is built for>).

This is how x86_64 looks like. I don't see any issues with `redhat-hardened-cc1`, it did what was expected, i.e. adding `-fPIE` to options. Btw, GNU AS does not have such option as `-fpic`.
```
 /usr/libexec/gcc/x86_64-redhat-linux/6.1.1/cc1 -quiet -v -D _FILE_OFFSET_BITS=64 -D_FORTIFY_SOURCE=2 huffman.c -quiet -dumpbase huffman.c -mtune=generic -march=x86-64 -auxbase-strip huffman.o -g -grecord-gcc-switches -O2 -Wall -Werror=format-security -version -fexceptions -fstack-protector-strong --param ssp-buffer-size=4 -fPIE -o - |
 as -v --64 -o huffman.o
```

aswaterman, 2016-09-10:
Ok, so, can you propose a fix?

davidlt, 2016-09-10:
No idea. You can also just replace `-specs=/redhat-hardened-cc1` with simply `-fPIE`.

The only different between GCC 5.3.0 and GCC 6.1.1 is that GCC 5.3.0 generates one relocation less.

GCC 6.1.1
```
$ riscv64-unknown-linux-gnu-readelf --wide -r huffman.o|grep stack_chk_guard
000000000000000c  000001f100000017 R_RISCV_PCREL_HI20     0000000000000000 __stack_chk_guard + 0
00000000000004f4  000001f100000017 R_RISCV_PCREL_HI20     0000000000000000 __stack_chk_guard + 0
```

GCC 5.3.0
```
$ riscv64-unknown-linux-gnu-readelf --wide -r huffman.o|grep stack_chk_guard
0000000000000010  000001f100000017 R_RISCV_PCREL_HI20     0000000000000000 __stack_chk_guard + 0
```

One should understand why these messages are printed (`unresolvable R_RISCV_PCREL_HI20 relocation against symbol stderr@@GLIBC_2.20`) and then if this is linker or code-gen issue.

aswaterman, 2016-09-12:
No... replacing -specs=/redhat-hardened-cc1 with -fPIE makes the problem go away.  If I pass -fPIE, GAS is correctly invoked with -fpic.  If I pass -specs=/redhat-hardened-cc1, GAS is incorrectly invoked without -fpic.

aswaterman, 2016-09-12:
The problem is the way that RISC-V GCC determines how to pass -fpic to GAS.  You actually have to write -fPIE in the command line for that to happen.  I'll see if I can fix that.

aswaterman, 2016-09-12:
OK, this is fixed via https://github.com/riscv/riscv-binutils-gdb/commit/8105c434160681f80c48bd5e630eca7c90984691 and https://github.com/riscv/riscv-gcc/commit/e90d8dc6f429856510a51371d21dacd97ef68d41

closed 2016-09-12 by aswaterman