riscv-collab/riscv-gnu-toolchain #178: gmp: relocation truncated to fit: R_RISCV_GPREL_I against `gmp_default_alloc'

davidlt, 2016-09-04:
All the details, code and built packages for Fedora bootstrapping are here: https://github.com/rwmjones/fedora-riscv

While we can cross-compile GMP, we cannot compile it natively out-of-the-box.

We hit again another relocation issue:
```
[..]
config.status: linking ../mpn/generic/gmp-mparam.h to gmp-mparam.h
config.status: executing libtool commands
configure: summary of build options:

  Version:           GNU MP 6.1.1
  Host type:         riscv64-unknown-linux-gnu
  ABI:               standard
  Install prefix:    /usr
  Compiler:          gcc
  Static libraries:  yes
  Shared libraries:  yes

+ sed -e 's|^hardcode_libdir_flag_spec=.*|hardcode_libdir_flag_spec=""|g' -e 's|^runpath_var=LD_RUN_PATH|runpath_var=DIE_RPATH_DIE|g' -e 's|-lstdc++ -lm|-lstdc++|' -i libtool
++ pwd
+ export LD_LIBRARY_PATH=//rpmbuild/BUILD/gmp-6.1.1/base/.libs
+ LD_LIBRARY_PATH=//rpmbuild/BUILD/gmp-6.1.1/base/.libs
+ make 'CFLAGS=-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches '
gcc `test -f 'gen-fac.c' || echo '../'`gen-fac.c -o gen-fac
./gen-fac 64 0 >fac_table.h || (rm -f fac_table.h; exit 1)
gcc `test -f 'gen-fib.c' || echo '../'`gen-fib.c -o gen-fib
./gen-fib header 64 0 >fib_table.h || (rm -f fib_table.h; exit 1)
./gen-fib table 64 0 >mpn/fib_table.c || (rm -f mpn/fib_table.c; exit 1)
gcc `test -f 'gen-bases.c' || echo '../'`gen-bases.c -o gen-bases -lm
/tmp/ccjIHVt2.o: In function `.L14':
/gen-bases.c:(.text+0x1b8): relocation truncated to fit: R_RISCV_GPREL_I against `gmp_default_alloc'
/usr/lib64/gcc/riscv64-unknown-linux-gnu/6.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: final link failed: Nonrepresentable section on output
collect2: error: ld returned 1 exit status
Makefile:1428: recipe for target 'gen-bases' failed
make: *** [gen-bases] Error 1
[..]

stage3:/rpmbuild/BUILD/gmp-6.1.1# gcc gen-bases.c -o gen-bases -lm
/tmp/ccgjqZAu.o: In function `.L14':
gen-bases.c:(.text+0x1b8): relocation truncated to fit: R_RISCV_GPREL_I against `gmp_default_alloc'
/usr/lib64/gcc/riscv64-unknown-linux-gnu/6.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: final link failed: Nonrepresentable section on output
collect2: error: ld returned 1 exit status
```

Full build directory: http://davidlt.web.cern.ch/davidlt/vault/riscv64_issues/gmp-6.1.1.tar.gz
Selected files (if you don't want to download tarball):
http://davidlt.web.cern.ch/davidlt/vault/riscv64_issues/gen-bases.c
http://davidlt.web.cern.ch/davidlt/vault/riscv64_issues/gen-bases.i
http://davidlt.web.cern.ch/davidlt/vault/riscv64_issues/gen-bases.o
http://davidlt.web.cern.ch/davidlt/vault/riscv64_issues/gen-bases.s

Toolchain information:
```
stage3:/rpmbuild/BUILD/gmp-6.1.1# gcc -dumpversion
6.1.0

stage3:/rpmbuild/BUILD/gmp-6.1.1# gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/riscv64-unknown-linux-gnu/6.1.0/lto-wrapper
Target: riscv64-unknown-linux-gnu
Configured with: ../configure --host=riscv64-unknown-linux-gnu --prefix=/usr --libdir=/usr/lib64 --enable-shared --enable-tls --enable-languages=c,c++ --disable-libmudflap --disable-libssp --disable-libquadmath --disable-nls --disable-multilib
Thread model: posix
gcc version 6.1.0 (GCC)

stage3:/rpmbuild/BUILD/gmp-6.1.1# ld --version
GNU ld (GNU Binutils) 2.26.0.20160516
Copyright (C) 2015 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.

stage3:/rpmbuild/BUILD/gmp-6.1.1# /lib64/libc.so.6
GNU C Library (GNU libc) stable release version 2.22, by Roland McGrath et al.
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.
Compiled by GNU CC version 5.3.0.
Available extensions:
        crypt add-on version 2.1 by Michael Glad and others
        GNU Libidn by Simon Josefsson
        Native POSIX Threads Library by Ulrich Drepper et al
        BIND-8.2.3-T5B
libc ABIs: UNIQUE
For bug reporting instructions, please see:
<http://www.gnu.org/software/libc/bugs.html>.
```

sorear, 2016-09-05:
FWIW, this is almost certainly a linker relaxation issue, compiling with `-Wl,--no-relax` works around it; and it can be reproduced in my docker container from last week.  It can also be reproduced in a _cross_ compiler (host=x86_64 target=riscv64).

aswaterman, 2016-09-05:
Thanks, Stefan.

sorear, 2016-09-05:
(I'm still on this)

sorear, 2016-09-05:
Going to stop as I'm not making much progress.  All I can tell is that it's sometimes making relocations that are _almost_ in range.  The most interesting thing I've found is that in gdb during the call to _bfd_riscv_relax_lui, riscv_global_pointer_value returns 0x11740, but in the final executable it's 0x11778.

aswaterman, 2016-09-06:
This no longer happens in binutils 2.27.  It doesn't look like it's because of a change we made, so it's not obvious if the problem still exists and is just being hidden, or if it is actually fixed.  As a precaution, I made a patch to disable the gp relaxation for relocations against code segments, under the theory that relaxations are fighting with each other, causing the gp reloc to go out of range.  If you want to try cherry-picking the commit, feel free... https://github.com/riscv/riscv-binutils-gdb/commit/1b6393112371fe67957e239896ca11071c410978

Not much else I can do...

sorear, 2016-09-07:
The bug is very difficult to reproduce; unlike #177 which affected everything, this only affects one program out of thousands we've compiled.  I can easily imagine that any change in code generation or relocation would mask it.  *We have a workaround, see above.*  I'm willing to stop work on this bug for a while, after we get the full build environment on Fedora we're likely to have a lot more examples and with a current compiler.

@aswaterman In general, that relaxation code scares me; since relocations are relaxed in no particular order (I can't find a sort call, and the ELF spec says nothing about relocation order in .o files), is there anything preventing a situation where one instruction of a two-instruction sequence is relaxed and the other isn't, resulting in broken code?

aswaterman, 2016-09-07:
@sorear It may be that the LUI relaxations are ill-conceived, but preventing them against code segments could suffice.  If the data segment doesn't change size as a result of relaxation, it should be fine...

sorear, 2016-09-07:
@aswaterman Relocation is possible to do safely, but:

1. It requires new promises from the compiler that need to be very carefully documented
2. If we want to support code that violates those promises but is not intended to be relaxable (I can think of some possibilities), we may need to add new relocation types; for safety it'd be best to make the carefully defined relaxable types new, and make the old types non-relaxable
3. I don't know if this is relevant to the problem at hand --- there might be a subtle bug in the relaxer _distinct_ from the specification problem

Let me know if you want me to write something for the ML about relaxation in RISC-V (a general approach: what compiler and linker writers need to do for safety and effectiveness)

aswaterman, 2016-09-07:
@sorear I agree we should have a coherent strategy and I would be interested in hearing more of your thoughts.

As to the problem at hand, the relaxer is making an incorrect assumption about the invariance of (symbol - gp).  I'm just not certain why.  My best guess is that a data alignment directive exists between symbol and gp, and so the magnitude of (symbol - gp) may increase as a result of the relaxation.

sorear, 2016-09-07:
@aswaterman the relaxer is making an incorrect assumption about the invariance of the value of `_gp` _itself_.  I ran ld under a debugger and the assumed value of _gp was different by a dozen bytes or so from the final value in the linked executable.  Since there are no text segments before _gp, something else is going on?

sorear, 2016-09-07:
@aswaterman OK, I'll write up the conditions for safe relaxation tomorrow-ish

aswaterman, 2016-09-07:
What I meant was, it's fine if _gp moves, as long as the symbols it references move with it.  Since it is anchored to the base of one of the data sections, perhaps the correct fix is to only allow relaxations to symbols in that section?

kito-cheng, 2016-09-12:
We have fix a bug for relaxing gp-relative instruction, maybe you can try it https://github.com/kito-cheng/riscv-binutils-gdb/commit/64f1b5dfe90f4d0e264a9f13942aa29d2c8cb44e

davidlt, 2016-09-12:
Will do that later today or tomorrow. Stay tuned.

davidlt, 2016-09-16:
Sorry for the delay. Building a native GCC in QEMU takes a while.

sorear, 2016-10-14:
We have a gcc and a gmp now and for unknown reasons gmp no longer hits the bug case.

Now that I know %pcrel_lo doesn't actually point directly to the symbol, I have a much better idea of how a fully correct relaxer needs to work; there are some vague proposals in the "[sw-dev] RISC-V Binutils and GDB Upstreaming Progress" thread (which I kinda derailed, sorry).

I still have very little idea about bfd's internals, but that can be fixed if necessary.

palmer-dabbelt, 2017-03-30:
I'm going to close this assuming it's fixed.  There's been a lot of relaxation fixes lately, and we pass most of the GCC test suite.  If this is old, feel free to open a new bug.

closed 2017-03-30 by palmer-dabbelt