riscv-collab/riscv-gnu-toolchain #180: Warnings from gcc/config/riscv/riscv.md (warning: operand 0 missing mode?)

davidlt, 2016-09-16:
While building a native GCC RPM for Fedora I notice the following warnings:
```
//rpmbuild/BUILD/gcc-6.1.0-20160912/obj-riscv64-redhat-linux/./prev-gcc/xg++ -B//rpmbuild/BUILD/gcc-6.1.0-20160912/obj-riscv64-redhat-linux/./prev-gcc/ -B/usr/riscv64-redhat-linux/bin/ -nostdinc++ -B//rpmbuild/BUILD/gcc-6.1.0-20160912/obj-riscv64-redhat-linux/prev-riscv64-redhat-linux/libstdc++-v3/src/.libs -B//rpmbuild/BUILD/gcc-6.1.0-20160912/obj-riscv64-redhat-linux/prev-riscv64-redhat-linux/libstdc++-v3/libsupc++/.libs  -I//rpmbuild/BUILD/gcc-6.1.0-20160912/obj-riscv64-redhat-linux/prev-riscv64-redhat-linux/libstdc++-v3/include/riscv64-redhat-linux  -I//rpmbuild/BUILD/gcc-6.1.0-20160912/obj-riscv64-redhat-linux/prev-riscv64-redhat-linux/libstdc++-v3/include  -I//rpmbuild/BUILD/gcc-6.1.0-20160912/libstdc++-v3/libsupc++ -L//rpmbuild/BUILD/gcc-6.1.0-20160912/obj-riscv64-redhat-linux/prev-riscv64-redhat-linux/libstdc++-v3/src/.libs -L//rpmbuild/BUILD/gcc-6.1.0-20160912/obj-riscv64-redhat-linux/prev-riscv64-redhat-linux/libstdc++-v3/libsupc++/.libs   -O2 -g -Wall -Werror=format-security -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -fprofile-use -DIN_GCC     -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -DGENERATOR_FILE -fno-PIE -static-libstdc++ -static-libgcc  -no-pie -o build/genrecog \
    build/genrecog.o build/rtl.o build/read-rtl.o build/ggc-none.o build/vec.o build/min-insn-modes.o build/gensupport.o build/print-rtl.o build/hash-table.o build/read-md.o build/errors.o build/inchash.o .././libiberty/libiberty.a
build/genrecog ../../gcc/common.md ../../gcc/config/riscv/riscv.md \
  insn-conditions.md > tmp-recog.c
../../gcc/config/riscv/riscv.md:2258: warning: operand 0 missing mode?
../../gcc/config/riscv/riscv.md:2278: warning: operand 1 missing mode?
../../gcc/config/riscv/riscv.md:2288: warning: operand 1 missing mode?
../../gcc/config/riscv/riscv.md:2312: warning: operand 0 missing mode?
../../gcc/config/riscv/riscv.md:2334: warning: operand 1 missing mode?
../../gcc/config/riscv/riscv.md:2346: warning: operand 1 missing mode?
Statistics for recog:
  Number of decisions:   1489
  longest path:            29 (code:    264)
  longest backtrack:        6 (code:    272)
Statistics for split_insns:
  Number of decisions:    101
  longest path:            30 (code:     23)
  longest backtrack:        3 (code:     23)
Statistics for peephole2_insns:
  Number of decisions:     46
  longest path:            33 (code:      4)
  longest backtrack:        1 (code:      3)
Shared 936 out of 2560 states by creating 272 new states, saving 664
/bin/sh ../../gcc/../move-if-change tmp-recog.c insn-recog.c
echo timestamp > s-recog
```

aswaterman, 2016-09-16:
Yeah, it is benign.  We tried to fix the warnings a while ago, and the obvious approach failed for a reason I can't recall.  We would happily accept a patch that fixes the warnings, but it's low priority for us.

We should keep this issue open for bookkeeping.

palmer-dabbelt, 2016-09-16:
For reference, here's what didn't work https://github.com/riscv/riscv-gnu-toolchain/pull/29.

aswaterman, 2016-09-16:
And the revert: https://github.com/riscv/riscv-gnu-toolchain/commit/f36ceb5804a510dd9c8bcd030a0969b0fa686ad0

TommyMurphyTM1234, 2022-09-29:
Benign warnings as explained here and in linked content by @aswaterman and @jim-wilson .

closed 2022-09-29 by TommyMurphyTM1234