riscv-collab/riscv-gnu-toolchain #181: CLOCK_BOOTTIME undeclared on redhat

colinschmidt, 2016-09-19:
Building a recent toolchain results in the following error on a redhat 6.7 system with devtoolset-2 enabled
```
/tools/projects/colins/clean-chip/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/sim/riscv/sim-main.c: In function ‘fetch_csr’:
/tools/projects/colins/clean-chip/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/sim/riscv/sim-main.c:93:21: error: ‘CLOCK_BOOTTIME’ undeclared (first use in this function)
  if (clock_gettime (CLOCK_BOOTTIME, &ts) == 0)
```

aswaterman, 2016-09-19:
this was fixed by https://github.com/riscv/riscv-binutils-gdb/commit/1a49dc378f60b9bc6939dabaabf64b4a1946f5d4

colinschmidt, 2016-09-19:
ok thanks, I guess rocket-chip is just a bit behind the tools

closed 2016-09-19 by colinschmidt